Koenkk/zigbee2mqtt #10915: Failed to interview ZB-RGBCW Lonsonho bulb

JeanProfite, 2022-01-20:
### What happened?

I tried several times to pair my 3 new light bulbs ([those](https://fr.aliexpress.com/item/1005003297096190.html?gatewayAdapt=glo2fra&spm=a2g0o.9042311.0.0.3bb56c37oIkGJe)) and the interview fail all time.
I followed the FAQ for this issue but non of the solution worked for me.

### What did you expect to happen?

The light bulbs interview get successful.

### How to reproduce it (minimal and precise)

Use RPI3 B+ with home assistant supervised.

Turn on/off the bulbs 5 times to go in pairing mode.
Activate Zigbee discovery.
Wait until the interview fail.

### Zigbee2MQTT version

1.22.2-dev (commit #a867b71)

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7907276/log.txt)

Koenkk, 2022-01-20:
Could you provide the herdsman debug log when pairing this device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

JeanProfite, 2022-01-20:
Of course, but I'm not sure to understand how to enable it and where to find it.
-Where I am supposed to put "DEBUG=zigbee-herdsman* npm start"?
-And then how to see STDOUT/STDERR in homeassisatant supervised?

JeanProfite, 2022-01-20:
Ok, normally there it is, thanks for the help.

[log_herdsman_en.txt](https://github.com/Koenkk/zigbee2mqtt/files/7908548/log_herdsman_en.txt)

Koenkk, 2022-01-21:
Thanks, could you provide the same logging again but then before pairing it force remove the device through the z2m frontend and restart z2m, then try to pair the device and provide the log here.

JeanProfite, 2022-01-21:
[log_herdsman_en_2.txt](https://github.com/Koenkk/zigbee2mqtt/files/7914615/log_herdsman_en_2.txt)
And on the frontend i see this:
[image]

Koenkk, 2022-01-21:
- Are you trying to pair multiple bulbs at the same time?
- Can the bulbs which are identified as `ZB-RGBCW` be controlled? (e.g turn on/off, change color)

JeanProfite, 2022-01-21:
- Yes i'm pairing 3 bulb at the same time. 
- No.

Here is the log ([log_herdsman_en_3.txt](https://github.com/Koenkk/zigbee2mqtt/files/7915666/log_herdsman_en_3.txt)) when I pair only one bulb and at the end I tried to turn it off via this pannel: 

[image]
Which didn't work and give me this message on the frontend:
`Publish 'set' 'state' to '0x00124b002450dce0' failed: 'Error: Command 0x00124b002450dce0/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 46455 - 1 - 7 - 6 - 11 after 10000ms)'`

And the multiple time I tried to pair the bulb, sometimes I see the `LQI` and some time not. It's the same for `Power`.

Koenkk, 2022-01-21:
For some reason the bulb indeed has trouble pairing. Could you try to pair the bulbs while having them close to the coordinator? (to ensure they pair via the coordinator and not via a router)

JeanProfite, 2022-01-21:
I only have a sonoff SNZB-03 (PIR) and SNZB-01 (Button) so no router.
Also I tried to pair the bulb as close as possible from the coordinator (10cm) and again had the same issue, "interview failed".

Koenkk, 2022-01-22:
The interview failing should be fixed now, but I'm not sure if it is controllable afterwards, can you check this?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

JeanProfite, 2022-01-22:
I tried to pair it again, i know it's too early the addon point on the commit [#a867b71](https://github.com/Koenkk/zigbee2mqtt/commit/a867b719dcdd5a02c8f3dd2ab32475d9d3ef0275) but I had this screen:
[image]
I put here the [initialState.json.zip](https://github.com/Koenkk/zigbee2mqtt/files/7918515/initialState.json.zip) and the [log_herdsman_en_4.txt](https://github.com/Koenkk/zigbee2mqtt/files/7918517/log_herdsman_en_4.txt), maybe it's useful for you ?

I will retry when the addon will point on the last commit, thank you 👍 .

JeanProfite, 2022-01-22:
:heavy_check_mark:The pairing issue is now resolved, all the interview was successful.
[image]

:x: But I can't control them I have this error (tried to switch off and change the brightness):
```
Zigbee2MQTT:debug 2022-01-22 15:50:53: Received MQTT message on 'zigbee2mqtt/0x00124b002450d596/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2022-01-22 15:50:53: Publishing 'set' 'state' to '0x00124b002450d596'
Zigbee2MQTT:debug 2022-01-22 15:51:06: Received MQTT message on 'zigbee2mqtt/0x00124b002450d596/set' with data '{"brightness":138}'
Zigbee2MQTT:debug 2022-01-22 15:51:06: Publishing 'set' 'brightness' to '0x00124b002450d596'
Zigbee2MQTT:error 2022-01-22 15:51:16: Publish 'set' 'state' to '0x00124b002450d596' failed: 'Error: Command 0x00124b002450d596/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24077 - 1 - 63 - 6 - 11 after 10000ms)'
Zigbee2MQTT:debug 2022-01-22 15:51:16: Error: Command 0x00124b002450d596/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24077 - 1 - 63 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
Zigbee2MQTT:info  2022-01-22 15:51:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x00124b002450d596' failed: 'Error: Command 0x00124b002450d596/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 24077 - 1 - 63 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x00124b002450d596"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2022-01-22 15:51:39: Publish 'set' 'brightness' to '0x00124b002450d596' failed: 'Error: Command 0x00124b002450d596/1 genLevelCtrl.moveToLevelWithOnOff({"level":138,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24077 - 1 - 64 - 8 - 11 after 10000ms)'
Zigbee2MQTT:debug 2022-01-22 15:51:39: Error: Command 0x00124b002450d596/1 genLevelCtrl.moveToLevelWithOnOff({"level":138,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24077 - 1 - 64 - 8 - 11 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
Zigbee2MQTT:info  2022-01-22 15:51:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to '0x00124b002450d596' failed: 'Error: Command 0x00124b002450d596/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":138,\"transtime\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 24077 - 1 - 64 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"0x00124b002450d596"},"type":"zigbee_publish_error"}'
```

Koenkk, 2022-01-23:
@JeanProfite seems the device somehow isn't connecting, could you make a sniff when pairing the device to z2m?

In case you also have the original hub, also provide a sniff when pairing it to that.

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

JeanProfite, 2022-01-23:
I'm currently trying to follow the sniffing procedure but stuck when launching Wireshark via ZBOSS. Apparently, ZBOSS does not detect the USB dongle :
[image]

I have flashed the dongle with the sniffing firmware and windows recognized it:
[image]

I don't have the original hub.

If i understand i need to have 2 Zigbee USB sticks? One for sniffing with my laptop and one for z2m?

Koenkk, 2022-01-24:
> If i understand i need to have 2 Zigbee USB sticks? 

Yes, what usb are you sniffing with?

JeanProfite, 2022-01-24:
I have 2 Sonoff Zigbee 3.0 USB Dongle Plus and yes the one I use to sniff is flashed with the sniffing firmware.

luizcorreia, 2022-01-25:
Same here, there is a solution?

luizcorreia, 2022-01-25:
Here my log:
```
Error Publish 'set' 'brightness' to 'Bathroom Light Bulb' failed: 'Error: Command 0x00124b0023b4cb39/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":450}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 59971 - 1 - 3 - 8 - 11 after 10000ms)'
Debug Error: Command 0x00124b0023b4cb39/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":450}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 59971 - 1 - 3 - 8 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'Bathroom Light Bulb' failed: 'Error: Command 0x00124b0023b4cb39/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":254,\"transtime\":450}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 59971 - 1 - 3 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"Bathroom Light Bulb"},"type":"zigbee_publish_error"}'
Debug Publishing 'set' 'transition' to 'Bathroom Light Bulb'
Debug Publishing 'set' 'color_temp' to 'Bathroom Light Bulb'
Debug Received MQTT message on 'zigbee2mqtt/Bathroom Light Bulb/set' with data '{"state": "ON", "transition": 45.0, "brightness": 254, "color_temp": 188}'
Debug Publishing 'set' 'brightness' to 'Bathroom Light Bulb'
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"rnhkc-1","value":true}'
```

Koenkk, 2022-01-25:
@JeanProfite not sure why it doesn't work, personally I always sniff with a CC2531 + wireshark and this works flawlessly.

luizcorreia, 2022-01-25:
I put the bulb on ZHA end works like a charm.
There is something I can do to help solve this. 
I am in love with Z2M.

JeanProfite, 2022-01-26:
> @JeanProfite not sure why it doesn't work, personally I always sniff with a CC2531 + wireshark and this works flawlessly.

Ok, I have tried also on linux but I just saw that my dongle is not based on CC2531 but a CC2652P. So obvyously,   ZBOSS and whsniff don't recognise it. So i don't know how to do ..

> I put the bulb on ZHA end works like a charm. There is something I can do to help solve this. I am in love with Z2M.

@luizcorreia  Maybe you can try to sniff the trafic like mentioned here ?

> @JeanProfite seems the device somehow isn't connecting, could you make a sniff when pairing the device to z2m?
> 
> In case you also have the original hub, also provide a sniff when pairing it to that.
> 
> I've reverted the interviewing fix since it doesn't solve the actual problem (device not staying in the network somehow)
> 
> https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

Koenkk, 2022-01-26:
@luizcorreia it would be great if you could provide a sniff when pairing it to z2m and to zha so I can check for any differences (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531)

luizcorreia, 2022-01-26:
> @luizcorreia it would be great if you could provide a sniff when pairing it to z2m and to zha so I can check for any differences (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531)

I had just one module, can´t sniff. I'll buy a new one.

closed 2022-03-05 by github-actions[bot]

JeanProfite, 2022-05-20:
Hello, I just wanted to say that the issue is now fixed !