Koenkk/zigbee2mqtt #10923: MiBoxer FUT039Z RGBCCT - Colors not working

plasticrake, 2022-01-21:
### What happened?

I've paired two MiBoxer FUT039Z RGBCCT devices and they both have the same behavior. I can only change the colors to cool white or warm white, and that is done incorrectly. When set to warm the cool LEDs are lit up. When it is set to Cool, the warm and cool LEDs are lit up. I cannot set them to RGB.

Someone else reported this issue here: #10762

### What did you expect to happen?

White color temperature to match.

RGB colors to be set.

### How to reproduce it (minimal and precise)

Set the color temperature, it will change but it will be incorrect.

Set an RGB color, nothing happens.

### Zigbee2MQTT version

1.22.2 commit: 414c51f

### Adapter firmware version

20211217

### Adapter

SMARTLIGHT CC2652P

### Debug log

```
debug Received MQTT message on 'zigbee2mqtt/Studio Cabinet/set' with data '{"color_temp":153}'
debug Skipping state because of Home Assistant
debug Publishing 'set' 'color_temp' to 'Studio Cabinet'
info MQTT publish: topic 'zigbee2mqtt/Studio Cabinet', payload '{"brightness":254,"color_mode":"color_temp","color_power_on_behavior":null,"color_temp":500,"do_not_disturb":null,"last_seen":"2022-01-20T21:02:05-08:00","linkquality":58,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/Studio Cabinet', payload '{"brightness":254,"color_mode":"color_temp","color_power_on_behavior":null,"color_temp":500,"do_not_disturb":null,"last_seen":"2022-01-20T21:02:05-08:00","linkquality":61,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/Studio Cabinet', payload '{"brightness":254,"color_mode":"color_temp","color_power_on_behavior":null,"color_temp":500,"do_not_disturb":null,"last_seen":"2022-01-20T21:02:08-08:00","linkquality":58,"state":"ON"}'
debug Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x84fd27fffe3b0fcd!
info MQTT publish: topic 'zigbee2mqtt/Studio Cabinet', payload '{"brightness":254,"color_mode":"color_temp","color_power_on_behavior":null,"color_temp":153,"do_not_disturb":null,"last_seen":"2022-01-20T21:02:08-08:00","linkquality":58,"state":"ON"}'
debug Received MQTT message on 'zigbee2mqtt/Studio Cabinet/set' with data '{"color_temp":500}'
debug Skipping state because of Home Assistant
debug Publishing 'set' 'color_temp' to 'Studio Cabinet'
info MQTT publish: topic 'zigbee2mqtt/Studio Cabinet', payload '{"brightness":254,"color_mode":"color_temp","color_power_on_behavior":null,"color_temp":153,"do_not_disturb":null,"last_seen":"2022-01-20T21:02:24-08:00","linkquality":54,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/Studio Cabinet', payload '{"brightness":254,"color_mode":"color_temp","color_power_on_behavior":null,"color_temp":153,"do_not_disturb":null,"last_seen":"2022-01-20T21:02:24-08:00","linkquality":54,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/Studio Cabinet', payload '{"brightness":254,"color_mode":"color_temp","color_power_on_behavior":null,"color_temp":153,"do_not_disturb":null,"last_seen":"2022-01-20T21:02:25-08:00","linkquality":58,"state":"ON"}'
debug Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x84fd27fffe3b0fcd!
info MQTT publish: topic 'zigbee2mqtt/Studio Cabinet', payload '{"brightness":254,"color_mode":"color_temp","color_power_on_behavior":null,"color_temp":500,"do_not_disturb":null,"last_seen":"2022-01-20T21:02:25-08:00","linkquality":58,"state":"ON"}'
debug Received MQTT message on 'zigbee2mqtt/Studio Cabinet/set' with data '{"color":{"hue":354,"saturation":100}}'
debug Skipping state because of Home Assistant
debug Publishing 'set' 'color' to 'Studio Cabinet'
error Publish 'set' 'color' to 'Studio Cabinet' failed: 'TypeError: Cannot read property 'h' of undefined'
debug TypeError: Cannot read property 'h' of undefined at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:3073:60) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52) at EventEmitter.emit (events.js:412:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:102:22) at MQTT.onMessage (/app/lib/mqtt.ts:115:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27) at WebSocket.emit (events.js:400:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1098:20) at Receiver.emit (events.js:400:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:528:14)
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'color' to 'Studio Cabinet' failed: 'TypeError: Cannot read property 'h' of undefined'","meta":{"friendly_name":"Studio Cabinet"},"type":"zigbee_publish_error"}'
debug Received MQTT message on 'zigbee2mqtt/Studio Cabinet/set' with data '{"color":{"hue":131,"saturation":100}}'
debug Skipping state because of Home Assistant
debug Publishing 'set' 'color' to 'Studio Cabinet'
error Publish 'set' 'color' to 'Studio Cabinet' failed: 'TypeError: Cannot read property 'h' of undefined'
debug TypeError: Cannot read property 'h' of undefined at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:3073:60) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52) at EventEmitter.emit (events.js:412:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:102:22) at MQTT.onMessage (/app/lib/mqtt.ts:115:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27) at WebSocket.emit (events.js:400:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1098:20) at Receiver.emit (events.js:400:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:528:14)
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'color' to 'Studio Cabinet' failed: 'TypeError: Cannot read property 'h' of undefined'","meta":{"friendly_name":"Studio Cabinet"},"type":"zigbee_publish_error"}'
debug Received MQTT message on 'zigbee2mqtt/Studio Cabinet/set' with data '{"color":{"hue":240,"saturation":100}}'
debug Skipping state because of Home Assistant
debug Publishing 'set' 'color' to 'Studio Cabinet'
error Publish 'set' 'color' to 'Studio Cabinet' failed: 'TypeError: Cannot read property 'h' of undefined'
debug TypeError: Cannot read property 'h' of undefined at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:3073:60) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52) at EventEmitter.emit (events.js:412:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:102:22) at MQTT.onMessage (/app/lib/mqtt.ts:115:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27) at WebSocket.emit (events.js:400:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1098:20) at Receiver.emit (events.js:400:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:528:14)
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'color' to 'Studio Cabinet' failed: 'TypeError: Cannot read property 'h' of undefined'","meta":{"friendly_name":"Studio Cabinet"},"type":"zigbee_publish_error"}'
```

captainbalu-hu, 2022-01-29:
Hi,

Do you have any progress since?
I've 3 of these controllers and having the same issue with the CCT, but I can control RGB (although is not perfect either).

I'm just barely scratching the surface about this Zigbee + MQTT thing, these are my very first devices. 
Using the [FUT039Z description](https://www.zigbee2mqtt.io/devices/FUT039Z.html) I found my way to send payload messages directly and have some interesting results.

The device has 2 mode of operation, CCT or RGB. So far I couldn't manage to adjust all the 5 LEDs (RGB+CW+WW) at once.
In the device status there is a "color_mode" which can be "color_temp" in CCT and "hs" in RGB mode.
To switch between the 2 mode just need to send a color value in hue/saturation or a color_temp value.
By the documentation in CCT mode the value range is 153-500 mired.
Just in your case, my device also has both CW and WW lighting at level 153 and CW remains on at 500.
Out of curiosity I've tried to send out of range values.
Greatest surprise, the controller responded and at value -171(! yes negative !) the CW was off and WW on. 
-170 was the last step with CW slightly lighting.
On the other side there is no change in LED's state above 281.
The usable range is -171 to 282 with negative being the warmest.
So it has 453 steps meanwhile the mired range has only 347.
The debug log is full of errors about the <153 values, however the controller does the job.

In RGB mode I can adjust a the color within a hue range of 1-361. If I set hue to 0, the controller set itself to hue:142. 
The same happen if I try to adjust brightness. Whatever was the color it turns to hue:142 and no change in brightness.
Also interesting that once I sent brightness value in RGB mode - with error - and then switch to CCT mode, the controller applied the previously sent brightness. Switching back to RGB mode will keep the brightness.

I've tried most of the commands from the documentation but none of the transition/moving/stepping worked. Neither the HSB, HSV, HSL forms of color settings.

It would be nice if you could try some of these (like the CCT range) to confirm my observations.

captainbalu-hu, 2022-01-31:
So, I tried to figure out these negative values.
According to the [miboxer.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/miboxer.js) script the system uses the 'tuya_led_control' function to convert to Zigbee.
`toZigbee: [tz.on_off, tz.tuya_led_control, tz.tuya_do_not_disturb, tz.tuya_color_power_on_behavior],`
In this part there is a range mapping which converts the 500 - 154 mired range to 0 - 254 where 500 the "warmest" goes to 0 and 154 the "coolest" goes to 254. 
`const zclData = {colortemp: utils.mapNumberRange(meta.message.color_temp, 500, 154, 0, 254), transtime: 0};`
I checked the mapping with my observed values.
It turns out that -172(WW) goes to 493 and 262(CW) goes to 175 after the mapping.
This led me to the conclusion that my device is waiting the color_temp values in mired scale instead of 0-254 values.
I'll try to make a new converter file to see if there is a way to correct this problem by that.

If anyone has any thoughts about this, please share. Thx.

captainbalu-hu, 2022-02-09:
I don't have any official remote to control the device, so I couldn't make a new converter file.
I'm planning to purchase a Tuya gateway to able to [find the right data points](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats).
Until then, I've changed my toZigbee and fromZigbee converter files. 
I didn't want to mess up with other tuya lights so I created a new 'tuya_led_control_miboxer' function based on the original and removed all range conversion from colortemp controls.
Then I changed the miboxer.js script to use my modified function and now my CCT LEDs are work as they supposed to.
At the 'coolest' preset only the CW LED is ON, at the 'warmest' only the WW is ON and the full range can be used to set the required colortemp.

In the meantime I contacted the manufacturer.
According to them, this LED controller supposed to drive both the RGB and CW+WW LEDs at the same time.
Right now Z2M controls it as it would be two exlusive modes of the device. Is it possible to change this?

plasticrake, 2022-02-09:
@captainbalu-hu can you publish a fork of your changes?

plasticrake, 2022-02-15:
Thanks @captainbalu-hu, I think the fix might be a little simpler than that. I just submitted a PR against [zigbee-herdsmann-converters](https://github.com/Koenkk/zigbee-herdsman-converters) that fixed the color temp and color issue for me.

captainbalu-hu, 2022-02-16:
Well... I think too. 
As said in my first comment I've just jumped in to this world whit these products.
I need to learn a lot. (I'm not a programmer JS is also new to me... :D )
I considered my "fix" as a proof-of-concept for the CCT issue, not a final solution.
Thanks @plasticrake for the PR. I've checked and going to try that solution with my devices.
However I will go to a very long trip on soon, so I won't have access to my stuff until April.
But then, I'd like to make them work properly.
Could you manage to use both the white and RGB at the same time?

zatloeri, 2023-04-23:
Is the RGB and CW+WW supposed to work correctly after this fix? It does not seem like its working on my end. I can either choose a color or warmth (RGB or CCT). Not both at the same time. @captainbalu-hu mentioned they should work like that though.

Georgeeden, 2024-10-23:
> I don't have any official remote to control the device, so I couldn't make a new converter file. I'm planning to purchase a Tuya gateway to able to [find the right data [...]

Hi @captainbalu-hu, Did you receive an update from MiBoxer / MiLight regarding both modes at the same time? Would be great to have the ability to use the white channels to make more pastel colours.