Koenkk/zigbee2mqtt #10927: Migrate coordinator from CC2531 To CC2652P

fabiobaldoibs, 2022-01-21:
### What happened?

The coordinator can't communicate with any device.

### What did you expect to happen?

The coordinator is able to communicate with all devices without a repair it

### How to reproduce it (minimal and precise)

- First off all I stopped 2MQTT Service; 
- I get the IEEE address from my CC2531 ("Settings" -> "About" -> "Coordinator IEEE Address") 
- I put the CC2652P to bootloader mode; 
- I used the ZigStar too to write the CC2531 IEEE address to CC2652P;
- I put the CC2652P to bootloader mode again;
- I flashed the firmware "CC1352P2_CC2652P_launchpad_coordinator_20220103"
- I'v installed the stick on my raspberry
- I'v changed the port configuration to "port: /dev/ttyACM0"
- I'v started 2MQTT Service; 
- I looked in the log, and none error are reported;
- I checked the IEEE address to guarantee that was the same; 

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20220103

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

kevincaradant, 2022-01-24:
I think you should paste the logs and the config of the coordinator to have some help about this.

BTW: If the channel is not the same than before, all devices will need to be re-paired.

Maybe this FAQ can help ? 
https://www.zigbee2mqtt.io/guide/faq/#how-do-i-migrate-from-one-adapter-to-another

I saw for example: 
```
If repairing was not required and your devices do not respond; restart some routers by removing them from the mains power for a few seconds.
```

or 

```
Requires repairing
You need to re-pair all you devices when:

Changing the network key (network_key), Zigbee channel (channel) or panID (pan_id) in configuration.yaml.
Switching between adapters requires repairing, except when:
When adapters are the same repairing is not required (e.g. CC2531 -> CC2531)
Switching from a CC2530 or CC2531 based adapter to a CC2538/CC2652/CC1352 based adapter does not require repairing (e.g. CC2531 -> CC2652)
Switching between CC2538, CC2652 and CC1352 based adapters does not require repairing (e.g. CC2538 -> CC2652
```

Are you concern ?

fabiobaldoibs, 2022-01-24:
I did the re-pair after all

closed 2022-01-24 by fabiobaldoibs