Koenkk/zigbee2mqtt #10934: Following update to 1.22.2; Nue 2-gang switch are detected as double GPO

jyavenard, 2022-01-21:
### What happened?

Recently did an update

Need to éd thst my two-gang switches no longer worked.

They now show as a double gang GPO instead. Attempting to unpair/pair them again didn't fix the issue.

[image]

### What did you expect to happen?

Things would continue to work as before, no change in behaviour of devices.

### How to reproduce it (minimal and precise)

Pair dual-gang Nue zigbee seitch

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210120

### Adapter

Zig-a-ziza

### Debug log

Zigbee2MQTT:info  2022-01-22 00:48:02: Device '0x00158d000462238d' joined
Zigbee2MQTT:info  2022-01-22 00:48:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d000462238d","ieee_address":"0x00158d000462238d"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-01-22 00:48:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d000462238d"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-01-22 00:48:02: MQTT publish: topic 'homeassistant/switch/0x00158d000462238d/switch_left/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x00158d000462238d/left/set","device":{"identifiers":["zigbee2mqtt_0x00158d000462238d"],"manufacturer":"Nue / 3A","model":"Smart Zigbee double power point (NUE-AUWZO2)","name":"0x00158d000462238d","sw_version":"1000-0001"},"json_attributes_topic":"zigbee2mqtt/0x00158d000462238d","name":"0x00158d000462238d left","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x00158d000462238d","unique_id":"0x00158d000462238d_switch_left_zigbee2mqtt","value_template":"{{ value_json.state_left }}"}'
Zigbee2MQTT:info  2022-01-22 00:48:02: MQTT publish: topic 'homeassistant/switch/0x00158d000462238d/switch_right/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x00158d000462238d/right/set","device":{"identifiers":["zigbee2mqtt_0x00158d000462238d"],"manufacturer":"Nue / 3A","model":"Smart Zigbee double power point (NUE-AUWZO2)","name":"0x00158d000462238d","sw_version":"1000-0001"},"json_attributes_topic":"zigbee2mqtt/0x00158d000462238d","name":"0x00158d000462238d right","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x00158d000462238d","unique_id":"0x00158d000462238d_switch_right_zigbee2mqtt","value_template":"{{ value_json.state_right }}"}'
Zigbee2MQTT:info  2022-01-22 00:48:02: MQTT publish: topic 'homeassistant/sensor/0x00158d000462238d/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000462238d"],"manufacturer":"Nue / 3A","model":"Smart Zigbee double power point (NUE-AUWZO2)","name":"0x00158d000462238d","sw_version":"1000-0001"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00158d000462238d","name":"0x00158d000462238d linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d000462238d","unique_id":"0x00158d000462238d_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2022-01-22 00:48:02: Starting interview of '0x00158d000462238d'
Zigbee2MQTT:info  2022-01-22 00:48:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d000462238d","ieee_address":"0x00158d000462238d","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-01-22 00:48:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d000462238d"},"type":"pairing"}'
Zigbee2MQTT:info  2022-01-22 00:48:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d000462238d","ieee_address":"0x00158d000462238d"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-01-22 00:48:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d000462238d"},"type":"device_announced"}'
Zigbee2MQTT:info  2022-01-22 00:48:03: Successfully interviewed '0x00158d000462238d', device has successfully been paired
Zigbee2MQTT:info  2022-01-22 00:48:03: Device '0x00158d000462238d' is supported, identified as: Nue / 3A Smart Zigbee double power point (NUE-AUWZO2)
Zigbee2MQTT:info  2022-01-22 00:48:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart Zigbee double power point","exposes":[{"endpoint":"left","features":[{"access":7,"description":"On/off state of the switch","endpoint":"left","name":"state","property":"state_left","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"right","features":[{"access":7,"description":"On/off state of the switch","endpoint":"right","name":"state","property":"state_right","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"NUE-AUWZO2","options":[],"supports_ota":false,"vendor":"Nue / 3A"},"friendly_name":"0x00158d000462238d","ieee_address":"0x00158d000462238d","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2022-01-22 00:48:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart Zigbee double power point","friendly_name":"0x00158d000462238d","model":"NUE-AUWZO2","supported":true,"vendor":"Nue / 3A"},"type":"pairing"}'
Zigbee2MQTT:info  2022-01-22 00:48:59: MQTT publish: topic 'zigbee2mqtt/0x00158d000462238d', payload '{"linkquality":75,"state_left":null,"state_right":null,"state_undefined":"OFF"}'
Zigbee2MQTT:info  2022-01-22 00:48:59: MQTT publish: topic 'zigbee2mqtt/0x00158d000462238d', payload '{"linkquality":75,"state_left":null,"state_right":null,"state_undefined":"OFF"}'

jyavenard, 2022-01-21:
Restoring previous 1.21.2; and the switches show up again as expected
```
Warning Device '0x00158d000462238d' left the network
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"ieee_address":"0x00158d000462238d"},"type":"device_leave"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x00158d000462238d"},"type":"device_removed"}'
Info Zigbee: allowing new devices to join.
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"zi9m8-3"}'
Info Device '0x00158d000462238d' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d000462238d","ieee_address":"0x00158d000462238d"},"type":"device_joined"}'
Info Configuring '0x00158d000462238d'
Info MQTT publish: topic 'homeassistant/switch/0x00158d000462238d/switch_top/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00158d000462238d/availability"}],"command_topic":"zigbee2mqtt/0x00158d000462238d/top/set","device":{"identifiers":["zigbee2mqtt_0x00158d000462238d"],"manufacturer":"Nue / 3A","model":"Smart light switch - 2 gang v2.0 (HGZB-42)","name":"0x00158d000462238d","sw_version":"Zigbee2MQTT 1.21.2"},"json_attributes_topic":"zigbee2mqtt/0x00158d000462238d","name":"0x00158d000462238d top","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x00158d000462238d","unique_id":"0x00158d000462238d_switch_top_zigbee2mqtt","value_template":"{{ value_json.state_top }}"}'
Info MQTT publish: topic 'homeassistant/switch/0x00158d000462238d/switch_bottom/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00158d000462238d/availability"}],"command_topic":"zigbee2mqtt/0x00158d000462238d/bottom/set","device":{"identifiers":["zigbee2mqtt_0x00158d000462238d"],"manufacturer":"Nue / 3A","model":"Smart light switch - 2 gang v2.0 (HGZB-42)","name":"0x00158d000462238d","sw_version":"Zigbee2MQTT 1.21.2"},"json_attributes_topic":"zigbee2mqtt/0x00158d000462238d","name":"0x00158d000462238d bottom","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x00158d000462238d","unique_id":"0x00158d000462238d_switch_bottom_zigbee2mqtt","value_template":"{{ value_json.state_bottom }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x00158d000462238d/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00158d000462238d/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00158d000462238d"],"manufacturer":"Nue / 3A","model":"Smart light switch - 2 gang v2.0 (HGZB-42)","name":"0x00158d000462238d","sw_version":"Zigbee2MQTT 1.21.2"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00158d000462238d","name":"0x00158d000462238d linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d000462238d","unique_id":"0x00158d000462238d_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d000462238d"},"type":"device_connected"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d000462238d","ieee_address":"0x00158d000462238d"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d000462238d"},"type":"device_announced"}'
```

jyavenard, 2022-01-21:
Interestingly, the model number is showing as the same in 1.21.2

Friendly name
master_switch
Last seen
N/A
Device type
Router
Zigbee Model
LXN-2S27LX1.0
Zigbee Manufacturer
3A Smart Home DE
Description
Smart light switch - 2 gang v2.0
Support status
Supported
IEEE Address
0x00158d000462238d
Network address
0x170B
Firmware build date
20190719
Firmware version
1000-0001
Manufacturer
Nue / 3A
Model
HGZB-42
Power
Interview completed
True

jyavenard, 2022-01-21:
Same as https://github.com/Koenkk/zigbee2mqtt/issues/10497

Doest that mean that 1.22.2 release isn't as recent as 1.22.2-dev?

Koenkk, 2022-01-21:
> Doest that mean that 1.22.2 release isn't as recent as 1.22.2-dev?

Yes, -dev comes after the release. You can switch to the dev branch or wait for the next release on 1 February. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ManfredBartz, 2022-01-23:
Same issue here.  Will try dev branch.
Nitpick: naming endpoints "right" and "left" does not help when the device is or can be mounted vertically.  1, 2, 3, ... is better.

ManfredBartz, 2022-01-23:
Just installed dev branch.  Issue with Nue light switch appears to be fixed, lights can be switched on/off.

closed 2022-01-23 by Koenkk