Koenkk/zigbee2mqtt #10938: Linkind water leak sensor unreliable pairing

ptr727, 2022-01-21:
### What happened?

I paired 8 Linkind water leak sensors.
5 of 8 failed to pair the first time with an interview failed error, repairing succeeds.

```
error 2022-01-19 13:03:54: Failed to configure '0x0c4314fffe7b3f7f', attempt 1 (Error: Bind 0x0c4314fffe7b3f7f/1 genPowerCfg from '0x00124b00258f2865/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))
```

### What did you expect to happen?

Pairing succeeds the first time

### How to reproduce it (minimal and precise)

Pair, good chance of failure

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Tubes CC2652P2 v2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7915885/log.txt)
[log1.txt](https://github.com/Koenkk/zigbee2mqtt/files/7915886/log1.txt)

closed 2022-03-01 by github-actions[bot]

airdrummingfool, 2022-10-19:
I'm having a similar issue with my Linkind LS21001 leak sensor - it pairs but the interview fails. @ptr727 did you get your devices to pair and work reliably?

ptr727, 2022-10-19:
I just repair them and eventually they pair and work fine, very solid after pairing.

airdrummingfool, 2022-10-21:
Thanks. After trying many times I returned it for a replacement, and the replacement paired and worked the first time. It might have been a bad unit, but for the record the one that wouldn't pair had date code `2111` and the one that did pair has date code `2104`.