Koenkk/zigbee2mqtt #10945: ZBWS01 not pairing

PM04290, 2022-01-22:
### What happened?

When pairing ZBWS01 (with dedicated button NET-KEY), led blinking normaly after few seconds, but nothing appear in Z2M, log is in debug mode and no lines displayed.

### What did you expect to happen?

pairing normally :)

### How to reproduce it (minimal and precise)

all times
i use this model from **Lonsonho**
https://zigbee.blakadder.com/Lerlink_ZBWS01.html
1 gang
https://fr.aliexpress.com/item/1005002235870518.html

### Zigbee2MQTT version

1.22.2-1

### Adapter firmware version

20211217

### Adapter

usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

nothing :(

PM04290, 2022-02-12:
@Koenkk I'm sorry to insist but I need to know if it's a possible Z2M problem or if the module is broken, in which case I have to have it replaced.

Koenkk, 2022-02-12:
Does it also fail to pair when holding it very close to the coordinator?

PM04290, 2022-02-27:
@Koenkk i have tryed at 5cm of coordinator, there wae absloutely no line on debug log.
A bought another scene switch : https://fr.aliexpress.com/item/1005002517960508.html
same problem, no lines on debug log.
can i do something  with a sniffer usb dongle to help you ? ha ve you an idae why there no lines on debug log. th two parts are zibbee 3.0.

Koenkk, 2022-02-28:
Do you maybe have an old cc2531 sniffer, if yes can you provide a sniff when trying to pair the device?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

PM04290, 2022-03-06:
@Koenkk , see attached file for 2 sniff, begin just after pairring authorized until led flashing end on devices.
[Sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/8192945/Sniff.zip)

Koenkk, 2022-03-06:
In Zigbee pairing is started by a device by sending a "Beacon request", I don't see that in your sniffs (should appear in the info column). This could mean 2 things:
- The device is not in pairing mode
- The device doesn't support all Zigbee channels, I suggest to also sniff on channel 11 and 15 when resetting the device and see if the device sends a "Beacon request".

Koenkk, 2022-03-10:
> Do you see any solution to solve my problem ?

Try to pair this device to the original gateway

PM04290, 2022-03-11:
it's a way, but i don't want to buy a dedicated hub just for a device, so much for that.

closed 2022-04-19 by github-actions[bot]