Koenkk/zigbee2mqtt #10946: [New device support]: BULB Tuya TS0505B - productor _TZ3210_remypqqm  not supported

mediaworld2007, 2022-01-22:
### Link

https://it.aliexpress.com/item/1005003667147607.html?gatewayAdapt=glo2ita&spm=a2g0o.order_list.0.0.21ef1802VJzrJi

### Database entry

{"id":41,"type":"Router","ieeeAddr":"0xa4c138a57fdc751c","nwkAddr":64144,"manufId":4417,"manufName":"_TZ3210_remypqqm","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_remypqqm","powerSource":1,"zclVersion":3,"appVersion":94,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":94,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1642849437480}

### Comments

I got 4 of this bulb TS0505B - _TZ3210_remypqqm but its seem not supported ( Stato supporto
Non supportato ) and I only see link quality status, not more features . 

I have tested 4 bulbs . 

Versione Zigbee2MQTT 1.21.1-dev commit: 5829663
Tipo Coordinator zStack3x0
Revisione Coordinator 20210708
Versione frontend 0.6.8
HA last version 

I checked it on web page and its seems supported https://www.zigbee2mqtt.io/devices/TS0505B.html   .. probably difefrent productors ?

Could you help me ?
Regards 
Luigi 

### Supported color modes

rgbcw

Koenkk, 2022-01-22:
Already supported in the latest dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-22 by Koenkk

mediaworld2007, 2022-01-23:
Hi Koenkk 

thanks for your support and your great work. I already use edge z2m version, and i read your guide, and now i have solved with an external converter 

It was my first time i had do it , it is not difficult . 

For others with my same issue : 

I created a file TS0505B.js and putted it in /share/zigbee2mqtt ( with filezilla ) , add it in congiguration file of zigbee2mqttedge  plugin

data_path: /share/zigbee2mqtt
external_converters:
  - TS0505B.js

file: TS0505B.js

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0505B'],
    model: 'TS0505B',
    vendor: '_TZ3210_remypqqm',
    description: 'LampRGBCW',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;

All work fine now. 

Thanks
Luigi