Koenkk/zigbee2mqtt #10947: Sonoff S26R2ZB leaves de network after ~30 seconds

jddcarreira, 2022-01-22:
### What happened?

Hi!

I have a Zigbee2MQTT running on Home Assistant with a [slae.sh CC2652RB](https://slae.sh/projects/cc2652/) coordinator and I've been try to connect the Sonoff plug S26R2ZB and after a few seconds (~30 seconds) the device leaves the network.

While the device is connected I'm able to properly control it, but after a few seconds it disconects.

One thing that I noticed is that the green led on the S26R2ZB is always blinking, even when connect to the network, which leads me to think that it might never leave the paring mode.

I have four brand new S26R2ZB and this is reproducible in all of them. 

I noticed that there is similar issue to this, but since the behaviour on[zigbee2mqtt#10282]( https://github.com/Koenkk/zigbee2mqtt/issues/10282) but I'm not sure if its related since my devices doesn't stick to mesh for more than 30 seconds.

Thanks

### What did you expect to happen?

The device would keep connected to the mesh.

### How to reproduce it (minimal and precise)

1. Turn on the device in paring mode
2. Allow to conenct
3. Wait for a successful pair
4. Wait a few seconds (~30 secs)
5. The device will self disconnect

### Zigbee2MQTT version

1.22.2-1

### Adapter firmware version

20211217

### Adapter

SONOFF S26R2ZB

### Debug log

Info Device 'Office :: Heater' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Office :: Heater","ieee_address":"0x00124b0024c0974a"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"Office :: Heater"},"type":"device_connected"}'
Info Starting interview of 'Office :: Heater'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Office :: Heater","ieee_address":"0x00124b0024c0974a","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"Office :: Heater"},"type":"pairing"}'
Debug Device 'Office :: Heater' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Office :: Heater","ieee_address":"0x00124b0024c0974a"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Office :: Heater"},"type":"device_announced"}'
Debug Received Zigbee message from 'Office :: Heater', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Office :: Heater', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Info Successfully interviewed 'Office :: Heater', device has successfully been paired
Info Device 'Office :: Heater' is supported, identified as: SONOFF Zigbee smart plug (S26R2ZB)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee smart plug","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"S26R2ZB","options":[],"supports_ota":false,"vendor":"SONOFF"},"friendly_name":"Office :: Heater","ieee_address":"0x00124b0024c0974a","status":"successful","supported":true},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Zigbee smart plug","friendly_name":"Office :: Heater","model":"S26R2ZB","supported":true,"vendor":"SONOFF"},"type":"pairing"}'
Debug Retrieving state of 'Office :: Heater' after reconnect
Debug Received Zigbee message from 'Office :: Heater', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Office :: Heater', payload '{"linkquality":36,"state":"OFF"}'
Warning Device 'Office :: Heater' left the network
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Office :: Heater","ieee_address":"0x00124b0024c0974a"},"type":"device_leave"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x00124b0024c0974a"},"type":"device_removed"}'
Info Device 'Office :: Heater' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Office :: Heater","ieee_address":"0x00124b0024c0974a"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"Office :: Heater"},"type":"device_connected"}'
Info Starting interview of 'Office :: Heater'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Office :: Heater","ieee_address":"0x00124b0024c0974a","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"Office :: Heater"},"type":"pairing"}'
Debug Device 'Office :: Heater' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Office :: Heater","ieee_address":"0x00124b0024c0974a"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Office :: Heater"},"type":"device_announced"}'
Debug Received Zigbee message from 'Office :: Heater', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Office :: Heater', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Info Successfully interviewed 'Office :: Heater', device has successfully been paired
Info Device 'Office :: Heater' is supported, identified as: SONOFF Zigbee smart plug (S26R2ZB)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee smart plug","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"S26R2ZB","options":[],"supports_ota":false,"vendor":"SONOFF"},"friendly_name":"Office :: Heater","ieee_address":"0x00124b0024c0974a","status":"successful","supported":true},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Zigbee smart plug","friendly_name":"Office :: Heater","model":"S26R2ZB","supported":true,"vendor":"SONOFF"},"type":"pairing"}'
Debug Retrieving state of 'Office :: Heater' after reconnect
Debug Received Zigbee message from 'Office :: Heater', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Office :: Heater', payload '{"linkquality":36,"state":"OFF"}'
Warning Device 'Office :: Heater' left the network

Koenkk, 2022-01-22:
Could you provide a sniff when pairing this device?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

jddcarreira, 2022-01-24:
Hi! 
Sorry for the delay, but had to buy the gear for the sniff.

Regarding the issue, I was able to fix it. 
My network was still in experimental state, so I still had the default Zigbee encryption key. Once I defined a proper new one, the devices worked like a charm.

By curiosity, is this kind of behaviour expected when using the default encryption key? Or does it worth to put has a note on the device page? https://www.zigbee2mqtt.io/devices/S26R2ZB.html

Thanks

Koenkk, 2022-01-25:
I don't expect that the issue was the default encryption key. Changing this key will reset your stick which so I guess that is what fixes this. Are you able to reproduce this with the default encryption key? If not, please close this issue.

Hedda, 2022-02-06:
FYI, there is more general discussions about non-specific performance issues with the ITead Sonoff S26R2ZB plugs here -> https://github.com/Koenkk/zigbee2mqtt/discussions/10909

Hedda, 2022-03-09:
FYI, Posted tip yesterday in https://github.com/Koenkk/zigbee2mqtt/issues/10282 that might be a good idea to try PTVO firmware but **select the CC2652R chip type (and not CC2652P) in the PTVO firmware configurator** as the  S26R2ZB radio board module has probably been optimised for use without a power amplifier enabled as per https://ptvo.info/custom-firmware-for-sonoff-zbmini-01minizb-and-ewelink-switch-zr-02-923/

closed 2022-04-17 by github-actions[bot]