Koenkk/zigbee2mqtt #10948: cc2531 as router generate warnings

Trzebu, 2022-01-22:
### What happened?

Hi.
I flash my cc2531 with router firmware and cc2351 connect to coordinator, end devices can connect to cc2351 but I can't see cc2351 in zigbee2mqtt frontend and it generates a lot of warnings like: 
Warning: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Warning: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
How can i fix this?

### What did you expect to happen?

CC2351 as router should be visible in frontend as "CC2530.ROUTER"

### Zigbee2MQTT version

1.21.2

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 2.0 usb dongle plus

lux73, 2022-01-22:
CC2531 as Router works for me - could not be a generally Problem:

[image]

Trzebu, 2022-01-22:
> CC2531 as Router works for me - could not be a generally Problem:
> 
> [image]

You have firmware from here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_Home_1.2/bin too?

lux73, 2022-01-22:
no, its flashed wit ptvo Router Firmware: https://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/

Router Firmware from 2018-09-26

Trzebu, 2022-01-23:
> no, its flashed wit ptvo Router Firmware: https://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/
> 
> Router Firmware from 2018-09-26

So it's OK to flash cc2531 with cc2530 firmware?

Trzebu, 2022-01-23:
Now I noticed that zigbee2mqtt think that cc2531 is coordinator. @lux73 have you firmware for cc2531 from 2018-09-26? Because I can't find this anywhere

lux73, 2022-01-23:
i bought my CC2531 pre-flashed as Router from here: https://www.ebay.de/itm/274300792162

closed 2022-03-03 by github-actions[bot]