Koenkk/zigbee2mqtt #10951: Aqara E1 door & window contact sensor [MCCGQ14LM] not working properly

Block4r, 2022-01-22:
### Link

https://www.zigbee2mqtt.io/devices/MCCGQ14LM.html#adapter-firmware

### Database entry

{"id":11,"type":"EndDevice","ieeeAddr":"0x54ef44100025955e","nwkAddr":11151,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.magnet.acn001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,64704],"outClusterList":[3,25],"clusters":{"ssIasZone":{"attributes":{"iasCieAddr":"0x0000000000000000","zoneState":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1642862408769,"defaultSendRequestWhen":"immediate"}

### Comments

Hi All!
Sensors are paring but i see only linkquality. 
I updated Conbee II firmware (according to https://www.zigbee2mqtt.io/devices/MCCGQ14LM.html#adapter-firmware) and trying few times paring and unpairing sensors - the same problem. 
[image]
[image]
[image]
Can you help me? or i should return sensors?

metlinux, 2022-01-24:
I have the exact same model and have no problem at all. I am using Zigbee2Mqtt 1.22.2 (not the dev but 1.22.2 commit: 414c51f)
It worked out the box for me:
[image]
[image]

Block4r, 2022-01-24:
It's Works! :) i Switched to Z2M without "dev", pair again and it worked. After that i Switched back to "Dev" and sensors working properly. Thank you.

closed 2022-01-24 by Block4r

ToXinE, 2022-02-11:
I have the same issue with 1.23.0 release.
[image]

 Does 1.22.2-dev have been merged ? Does 1.22.2-dev contains fixes that are not in 1.23.0 ?

[image]
[image]

metlinux, 2022-02-13:
In 1.23.0 it is still working for me. Have you upgraded the firmware of your adapter ? Maybe that is causing this.....

PS this issue was closed.....

ToXinE, 2022-02-13:
yep ! as you can see in screenshots,my coordinator is conbee2 with last firmware 0x26720700