Koenkk/zigbee2mqtt #10952: Incorrectly recognized tuya switch module model

drbios, 2022-01-22:
### What happened?

Some time ago I requested to add these modules to the database #9465. Today I reinstalled the system but the TS0011_switch_module module was recognized again as TS0011, it does not expose the switch type, or the initial state as it did before after the modification.

### What did you expect to happen?

TS0011 get recognize as TS0011_switch_module module

### How to reproduce it (minimal and precise)

just installing

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

egony cc2652p_cc1352p RF-STAR

drbios, 2022-01-22:
[image]
this is how the module is recognize

Koenkk, 2022-01-23:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/3630

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-23 by Koenkk