Koenkk/zigbee2mqtt #10953: Zigbee RGB Light missing extra feature

McGyver64, 2022-01-22:
### What happened?

After a crash and rebuild of RPI + Z2M + Domoticz etc. I was in process of pairing all devices again, but one device is missing functionality. I restored an pre-crash backup and it shows 2 devices, as after the crash I only get one device. I have both /domoticz/plugins data available

### What did you expect to happen?

That after repairing this Zigbee RGB light , the whole functionality was back

### How to reproduce it (minimal and precise)

when I restore an pra-crash database of Domoticz I get both devices. After new install en repairing devices, I only get one device

After new install and repairing

[image]
[image]
<img width="791" alt="Schermafbeelding 2022-01-22 om 15 26 25" src="https://user-

After restoring Pre-crash Database

images.githubusercontent.com/69874589/150643249-2e540290-21a7-4c0f-b0b1-68b96b5fb916.png">
[image]
 

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211217

### Adapter

CC2562RB

Koenkk, 2022-01-23:
Can you check if it works correctly via the z2m frontend? If yes, please create an issue here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues

McGyver64, 2022-01-23:
[image]

[image]

closed 2022-01-23 by McGyver64