Koenkk/zigbee2mqtt #10954: OWON CB432 does not report power and energy correctly

JJanssen123, 2022-01-22:
### What happened?

OWON CB432 power circuit breaker does not report power and energy; these values are always 0. State and linkquality are reported correctly.

{"energy":0,"linkquality":204,"power":0,"state":"ON"}

https://www.zigbee2mqtt.io/devices/CB432.html

### What did you expect to happen?

I expect the power and energy values to change when current is drawn through this circuit breaker.

### How to reproduce it (minimal and precise)

Switch on the circuit breaker, have some apparatus use energy. 

### Zigbee2MQTT version

1.22.2-dev commit: d5de1c62

### Adapter firmware version

20220103

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

```
# switch on from z2m frontend
Zigbee2MQTT:debug 2022-01-22 15:47:15: Received MQTT message on 'domotica/zigbee2mqtt/relais_autolader/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2022-01-22 15:47:15: Publishing 'set' 'state' to 'relais_autolader'
Zigbee2MQTT:info  2022-01-22 15:47:15: MQTT publish: topic 'domotica/zigbee2mqtt/relais_autolader', payload '{"energy":0,"linkquality":222,"power":0,"state":"ON"}'

# refresh state from z2m frontend
Zigbee2MQTT:debug 2022-01-22 15:47:48: Received MQTT message on 'domotica/zigbee2mqtt/relais_autolader/get' with data '{"state":""}'
Zigbee2MQTT:debug 2022-01-22 15:47:48: Publishing get 'get' 'state' to 'relais_autolader'
Zigbee2MQTT:debug 2022-01-22 15:47:48: Received Zigbee message from 'relais_autolader', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-01-22 15:47:48: MQTT publish: topic 'domotica/zigbee2mqtt/relais_autolader', payload '{"energy":0,"linkquality":185,"power":0,"state":"ON"}'

# refresh state from z2m frontend
Zigbee2MQTT:debug 2022-01-22 15:48:00: Received MQTT message on 'domotica/zigbee2mqtt/relais_autolader/get' with data '{"state":""}'
Zigbee2MQTT:debug 2022-01-22 15:48:00: Publishing get 'get' 'state' to 'relais_autolader'
Zigbee2MQTT:debug 2022-01-22 15:48:00: Received Zigbee message from 'relais_autolader', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-01-22 15:48:00: MQTT publish: topic 'domotica/zigbee2mqtt/relais_autolader', payload '{"energy":0,"linkquality":204,"power":0,"state":"ON"}'

# switch off from z2m frontend
Zigbee2MQTT:debug 2022-01-22 15:48:03: Received MQTT message on 'domotica/zigbee2mqtt/relais_autolader/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2022-01-22 15:48:03: Publishing 'set' 'state' to 'relais_autolader'
Zigbee2MQTT:info  2022-01-22 15:48:03: MQTT publish: topic 'domotica/zigbee2mqtt/relais_autolader', payload '{"energy":0,"linkquality":207,"power":0,"state":"OFF"}'

# refresh state from z2m frontend
Zigbee2MQTT:debug 2022-01-22 15:48:12: Received MQTT message on 'domotica/zigbee2mqtt/relais_autolader/get' with data '{"state":""}'
Zigbee2MQTT:debug 2022-01-22 15:48:12: Publishing get 'get' 'state' to 'relais_autolader'
Zigbee2MQTT:debug 2022-01-22 15:48:12: Received Zigbee message from 'relais_autolader', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-01-22 15:48:12: MQTT publish: topic 'domotica/zigbee2mqtt/relais_autolader', payload '{"energy":0,"linkquality":225,"power":0,"state":"OFF"}'
```

JJanssen123, 2022-01-22:
I did some sniffing using my old CC2531 stick. I'm not sure what one would like to see here. I tried reading an attribute using the z2m developer console but i don't know what attributes to read. This is reading activePower and also acFrequency (that should not be supported):

```
Frame 63: 57 bytes on wire (456 bits), 55 bytes captured (440 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x222f
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x222f
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 156
    Frame Control Field: Profile-wide (0x18)
    Sequence Number: 156
    Command: Read Attributes Response (0x01)
    Status Record, Int16: 0
        Attribute: Active Power (0x050b)
        Status: Success (0x00)
        Data Type: 16-Bit Signed Integer (0x29)
        Int16: 0
    Status Record
        Attribute: AC Frequency (0x0300)
        Status: Unsupported Attribute (0x86)
```

JJanssen123, 2022-01-22:
I also read currentSummDelivered and that is at least showing a value:

```
Frame 137: 58 bytes on wire (464 bits), 56 bytes captured (448 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x222f
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x222f
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 231
    Frame Control Field: Profile-wide (0x18)
    Sequence Number: 231
    Command: Read Attributes Response (0x01)
    Status Record, Uint: 2178
        Attribute: Current Summation Delivered (0x0000)
        Status: Success (0x00)
        Data Type: 48-Bit Unsigned Integer (0x25)
        Uint48: 2178 (0x0000000000000882)
```

JJanssen123, 2022-01-22:
After checking out what is in owon.js I used the developer console to read instantaneousDemand and currentSummDelivered. That works as both values are reported: {"energy": 4.18, "linkquality": 174, "power": 2, "state": "ON"}. However they are only updated after this manual request, not automatically.

JJanssen123, 2022-01-23:
I use the above now as a workaround: I have a cronjob running that sends the request to read the attributes every minute.

`mosquitto_pub -t zigbee2mqtt/relais_autolader/1/set -m '{"read":{"attributes":["instantaneousDemand","currentSummDelivered"],"cluster":"seMetering","options":{}}}'`

closed 2022-03-05 by github-actions[bot]

JJanssen123, 2022-03-06:
Sorry this is not solved (running 1.24.0-dev now) but I was too late to prevent closing it by the bot. Can this be reopened? And maybe someone (@khaidakin?) can take a look into the issue?

SveinHaDD, 2022-12-22:
Any news on this one? I had to upgrade zigbeee2mqtt to get some new HUE units to work (which they did) and then ALL of my OWON based energy logging units died... Now running z2m ver 1.28.4. The workaround from Jan 23 is working but not as a permanent solution...