Koenkk/zigbee2mqtt #10955: [New device support]: New type SilverCrest/LIDL Motion sensor HG07310

FlexxFR, 2022-01-22:
### Link

https://customercare.lidl.nl/SelfServiceNL/s/product2?productId=01t6N000001G6kfQAC#q=369229

### Database entry

{"id":130,"type":"EndDevice","ieeeAddr":"0x847127fffea6095a","nwkAddr":47161,"manufId":4619,"manufName":"_TZ1800_fcdjzz3s","powerSource":"Battery","modelId":"TY0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0021cbf157","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":31}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf157","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":67,"stackVersion":2,"hwVersion":1,"dateCode":"2021.1.5","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1642868242653,"useImplicitCheckin":true}

### Comments

This is a new type from SilverCrest/LIDL, model type according the box is HG07310.

However, after pairing it is recognized as 
https://www.zigbee2mqtt.io/devices/HG06335.html

Didn't work at first so I believed some work was needed for this new model, however after two more tries (??) it's actually working as normal :), So at the end no big issue.....

Could find any information on this forum about the model HG07310
In the end it's only the model type and picture that are wrong....for what it's worth :)

Koenkk, 2022-01-23:
Both use the same manufacturer name so we cannot distinguish them. I've changed the model to `HG06335/HG07310`

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-23 by Koenkk

Wuehli, 2024-10-01:
Hi,

i just started to migrate my devices over to z2m and a have this device. 
Unfortunately it doesn't work as expected. It get's recognized but after that the values for `Occupancy`, `Battery low` and `Tamper`
aren't updated and stay as "N/A", only `Battery `gets updated.
I tried two devices and re-paired them several times with no effect, in Deconz/Phoscon they are working fine.