Koenkk/zigbee2mqtt #10959: Null values battery, voltage and temp attributes after pairing Aqara RTCGQ11LM. Solved by downgrade to 1.22.0.

denevecon, 2022-01-22:
### What happened?

Pairing a new Aqara motion sensor RTCGQ11LM leaves the battery and voltage attributes empty (NULL). After downgrading to release 1.22.0, deleting the device and pairing presented the missing values.  I noticed this behavior in the z2m add-on in HA.  To exclude the HA interference I did a z2m install on a VM.

### What did you expect to happen?

That all exposed attributes would be present after pairing.

### How to reproduce it (minimal and precise)

Stand-alone install Ubuntu 20.04.02 release 1.22.2 
Pair device 
Notice the mqqt message : {"battery":null,"illuminance":4,"illuminance_lux":4,"linkquality":141,"occupancy":true,"temperature":null,"voltage":null}
Downgrade to 1.22.0 (git checkout tags/1.22.0)
npm start
pair device
mqtt message : {"battery":100,"illuminance":17,"illuminance_lux":17,"linkquality":111,"occupancy":true,"temperature":25,"voltage":3055}

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210708

### Adapter

zStack3x0

McGiverGim, 2022-01-24:
I had the same problem with latest version and the RTCGQ01LM. 
I upgraded to the edge version and repaired it and now it works perfectly. Can you check against the edge version?

denevecon, 2022-01-24:
I actually did that before downgrading to 1.22.0. With negative results.

kevincaradant, 2022-01-24:
I got the same issue with the water leak sensor Xiaomi. I fixed it with the force ... but it worked :D

1- 'force to remove the device' from the UI.
2 - remove the device line (0x) in the database.db file. 
==> Please be careful and do a backup before to modify it.

Then restart Z2M then push the button on the device 10s, a message will say something like 'device left the network' then re-click on the button and it will be re-paired again.

augard, 2022-01-27:
I have the exactly same problem

McGiverGim, 2022-01-28:
In the latest days I've received three devices. From Aqara (with luminance) and Xiaomi (without). I can confirm that:
- The problem exists. I have suffered it with the three devices.
- To fix it, simply repair it. You don't need to remove it. Enable pairing and press the button serveral seconds in the device.
- After being paired the second time the battery and the other values are ok.

@Koenkk do you have some idea about what has changed and where the problem can be? Is strange that the first time is only half working, and the second time it works perfectly.

Koenkk, 2022-01-28:
@McGiverGim after pairing it the second time, are you able to reproduce this by:
- force removing the device via the frontend
- stop z2m, start z2m
- pair device

kevincaradant, 2022-01-28:
> In the latest days I've received three devices. From Aqara (with luminance) and Xiaomi (without). I can confirm that:
>
> * The problem exists. I have suffered it with the three devices.
> * To fix it, [...]

If I said to remove it, it's because in my case ( not your apparently 😉) it didn't work without all steps mentioned before.
NB: If I don't remove the line from database, the problem was still here.
However, a complete repair + clean database file, has fixed the issue.

McGiverGim, 2022-01-28:
> * moving the device via the frontend
> * stop z2m, start z2m
> * pair device

Yes, I can (restarting z2m, if not it does not happen). Here is the log:
```
info  2022-01-28 20:11:32: Zigbee: allowing new devices to join.
info  2022-01-28 20:11:56: Device '0x00158d0007d762e0' joined
info  2022-01-28 20:11:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0007d762e0","ieee_address":"0x00158d0007d762e0"},"type":"device_joined"}'
info  2022-01-28 20:11:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d0007d762e0"},"type":"device_connected"}'
info  2022-01-28 20:11:56: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"last_seen":"2022-01-28T20:11:56+01:00","occupancy":false}'
info  2022-01-28 20:11:56: Starting interview of '0x00158d0007d762e0'
info  2022-01-28 20:11:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0007d762e0","ieee_address":"0x00158d0007d762e0","status":"started"},"type":"device_interview"}'
info  2022-01-28 20:11:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0007d762e0"},"type":"pairing"}'
debug 2022-01-28 20:11:57: Received Zigbee message from '0x00158d0007d762e0', type 'attributeReport', cluster 'genBasic', data '{"appVersion":5,"modelId":"lumi.sensor_motion.aq2"}' from endpoint 1 with groupID 0
debug 2022-01-28 20:11:57: Skipping message, definition is undefined and still interviewing
info  2022-01-28 20:11:57: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"last_seen":"2022-01-28T20:11:57+01:00","linkquality":61,"occupancy":false}'
debug 2022-01-28 20:11:57: Received Zigbee message from '0x00158d0007d762e0', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3115,"10":23908,"100":0,"11":33,"3":27,"4":424,"5":10,"6":[0,0]}}' from endpoint 1 with groupID 0
debug 2022-01-28 20:11:57: Skipping message, definition is undefined and still interviewing
info  2022-01-28 20:11:57: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"last_seen":"2022-01-28T20:11:57+01:00","linkquality":61,"occupancy":false}'
info  2022-01-28 20:11:57: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"last_seen":"2022-01-28T20:11:57+01:00","linkquality":61,"occupancy":false}'
debug 2022-01-28 20:11:57: Device '0x00158d0007d762e0' announced itself
info  2022-01-28 20:11:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0007d762e0","ieee_address":"0x00158d0007d762e0"},"type":"device_announce"}'
info  2022-01-28 20:11:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d0007d762e0"},"type":"device_announced"}'
info  2022-01-28 20:11:57: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"last_seen":"2022-01-28T20:11:57+01:00","linkquality":61,"occupancy":false}'
debug 2022-01-28 20:11:58: Received Zigbee message from '0x00158d0007d762e0', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.sensor_motion.aq2"}' from endpoint 1 with groupID 0
debug 2022-01-28 20:11:58: Skipping message, definition is undefined and still interviewing
info  2022-01-28 20:11:58: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"last_seen":"2022-01-28T20:11:58+01:00","linkquality":61,"occupancy":false}'
debug 2022-01-28 20:12:00: Received Zigbee message from 'temperatura_banyo_pasillo', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":2775,"10":30902,"100":1907,"101":4297,"4":17320,"5":36,"6":[0,2]}}' from endpoint 1 with groupID 0
info  2022-01-28 20:12:01: MQTT publish: topic 'zigbee2mqtt/temperatura_banyo_pasillo', payload '{"battery":23,"humidity":43,"last_seen":"2022-01-28T20:11:59+01:00","linkquality":123,"temperature":19.1,"voltage":2775}'
debug 2022-01-28 20:12:01: Received Zigbee message from '0x00158d0007d762e0', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.sensor_motion.aq2"}' from endpoint 1 with groupID 0
debug 2022-01-28 20:12:01: Skipping message, definition is undefined and still interviewing
info  2022-01-28 20:12:01: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"last_seen":"2022-01-28T20:12:01+01:00","linkquality":61,"occupancy":false}'
debug 2022-01-28 20:12:07: Received Zigbee message from '0x00158d0007d762e0', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":60}' from endpoint 1 with groupID 0
debug 2022-01-28 20:12:07: Skipping message, definition is undefined and still interviewing
info  2022-01-28 20:12:07: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"last_seen":"2022-01-28T20:12:07+01:00","linkquality":61,"occupancy":false}'
debug 2022-01-28 20:12:07: Received Zigbee message from '0x00158d0007d762e0', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
debug 2022-01-28 20:12:07: Skipping message, definition is undefined and still interviewing
info  2022-01-28 20:12:07: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"last_seen":"2022-01-28T20:12:07+01:00","linkquality":61,"occupancy":false}'
info  2022-01-28 20:12:11: Successfully interviewed '0x00158d0007d762e0', device has successfully been paired
[... cut ...]
info  2022-01-28 20:13:38: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"last_seen":"2022-01-28T20:13:38+01:00","linkquality":123,"occupancy":true,"temperature":27,"voltage":3115}'
debug 2022-01-28 20:13:40: Received Zigbee message from '0x00158d0007d762e0', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":3}' from endpoint 1 with groupID 0
info  2022-01-28 20:13:40: MQTT publish: topic 'zigbee2mqtt/0x00158d0007d762e0', payload '{"battery":100,"illuminance":3,"illuminance_lux":3,"last_seen":"2022-01-28T20:13:40+01:00","linkquality":112,"occupancy":true,"temperature":27,"voltage":3115}'
```

Koenkk, 2022-01-29:
@McGiverGim your logs looks like expected. Upon pairing z2m doesn't know the battery percentage yet, so it will publish `battery: null`. This is done to keep HA satisfied, and is only done when the HA integration is enabled.

McGiverGim, 2022-01-29:
@Koenkk My log contains "two" pairing. The first one shows the battery null as value always, after the second pairing is when it shows values. 
If I don't  make the second pairing, the battery remains null. Or at least by a long period of time.

Koenkk, 2022-01-29:
@McGiverGim 

> Or at least by a long period of time.

I would expect this, it will be null as long as the sensor doesn't report the battery (could take 1 day). However the pairing button also functions as a send battery report button (if I remember correctly). Since the device has already been paired, z2m receives the battery report and publishes it.  

So this might also work:
- pair device (1st time)
- press pair button once

McGiverGim, 2022-01-29:
@Koenkk Maybe you are right. I will do the test that you are suggesting. I can pair it and let it one or two days alone too. I will report back.

McGiverGim, 2022-01-29:
Tested the button after pairing, it did not work. I will let alone to see if it works...

Koenkk, 2022-01-29:
Strange, looks like some strange behaviour of the device then. Anyway I don't expect this is a z2m issue, it just waits until the device reports its battery state and publishes `null` in the meantime (only when the HA integration is enabled). After pairing it the 1st time and letting it sit for a day, is battery reported?

McGiverGim, 2022-01-29:
As the other user said, it seems that downgrading to 1.22.0 fixes the issue. I have not tested it so I can't confirm. I will let it alone for one day and report tomorrow.

McGiverGim, 2022-01-29:
@Koenkk you were right again. After some hours my values appeared alone. I don't know why repairing made they appear sooner.

denevecon, 2022-01-30:
After further research, I updated the FW on my coordinator to version 20211217. My coordinator is Sonoff ZigBee Dongle Plus. Together with the FW upgrade, I switched to the dev branch of z2m. The symptoms are now gone. I'll close the issue.

closed 2022-01-30 by denevecon