Koenkk/zigbee2mqtt #10960: Miboxer FUT105Z [New device support]:

soukal24, 2022-01-23:
### Link

https://miboxer.com/light/xqy/fut105z.html

### Database entry

{"id":31,"type":"Router","ieeeAddr":"0x540f57fffe19e6c8","nwkAddr":8782,"manufId":4098,"manufName":"_TZ3210_wslkvrau","powerSource":"Mains (single phase)","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterLi$

### Comments

Hi, I tried to create support for the new Miboxer FUT105Z bulb, but unfortunately I failed to get it up and running. Is there any way to add support for this light bulb? Thank you in advance!
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3210_wslkvrau'}],
    model: 'FUT105Z',
    vendor: 'Miboxer',
    description: 'Mi-Light RGB+CCT LED Bulb 1100lm 12W',
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;
```

### Supported color modes

color (RGB+CCT)

### Color temperature range

Do not know :/

soukal24, 2022-01-27:
Hi!
I found it. I added fingerprint '_TZ3210_wslkvrau' to TS0505B in node_modules/zigbee-herdsman-converters/devices/tuya.js.
Same I add fingerprint '_TZ3210_i680rtja' to TS0501B for: see picture.

I think it is worth to be added there for ever

[image]

Koenkk, 2022-01-27:
thanks, added both, will be included in the next release

closed 2022-01-27 by Koenkk