Koenkk/zigbee2mqtt #10963: TuYa WSD500A does not expose battery or voltage

oh2xx, 2022-01-23:
### What happened?

TuYa WSD500A does not expose battery or voltage. The device [document](https://www.zigbee2mqtt.io/devices/WSD500A.html) should be updated accordingly.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211210

### Adapter

Sonoff Zigbee 3.0

rwb196884, 2022-01-29:
I have experinced this and thought it was probably a bug. But apparently it _can't_ work?

oh2xx, 2022-01-30:
> 

Yes, it looks like those parameters are not exposed:
[image]

closed 2022-03-09 by github-actions[bot]

starox, 2022-04-01:
Same problem with z2m 1.24.0 commit: 7a2ddf2)