Koenkk/zigbee2mqtt #10966: [New device support]: Alchemy VIII  AL8TC13W-AP

ManfredBartz, 2022-01-23:
### Link

https://urneeds.com.au/13w-90mm-smart-zigbee-tri-colour-led-downlight-kit-for-echo-plus-smartthings-hub-hue/

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0x00124b0019d319a6","nwkAddr":18138,"manufId":0,"manufName":"ALCHEMY_VIII","powerSource":"Mains (single phase)","modelId":"AL8TC13W-AP","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":544,"inClusterList":[0,3,4,5,\ 6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"AL8TC13W-AP","manufacturerName":"ALCHEMY_VIII","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20170724","swBuildId":"1.0.3"}},"lightingColorCtrl":{"at\ tributes":{"colorCapabilities":16}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{}}},"binds":[{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c9ec49","endpointID":1}],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId"\ :13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20170724","swBuildId":"1.0.3","zclVersion":1,"interviewCompleted":false,"meta":{},"l\ astSeen":1642761656532,"useImplicitCheckin":true}

### Comments

I could not find the original manufacturer's website.  Supplied link was the best I could find.
Does not seem to respond to "colorTempPhysicalMin" or "...Max".
The separate driver for the light has model number "GW-ZLTH10W01" and it uses a Zigbee module with a "CC2530F256" on it.

[image]
[image]
[image]

### External converter

```shell
Sorry, I am completely new to this.
```

### Supported color modes

color temperature 2700K - 6500K, white, no RGB

ManfredBartz, 2022-02-12:
extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]})  provides basic functionality.

I still cannot determine who the original manufacturer is.  It probably is "TuYa" and has just been branded "Alchemy VIII".

Koenkk, 2022-02-13:
Added. 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ManfredBartz, 2022-02-18:
Thanks.
Confirming functionality.  Now just need a photo - is the one I supplied previously ok or should I supply one without the mounting clips?
[image]

Koenkk, 2022-02-18:
The picture will show up once it is included in a release (next on 1 March).

closed 2022-02-18 by Koenkk