Koenkk/zigbee2mqtt #10971: Xiaomi ZNJLBL01LM Stop command not working

invis-z, 2022-01-23:
### What happened?

My device came with fw with a build date of 11-24-2021. While the device is running, setting the state to stop does nothing and the device will continue to move. 

Based on #4585, this issue seems to be only affecting the ZNJLBL01LM but not other Xiaomi cover/curtain devices

The current position is not reported while it is running (nor the running status ever become true if controlled using zigbee). Currently, a way to get the position is to set a new location on the oppsite of the current one (make the device change direction), if the value is set toward the same direction (e.g. 50 while the device is moving from 0 to 100 but currently at 10), the value will be overwritten to the one being set (50 instead of 10 in the previous example). This means quickly setting the position to 0 and then 100 (or 100 and then 0) will get the current position and writing it again seems can stop the device. This approach is rather hacky but seems to work when I tried manually doing it. 
Using the button on the device does make the running state to true, but the position is still not updating. 

### What did you expect to happen?

The motor stops.

### How to reproduce it (minimal and precise)

While the device is running, set state to stop

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Vision CC2652 dongle

### Debug log

```
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"OPEN"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received Zigbee message from 'Bedroom Curtain', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":117440661,"presentValue":1}' from endpoint 1 with groupID 0
No converter available for 'ZNJLBL01LM' with cluster 'genMultistateOutput' and type 'attributeReport' and data '{"61440":117440661,"presentValue":1}'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"STOP"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"CLOSE"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/Bedroom Curtain/set' with data '{"state":"CLOSE"}'
Publishing 'set' 'state' to 'Bedroom Curtain'
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Bedroom Curtain', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":117440689,"presentValue":1}' from endpoint 1 with groupID 0
No converter available for 'ZNJLBL01LM' with cluster 'genMultistateOutput' and type 'attributeReport' and data '{"61440":117440689,"presentValue":1}'
Received Zigbee message from 'Bedroom Curtain', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Bedroom Curtain', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":56791040,"presentValue":0}' from endpoint 1 with groupID 0
No converter available for 'ZNJLBL01LM' with cluster 'genMultistateOutput' and type 'attributeReport' and data '{"61440":56791040,"presentValue":0}'
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":8}' from endpoint 1 with groupID 0
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":8}' from endpoint 1 with groupID 0
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":8}' from endpoint 1 with groupID 0
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":8}' from endpoint 1 with groupID 0
Received Zigbee message from 'Bedroom Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":8}' from endpoint 1 with groupID 0
```

redphx, 2022-01-26:
Same for me after upgrading roller's firmware from version 25 to 27.

invis-z, 2022-01-30:
Fixed in Koenkk/zigbee-herdsman-converters#3757

closed 2022-01-30 by invis-z