Koenkk/zigbee2mqtt #10972: [Problem]: Legrand 067773 doesn't properly send off event

Crocmagnon, 2022-01-23:
### What happened?

I have a bunch of Legrand 067773 around my house. Most are linked to their light bulb with a zigbee group and work fine.
For some other, I'm using more advanced automations and I'm using the button press events. They are *also* linked to a bulb with a zigbee group to provide a degraded mode in case my Home Assistant is unavailable for some reason.

Everything worked fine, until a few weeks ago.

Now these buttons start acting up: they don't trigger the "advanced" action they're supposed to on the first press, I have to press a second time. The bulb switches off on the first try though, so the button "worked".

I tried measuring it by subscribing to the action MQTT endpoint but when I'm subscribed, it always works on the first try.

I already tried repairing the faulty units but it didn't change anything.

When this happens, here's what I observe in the logs:

<details>
<summary>Logs</summary>

```
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Interrupteur toilettes","ieee_address":"0x0004740000ad3839"},"type":"device_announce"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Interrupteur toilettes"},"type":"device_announced"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"dateCode":"0x26720700","friendly_name":"Coordinator","ieeeAddr":"0x00212effff059d1f","lastSeen":1642951214989,"networkAddress":0,"softwareBuildID":"ConBee2/RaspBee2","type":"Coordinator"},{"dateCode":"20190219","description":"Hue motion sensor","friendly_name":"Détecteur toilettes","hardwareVersion":1,"ieeeAddr":"0x001788010862e69e","lastSeen":1642951211496,"manufacturerID":4107,"manufacturerName":"Philips","model":"9290012607","modelID":"SML001","networkAddress":48011,"powerSource":"Battery","softwareBuildID":"6.1.1.27575","type":"EndDevice","vendor":"Philips"},{"dateCode":"20211021","description":"Hue white ambiance E27 with Bluetooth","friendly_name":"Ampoule toilettes","hardwareVersion":2,"ieeeAddr":"0x0017880108342537","lastSeen":1642951211920,"manufacturerID":4107,"manufacturerName":"Philips","model":"9290022169","modelID":"LTA001","networkAddress":38711,"powerSource":"Mains (single phase)","softwareBuildID":"1.93.3","type":"Router","vendor":"Philips"},{"dateCode":"20210525","description":"Hue white and color ambiance E26/E27","friendly_name":"Ampoule bureau","hardwareVersion":5,"ieeeAddr":"0x001788010679e743","lastSeen":1642951180059,"manufacturerID":4107,"manufacturerName":"Philips","model":"9290022166","modelID":"LCA001","networkAddress":64519,"powerSource":"Mains (single phase)","softwareBuildID":"1.90.1","type":"Router","vendor":"Philips"},{"dateCode":"20210525","description":"Hue white and color ambiance E26/E27","friendly_name":"Ampoule chambre","hardwareVersion":5,"ieeeAddr":"0x0017880106a07f9f","lastSeen":1642949635418,"manufacturerID":4107,"manufacturerName":"Philips","model":"9290022166","modelID":"LCA001","networkAddress":40583,"powerSource":"Mains (single phase)","softwareBuildID":"1.90.1","type":"Router","vendor":"Philips"},{"dateCode":" \u000020191030\u0000\u0000\u0000\u0000\u0000","description":"Wireless remote switch","friendly_name":"Interrupteur toilettes","hardwareVersion":1,"ieeeAddr":"0x0004740000ad3839","lastSeen":1642951214956,"manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067773","modelID":" Remote switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","networkAddress":64387,"powerSource":"Battery","softwareBuildID":"001e","type":"EndDevice","vendor":"Legrand"},{"dateCode":" \u000020191030\u0000\u0000\u0000\u0000\u0000","description":"Wireless remote switch","friendly_name":"Interrupteur bureau","hardwareVersion":1,"ieeeAddr":"0x0004740000ad3838","lastSeen":1642951179392,"manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067773","modelID":" Remote switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","networkAddress":21872,"powerSource":"Battery","softwareBuildID":"001e","type":"EndDevice","vendor":"Legrand"},{"dateCode":" \u000020191030\u0000\u0000\u0000\u0000\u0000","description":"Wireless remote switch","friendly_name":"Interrupteur chambre","hardwareVersion":1,"ieeeAddr":"0x0004740000ad383c","lastSeen":1642929644986,"manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067773","modelID":" Remote switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","networkAddress":3066,"powerSource":"Battery","softwareBuildID":"001e","type":"EndDevice","vendor":"Legrand"},{"dateCode":"20210525","description":"Hue white and color ambiance E26/E27","friendly_name":"Lampadaire bibliothèque","hardwareVersion":5,"ieeeAddr":"0x0017880108340f23","lastSeen":1642949485377,"manufacturerID":4107,"manufacturerName":"Philips","model":"9290022166","modelID":"LCA001","networkAddress":35896,"powerSource":"Mains (single phase)","softwareBuildID":"1.90.1","type":"Router","vendor":"Philips"},{"dateCode":"20210422","description":"Hue white and color ambiance LightStrip plus","friendly_name":"Bandeau LED bibliothèque","hardwareVersion":0,"ieeeAddr":"0x0017880108b4bff1","lastSeen":1642950833569,"manufacturerID":4107,"manufacturerName":"Philips","model":"8718699703424","modelID":"LCL001","networkAddress":33777,"powerSource":"Mains (single phase)","softwareBuildID":"1.88.2","type":"Router","vendor":"Philips"},{"dateCode":"20210331","description":"Hue Go with Bluetooth","friendly_name":"Bol salon","hardwareVersion":0,"ieeeAddr":"0x0017880106f9ffbb","lastSeen":1642950051246,"manufacturerID":4107,"manufacturerName":"Philips","model":"7602031P7","modelID":"LCT026","networkAddress":32441,"powerSource":"Mains (single phase)","softwareBuildID":"1.88.1","type":"Router","vendor":"Philips"},{"dateCode":"20210422","description":"Hue white and color ambiance LightStrip plus","friendly_name":"Bandeau LED salon","hardwareVersion":0,"ieeeAddr":"0x0017880108d5664e","lastSeen":1642951205519,"manufacturerID":4107,"manufacturerName":"Philips","model":"8718699703424","modelID":"LCL001","networkAddress":51232,"powerSource":"Mains (single phase)","softwareBuildID":"1.88.2","type":"Router","vendor":"Philips"},{"dateCode":"20210422","description":"Hue white and color ambiance LightStrip outdoor 2m","friendly_name":"Bandeau LED extérieur","hardwareVersion":1,"ieeeAddr":"0x0017880108c47bb3","lastSeen":1642950826056,"manufacturerID":4107,"manufacturerName":"Philips","model":"9290022890","modelID":"LCL002","networkAddress":45188,"powerSource":"Mains (single phase)","softwareBuildID":"1.88.2","type":"Router","vendor":"Philips"},{"dateCode":"20210525","description":"Hue white ambiance E27 with Bluetooth","friendly_name":"Ampoule couloir nord","hardwareVersion":2,"ieeeAddr":"0x00178801083a42a7","lastSeen":1642951212968,"manufacturerID":4107,"manufacturerName":"Philips","model":"9290022169","modelID":"LTA001","networkAddress":32338,"powerSource":"Mains (single phase)","softwareBuildID":"1.90.1","type":"Router","vendor":"Philips"},{"dateCode":"20210525","description":"Hue white ambiance E27 with Bluetooth","friendly_name":"Ampoule couloir est","hardwareVersion":2,"ieeeAddr":"0x00178801082d13f3","lastSeen":1642949529712,"manufacturerID":4107,"manufacturerName":"Philips","model":"9290022169","modelID":"LTA001","networkAddress":39224,"powerSource":"Mains (single phase)","softwareBuildID":"1.90.1","type":"Router","vendor":"Philips"},{"dateCode":"20210525","description":"Hue white ambiance E27 with Bluetooth","friendly_name":"Ampoule couloir sud","hardwareVersion":2,"ieeeAddr":"0x00178801083a48d7","lastSeen":1642949876188,"manufacturerID":4107,"manufacturerName":"Philips","model":"9290022169","modelID":"LTA001","networkAddress":407,"powerSource":"Mains (single phase)","softwareBuildID":"1.90.1","type":"Router","vendor":"Philips"},{"dateCode":"20210525","description":"Hue white and color ambiance E26/E27","friendly_name":"Ampoule bibliothèque","hardwareVersion":5,"ie [... cut ...]
debug 2022-01-23 16:32:16: Received Zigbee message from 'Détecteur toilettes', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID null
info  2022-01-23 16:32:16: MQTT publish: topic 'zigbee2mqtt/Détecteur toilettes', payload '{"battery":100,"illuminance":20560,"illuminance_lux":114,"led_indication":false,"linkquality":255,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"temperature":19.53,"update":{"state":"idle"},"update_available":false}'
```
</details>

rdorys, 2022-01-23:
@Crocmagnon : This exact issue is described here : https://github.com/Koenkk/zigbee2mqtt/issues/10862

It also appears on 067774 (the double switch). And I also get it on the 067773.
What is your firmware version of 067773 ? Mine is 
[image]

I only get "toggle" action on single clic, on "off" action on double clic. But no "on" action.

Koenkk, 2022-01-23:
Lets continue in #10862

closed 2022-01-23 by Koenkk