Koenkk/zigbee2mqtt #10973: Sonoff SNZB-01

hijamie32, 2022-01-23:
### What happened?

All my Sonoff SNZB-01 fail to send any data after pairing.

I have confirmed with the supplier of the Smartlight zigbee lan CC2652P they also are having the same issue.

Pairs fine and recognises it as the correct switch but then get this error during configuration:

Failed to configure '0x00124b002135b2c0', attempt 1 (Error: Bind 0x00124b002135b2c0/1 genOnOff from '0x00124b0014dbbe5d/0' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))

Many Thanks

### What did you expect to happen?

To pair and work correctly. 

### How to reproduce it (minimal and precise)

Remove and try to repair with new CC2652P coordinator.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210319

### Adapter

Smartlight zigbee lan CC2652P

Bumblebeee19, 2022-01-23:
Hello,

I had a similar problem with the SNZB-02. I found the solution on a forum (but I can't remember which one), after pairing, you have to press the reconfigure button on Z2M 
[image]
 while pressing the SNZB button very regularly (1x per second).

 This operation must be repeated until there are no more error messages.

Hope this works for you.

hijamie32, 2022-01-23:
Hey,

Thanks for your suggestion, sadly just tried this with all three and still no response.

Strange pairs straight away but then is like it instantly drops from the network again......

Cheers

s4ntz, 2022-01-31:
> 

OMG! you fixed my problems!!!
thank you! so obvious.

enigmaspb, 2022-03-10:
Same problem - reconfigure and  fast click  no helped

Heisenberg2980, 2022-03-16:
@Koenkk it looks like some update has broken the integration with these 3 Sonoff devices and they cannot be paired anymore: SNZB-01, SNZB-02 and SNZB-03. According to https://github.com/Koenkk/zigbee2mqtt/discussions/9216 the issue might started in October 2021, can you help with this? 

I can confirm the issue is happening with my SNZB-02. I haven´t been able to pair it no matter how many times I have tried while pressing the button in the device every second. I also try to hit reconfigure (as mentioned above) with no success.

I have capture sniffer logs while pairing, happy to share them if they help.

Koenkk, 2022-03-17:
@Heisenberg2980 could you provide a sniff when pairing it to z2m 1.20.0 and 1.24.0, provide the pcapng files here.

Heisenberg2980, 2022-03-17:
> @Heisenberg2980 could you provide a sniff when pairing it to z2m 1.20.0 and 1.24.0, provide the pcapng files here.

Sure, is there a way to downgrade z2m to an earlier version in Home Assistant OS? I have done that with integrations by downloading the repo to the custom folder, not sure if something similar can be done with addons.

Heisenberg2980, 2022-03-17:
@Koenkk I setup mosquitto and z2m on another pi running Debian version 11 (bullseye), and the first time I tried to pair the device using the latest version of z2m (1.24.0) it actually paired successfully, which was unexpected tbh. After that I tried to pair it again and it failed every time I tried.

I am not sure why it worked the first time (maybe for some reason the fresh installed made the pairing to succeed), but I got the sniff of both executions:
[Sniffer in Rpi 4.zip](https://github.com/Koenkk/zigbee2mqtt/files/8292965/Sniffer.in.Rpi.4.zip)
 
This is a screenshot of the log when the pairing fails:

[image]

Heisenberg2980, 2022-03-17:
I just figured out how to run previous versions of z2m, so this is the sniff of a successful pair using z2m 1.20.0:

[SNZB-02 paired successful - z2m_1.20.0.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/8294701/SNZB-02.paired.successful.-.z2m_1.20.0.pcapng.zip)

With this version 1.20.0 the pairing is successful every single time I try.

Koenkk, 2022-03-18:
@Heisenberg2980 interesting, could you check wether the device always pairs successfully with z2m 1.24.0 when:
- force remove the sonoff device via the frontend
- restart zigbee2mqtt
- start zigbee2mqtt, pair the device

If it does not pair successfully, could you provide the sniff again + herdsman debug log of this? Note that the restart is crucial here, between z2m 1.20.0 and 1.24.0 there were some changes to the interviewing process. When a device did already join before, z2m will not fully interview it anymore which may explain the issues.

Heisenberg2980, 2022-03-18:
@Koenkk you are right!! Following those steps (force-remove device, restart z2m, pair) the device pair successfully on z2m 1.24.0 every time (I have tested it twice), so that must be the issue.

I guess this half-interview if the device has already been paired before was implemented to improve performance in the pairing process, but if it is making some devices to fail the interview (not sure if it is happening just with these sonoff or with other devices as well), maybe it is better to do the full interview? or is it something can be enable/disable by device?

Koenkk, 2022-03-19:
@Heisenberg2980 thanks for confirming. Think I've found the issue, could you check wether it is fixed in the latest dev?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Heisenberg2980, 2022-03-19:
Great job @Koenkk, I can confirm issue is fixed in the dev branch.

@hijamie32 please test it using the dev branch (alternatively you can wait until the changes are applied to the main branch) and if it works you can close this issue.

snowyu, 2022-04-01:
I've tried the dev branch ad24c4f677f4d968575fc7bbb357c170a69b1421. But

1. SNZB-01 can not unbind the default one: `Failed to unbind cluster 'genOnOff' from '0xXXXXXX' to 'Coordinator' (Error: Unbind 0xXXXXX/1 genOnOff from '0xXXXXXX/1' failed (AREQ - ZDO - unbindRsp after 10000ms))`
2. SNZB-01 can not bind new one: `Failed to unbind cluster 'genOnOff' from '0xXXXXXX' to 'Coordinator' (Error: Unbind 0xXXXXX/1 genOnOff from '0xXXXXXX/1' failed (AREQ - ZDO - unbindRsp after 10000ms))`

Heisenberg2980, 2022-04-01:
@snowyu have you tried to remove the device using "force remove" and pair it again?

snowyu, 2022-04-01:
Yes, I've tried "force remove". Still same error.

Heisenberg2980, 2022-04-01:
> I've tried the dev branch [ad24c4f](https://github.com/Koenkk/zigbee2mqtt/commit/ad24c4f677f4d968575fc7bbb357c170a69b1421). But
> 
> 1. SNZB-01 can not unbind the default one: `Failed to unbind cluster 'genOnOff' from '0xXXXXXX' to 'Coordinator' (Error: Unbind 0xXXXXX/1 genOnOff from '0xXXXXXX/1' failed (AREQ - ZDO - unbindRsp after 10000ms))`
> 2. SNZB-01 can not bind new one: `Failed to unbind cluster 'genOnOff' from '0xXXXXXX' to 'Coordinator' (Error: Unbind 0xXXXXX/1 genOnOff from '0xXXXXXX/1' failed (AREQ - ZDO - unbindRsp after 10000ms))`

Both error messages say "Failed to unbind", can you try again ensuring the device has been removed from z2m and then try to pair it again? If you still get an error please share the error message.

snowyu, 2022-04-02:
Sorry, maybe my typo. Anyway, I've cleaned all devices. and try again:

First, pair The SNZB-01 button OK, then unbind the default:

`Failed to unbind cluster 'genOnOff' from '0xXXXXX' to 'Coordinator' (Error: Unbind 0xXXXXX/1 genOnOff from '0xXXXX/242' failed (AREQ - ZDO - unbindRsp after 10000ms))`

Then add another switch to test bind:

`Failed to bind cluster 'genOnOff' from '0xXXXXXX' to '0xXXXXX' (Error: Bind 0xXXXXXX/1 genOnOff from '0xXXXXXX/1' failed (AREQ - ZDO - bindRsp after 10000ms))`

cat2devnull, 2022-04-04:
This error has been an issue for me with multiple Sonoff devices but I just upgraded to Zigbee2MQTT
1.25.0 commit: 6f1460e and also upgraded my Sonoff USB Zigbee 3.0 Coordinator to zStack3x0 revision 20220219 and for the first time ever, every device is connected and working.

snowyu, 2022-05-07:
Did someone try to bind or unbind it working fine?

closed 2022-06-15 by github-actions[bot]

avviano, 2022-10-30:
It does not really matter if you use Zigbee2MQTT or ZHA integration, I tried both with IKEA and SonOFF buttons, they work a few days, if you are lucky a few weeks. Then one or two buttons stop working. Since this keeps recurring, I simply gave up on the button and only use NFC tags. It's the only thing I was able to get to works realiably so far. All my WLAN devices work great. Zigbee is terrible. The worst of all my devices, as they are totally unreliable. At least in my experience. I didn't count, but I must have reconfigured all my Zigbee devices a gazillion times. They work a week or two, then just stop working.

rafaelreis-r, 2023-01-16:
Hey @Koenkk this is still going on with the latest version of Z2M. I had a button work for years and lately it went kaplooft after updating HA and Z2M. 

Let me know if you need any pcap sniffs, I have all the stuff I used to try and get the A100 working at #13087

vaderag, 2024-09-30:
This issue is still going in 2024... Having the same problem with the switch. Was connected, but needed to reconnect and after removal it's not connecting again with issues above

Bmiest, 2025-01-10:
Had this issue, reconnected after force remove, seems to work for the time being.