Koenkk/zigbee2mqtt #10978: TS011F_plug_3 won't report energy, current or power

marcgarciamarti, 2022-01-23:
### What happened?

I purchased the device TS011F_plug_3 which is supported and it won't report some variables like current, energy or power

### What did you expect to happen?

I would expect for the plug to report the values above

### How to reproduce it (minimal and precise)

just put any load on the plug and it won't report any of the values above

### Zigbee2MQTT version

1.22.1-1

### Adapter firmware version

20210120

### Adapter

Sonoff

aIexus, 2022-01-25:
try this
https://github.com/Koenkk/zigbee2mqtt/issues/10889

marcgarciamarti, 2022-01-31:
> try this
> 
> https://github.com/Koenkk/zigbee2mqtt/issues/10889

I re-joined the device without first removing it, and it seems to be working! thanks for the suggestion!