Koenkk/zigbee2mqtt #10981: Device Availability works but not showing in State

Papi77, 2022-01-23:
### What happened?

When **Availability** & **Last seen** are enabled they both show-up in the dashboard as expected.
However, when viewing the State of the device - only **Last seen** is included - **availability** is included in the message.

This likely causes MQTT broker on HA not to discover the **Availability** state of the device.

[image]

### What did you expect to happen?

To see "Availability" in the parameters of the state and to see it auto-discovered in HA

### How to reproduce it (minimal and precise)

Enable availability: true -> view that it is shown in Z2M dashboard (column Availability will display Online or Offline) -> but no availability entity is created in HA

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210708

### Adapter

cc2652p2lan

sjorge, 2022-01-23:
Availability is published under a different topic, it is published to `zigbee2mqtt/<device>/availability` with a payload of `offline` or `online`.

Papi77, 2022-01-24:
So, no ability to expose it as a device entity in HA? I think it would be helpful if we also get it configured as an entity….to possibility easily display this information in Lovelace, use in automations etc. - just like we can do with “last seen”

Koenkk, 2022-01-24:
> So, no ability to expose it as a device entity in HA?

It is not exposed separately indeed, all entities of the device will be marked as `unavailable` when it is unavailable, so you can pick any entity of the device.

closed 2022-01-24 by Koenkk