Koenkk/zigbee2mqtt #10982: Danfoss TRV External Sensor not showing correct Temperature

rubenkr, 2022-01-23:
### What happened?

- Danfoss eTRV0100 paired seamlessly to Z2M (awesome thank you sooo much for this project!)
- According to [Z2M Doc](https://www.zigbee2mqtt.io/devices/014G2461.html) i update _External_measured_room_sensor_ every 15Minutes
- Z2M config for TRV is updated external_measured_room_sensor (see below) and radiator_covered = TRUE
-  External Sensor is BLE Xiaomi updating every 10Minutes
- Node-Red is updating the entity with this [flow](<script src="https://gist.github.com/rubenkr/49493e864d68ef10aaf61a2b55615ed4.js"></script>)
- Home-Assistant climate entity generated displaying the wrong temperature of the valve (see  [here](https://imgur.com/mnuENM9)

### What did you expect to happen?

- If external_measured_room_sensor = -8000 AND radiator_covered = TRUE --> Publish external_measured_room_sensor as heater temperature
- Auto Generated MQTT Climate in Home-Assistant is displaying external temperature sensorr

### How to reproduce it (minimal and precise)

see above 
1. publish external sensor via mqtt to external_measured_room_sensor 
2. set radiator_covered = TRUE
3. generated mqtt climate in home-assistant displaying 26C° despite room temperature is 22°

AGAIN: Thank you so much for consideration and ideas how to fix this. Maybe also to rewrite in home-assistant as first workaround?

### Zigbee2MQTT version

1.22.2-dev commit: d5de1c6

### Adapter firmware version

20211217 

### Adapter

HW: zStack3x0 CC2652RB

sjorge, 2022-01-24:
I think this is by design, the TRV keeps publishing the local_temperature it is sensing, the internal PID is just not using the value. 

And the climate device in HA just fetches the temperature from that attribute.

rubenkr, 2022-01-25:
@sjorge thank you for clarifying. But in the later control function it looks weird and hard to justify why the local temperature is 29C and the target temperature is 20.5C. See example here 
[image]

So you suggest to fix it in home-assistant, right?

sjorge, 2022-01-25:
If possible I'd try swapping the temperature sensor used for the climate device to the external one. Although I don't know exactly how to do that.

closed 2022-03-04 by github-actions[bot]