Koenkk/zigbee2mqtt #10988: [New device support]: Schneider Electric / ELKO SmartPir with switch 10A Plus SO

aravndal, 2022-01-23:
### Link

https://proff.elko.no/smartpir-m-bryter-10a/smartpir-m-bryter-10a-plus-so-article3216-3066.html

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0x680ae2fffe11833b","nwkAddr":53555,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"NHMOTION/SWITCH/1","epList":[1,37,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"dateCode":"","zclVersion":3,"hwVersion":1,"manufacturerName":"Schneider Electric","modelId":"NHMOTION/SWITCH/1","appVersion":2}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"genOnOffSwitchCfg":{"attributes":{}},"genAlarms":{"attributes":{}},"genPowerCfg":{"attributes":{}},"haDiagnostic":{"attributes":{"lastMessageLqi":255,"lastMessageRssi":0}},"genIdentify":{"attributes":{"identifyTime":0}},"genGroups":{"attributes":{"nameSupport":0}},"genScenes":{"attributes":{"count":0,"currentScene":0}},"genLevelCtrl":{"attributes":{}},"haElectricalMeasurement":{"attributes":{}},"msOccupancySensing":{"attributes":{}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1},{"cluster":2821,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1}],"configuredReportings":[],"meta":{}},"37":{"profId":260,"epId":37,"devId":2080,"inClusterList":[0,3,1024,1030,2821,65305],"outClusterList":[3,6,8,1030],"clusters":{"genBasic":{"attributes":{"dateCode":"","powerSource":1,"zclVersion":3}},"msOccupancySensing":{"attributes":{"occupancy":0,"occupancySensorType":0,"pirOToUDelay":350}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":12553}},"elkoOccupancySettingClusterServer":{"attributes":{}},"genOnOff":{"attributes":{}},"msIlluminanceLevelSensing":{"attributes":{}},"genLevelCtrl":{"attributes":{}},"genIdentify":{"attributes":{"identifyTime":0}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1},{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1},{"cluster":65305,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1},{"cluster":1030,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00212effff069298","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.005.024 R","zclVersion":3,"interviewCompleted":true,"meta":{"configured":469669313},"lastSeen":1642971122866,"useImplicitCheckin":true}

### Comments

Hi, I've created support for Schneider Electric branded ELKO Smart PIR Switch.
It's possible to change occupancy_timeout through Dev Console, but I did not manage to get that working in the converter file.
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes;
const ep = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['NHMOTION/SWITCH/1'],
    model: 'NHMOTION',
    vendor: 'Schneider Electric',
    description: 'SmartPir med dimmer Plus SO',
    whiteLabel: [{vendor: 'ELKO', model: 'EKO06991'}],
    fromZigbee: [fz.on_off, fz.illuminance, fz.occupancy, fz.occupancy_timeout],
    toZigbee: [tz.on_off, tz.occupancy_timeout],
    exposes: [ep.switch(), ep.occupancy(), ep.illuminance_lux(), ep.illuminance(),
              e.numeric('occupancy_timeout', ea.ALL).withValueMin(0).withValueMax(5000).withUnit('s')
             .withDescription('Time in seconds after which occupancy is cleared after detecting it')
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        // Endporint 1
        const endpoint1 = device.getEndpoint(1);
        const binds1 = ['genBasic','genIdentify','haDiagnostic','genOnOff'];
        await reporting.bind(endpoint1, coordinatorEndpoint, binds1);
        // Endporint 37
        const endpoint37 = device.getEndpoint(37);
        const binds37 = ['genBasic', 'genIdentify', 'msIlluminanceMeasurement', 'elkoOccupancySettingClusterServer',
                'msOccupancySensing'];
        await reporting.bind(endpoint37, coordinatorEndpoint, binds37);
    },
};

module.exports = definition;
```

briis, 2022-01-26:
I have a similar component, just the Danish version of it, mine is a Relay and not a Dimmer (it does exist in a Dimmer version also). I hacked together the below which works for me, also setting the `occupancy_timeout`. As I never did any of these before, I am not sure this is the right way to do it, but it does work. Maybe you can use that for inspiration?

[image]

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['NHMOTION/SWITCH/1'],
    model: '545D6306',
    vendor: 'Schneider Electric',
    description: 'LK FUGA Wiser wireless PIR with Relay',
    fromZigbee: [fz.on_off, fz.illuminance, fz.occupancy, fz.occupancy_timeout],
    exposes: [e.switch().withEndpoint('l1'), e.occupancy(), e.illuminance_lux(), e.illuminance(),
        exposes.numeric('occupancy_timeout', ea.ALL).withUnit('second').withValueMin(0).withValueMax(3600).withDescription('Time in seconds after which occupancy is cleared after detecting it')
    ],
    toZigbee: [tz.on_off, tz.occupancy_timeout],
    endpoint: (device) => {
        return { 'default': 37, 'l1': 1, 'l2': 37 };
    },
    meta: { multiEndpoint: true },
    configure: async(device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        const binds1 = ['genBasic', 'genIdentify', 'genOnOff'];
        await reporting.bind(endpoint1, coordinatorEndpoint, binds1);
        await reporting.onOff(endpoint1);
        // read switch state
        await endpoint1.read('genOnOff', ['onOff']);

        const endpoint37 = device.getEndpoint(37);
        const binds37 = ['msIlluminanceMeasurement', 'msOccupancySensing'];
        await reporting.bind(endpoint37, coordinatorEndpoint, binds37);
        await reporting.occupancy(endpoint37);
        await reporting.illuminance(endpoint37);
        // read occupancy_timeout
        await endpoint37.read('msOccupancySensing', ['pirOToUDelay']);
    },
};

module.exports = definition;
````

aravndal, 2022-01-28:
> I have a similar component, just the Danish version of it, mine is a Relay and not a Dimmer (it does exist in a Dimmer version also). I hacked together the below which works for me, also setting the [...]

Thank's this worked perfect on the ELKO switch too 👍

briis, 2022-01-28:
Great to hear. The above is already merged in the dev branch, so it should show up in one of the next releases.

Koenkk, 2022-01-29:
Will be included in the 1 February release. Assuming this can be closed.

closed 2022-01-29 by Koenkk

aravndal, 2022-01-29:
> Will be included in the 1 February release. Assuming this can be closed.

Thank you 😁