Koenkk/zigbee2mqtt #10991: [New device support]: Pro Line ZigBee Dry Contact On/Off by Tuya

volodymyrmarch, 2022-01-24:
### Link

https://tervix.ua/product/Tervix/rozumniy_budinok/431181_rozumniy_peremikach_tervix_pro_line_zigbee_dry_contact_on_off_rele_z_sukhim_kontaktom/

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x847127fffec43f48","nwkAddr":25582,"manufId":4098,"manufName":"_TZE200_iv1mzha2","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_iv1mzha2","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{}},"genPowerCfg":{"attributes":{}},"genIdentify":{"attributes":{}},"genOnOffSwitchCfg":{"attributes":{}},"nearestGateway":{"attributes":{}},"aqaraOpple":{"attributes":{}},"genOta":{"attributes":{}},"manuSpecificTuya_2":{"attributes":{}},"manuSpecificTuya_3":{"attributes":{}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc3730","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc3730","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1643029400717,"useImplicitCheckin":true}

### Comments

Hello! Tried to add relay for Gas Boiler that works with tuya smart app. On application it has 2 state - On and Off.

tried to add converter but nothing happened after I change state. And after refresh state I get this error

`Publish 'get' 'state' to '0x847127fffec43f48' failed: 'Error: Read 0x847127fffec43f48/1 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

From iot.tuya debug I see this field 

`switch | Boolean | "{true,false}"` 

and json is 

`{ 'switch': 'true'}`

 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_iv1mzha2'}],
        model: 'TZE200',
        vendor: 'Tervix',
        description: 'Pro Line ZigBee Dry Contact On/Off',
        extend: extend.switch(),
        fromZigbee: [e.switch()],
        toZigbee: [tz.on_off],
        exposes: [e.switch()],
    };

module.exports = definition;
```

closed 2022-01-31 by volodymyrmarch

back2war, 2025-05-22:
@volodymyrmarch any luck making this work with zigbee2mqtt. For me it is still shown as Not supported.

back2war, 2025-05-22:
In case if somebody needs 

`/homeassistant/zigbee2mqtt/external_converters/ts0601_iv1mzha2.mjs`

```
import * as tuya from "zigbee-herdsman-converters/lib/tuya";
import exposes from "zigbee-herdsman-converters/lib/exposes";

const e = exposes.presets;
const ea = exposes.access;

export default {
    fingerprint: tuya.fingerprint("TS0601", ["_TZE200_iv1mzha2"]),
    model: "Tervix-Dry-Relay",
    vendor: "Tuya",
    description: "Tervix Dry Relay",
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [e.switch().setAccess("state", ea.STATE_SET)],
    extend: [],
    meta: {
        tuyaDatapoints: [[1, "state", tuya.valueConverter.onOff]],
    },
};

```