Koenkk/zigbee2mqtt #10993: Sunricher SR-ZGP2801K4-DIM does not match fingerprint in devices.js

Andreas-Hellberg, 2022-01-24:
### What happened?

Tried to pair a new Sunricher SR-ZGP2801K4-DIM, and it shows up as unsupported. Found that it's because the IEEE address did not match the fingerprint in sunricher.js. My device shipped with the following IEEE 0x0000000057e24e5d (0x4E5D)

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/sunricher.js
{
    fingerprint: [{modelID: 'GreenPower_2', ieeeAddr: /^0x000000005d5.....$/}],
    model: 'SR-ZGP2801K4-DIM',
    vendor: 'Sunricher',
    description: 'Pushbutton transmitter module',
    fromZigbee: [fz.sunricher_switch2801K4],
    toZigbee: [],
    exposes: [e.action(['press_on', 'press_off', 'press_high', 'press_low', 'hold_high', 'hold_low', 'release'])],
}

### What did you expect to happen?

Expected it to pair and get identified. 

### How to reproduce it (minimal and precise)

Buy a Sunricher SR-ZGP2801K4-DIM that does not have a IEEE address that starts with 5d5. 

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26700700

### Adapter

Conbee II

Andreas-Hellberg, 2022-01-24:
Yes, I ended up adding a external converter and my switch works just fine. Copied the settings from the sunricher.js and just modified the fingerprint. To the following, but yes  /^0x0000000057e.....$/}] should work fine also: 

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;

const definition =  {
	fingerprint: [{modelID: 'GreenPower_2', ieeeAddr: /^0x000000005.......$/}],
	model: 'SR-ZGP2801K4-DIM',
	vendor: 'Sunricher',
	description: 'Pushbutton transmitter module',
	fromZigbee: [fz.sunricher_switch2801K4],
	toZigbee: [],
	exposes: [e.action(['press_on', 'press_off', 'press_high', 'press_low', 'hold_high', 'hold_low', 'release'])],
};

module.exports = definition;

Andreas-Hellberg, 2022-01-24:
[image]

Koenkk, 2022-01-25:
thanks, added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-25 by Koenkk

ananace, 2022-07-15:
Throwing a note here that a SR-ZGP2801K4-DIM switch I just received also did not match the included fingerprints, mine shipped with the address `0x00000000f126c331`

Adding an external converter like this has worked fine though;
```js
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;

const definition = {
  fingerprint: [{modelID: 'GreenPower_2', ieeeAddr: /^0x00000000f.......$/}],
  model: 'SR-ZGP2801K4-DIM',
  vendor: 'Sunricher',
  description: 'Pushbutton transmitter module',
  fromZigbee: [fz.sunricher_switch2801K4],
  toZigbee: [],
  exposes: [e.action(['press_on', 'press_off', 'press_high', 'press_low', 'hold_high', 'hold_low', 'release'])],
};

module.exports = definition;
```

[image]

Koenkk, 2022-07-17:
@ananace added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

charlesrg, 2023-05-12:
Please, where did you buy your SR-ZGP2801K4-DIM ? Looking to buy them but can't find. Just really loved the batteryless switches.

charlesrg, 2023-05-13:
> @charlesrg I bought them at the store below. But try Google for sunricher kinetic switch.
> 
> https://www.kjell.com/se/varumarken/sunricher/smarta-hem/philips-hue/philips-hue-tillbehor/sunricher-kinetisk-fjarrstrombrytare-for-vaggmontering-p51601

Thank you . I definitely tried googling before asking. It looks like it's available only Swedish stores are selling it. Can't find it in USA.

Testo17, 2023-07-19:
I got the Sunricher SR-ZGP2801K4-DIM paired correctly, but I can't get the second rocker to work. When pairing both rockers separately, those will appear as separate devices with unique IEEE addresses. Problem is that even though I press either of the rockers, it will affect only to the rocker/device that was paired first. The second rocker doesn't update the LQI at all and with that device you can't do anything. Any idea how to get the second rocker to work?