Koenkk/zigbee2mqtt #11002: Aqara WRS-R02 double rocker binding not binding the second button

Steekvlam, 2022-01-25:
### What happened?

Im trying to bind the Aqara H1 wireless double rocker (WRS-R02) with the left button for lights hallway & right button for lights upstairs.
Now when I bind the left key, it works great as expected, but somehow the right key also binds with the left button so I can only bind 1 devices for both buttons.

### What did you expect to happen?

I expect to bind both left and right key to different devices.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26580700

### Adapter

DECONZ ConBee II Zigbee

Koenkk, 2022-01-26:
@kirovilya do you know if the device supports this?

kirovilya, 2022-01-26:
I couldn't get it to work in a two-button binding.
left a comment about this here https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js#L1577
Perhaps Aqara will fix it with new firmware, but I have not seen them yet.

closed 2022-03-05 by github-actions[bot]

bzeiss, 2022-10-22:
Even though this issue is closed, I do wonder if there is any chance in making this work (see also last comment in https://github.com/Koenkk/zigbee-herdsman-converters/issues/2620). It would be really great if it was possible to differentiate the buttons when the device is in command mode.