Koenkk/zigbee2mqtt #11016: Tuya zigbee motor (TS0601 _TZE200_nogaemzt) does not work correctly

mvbl, 2022-01-25:
### What happened?

Motor was connect to z2m correctly.

```
Model Zigbee: TS0601
Producer Zigbee: _TZE200_nogaemzt
Producer: TuYa
Model: TS0601_cover
```

1. When trying to open or close, the motor does not stop.

The only way to stop is to send a stop command, but an error message appears:

```
warn  2022-01-25 20:52:30: TuYa_cover_control: Unhandled DP #6 for _TZE200_nogaemzt:
                {"seq":19968,"dp":6,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}
```

2. Position and motor speed don't working. For example, i tried to send commands:

- Open
- Close
- Stop
- Set position to 21
- Set speed to 62

Got a log like this:
```
debug 2022-01-25 21:32:44: Received MQTT message on 'zigbee2mqtt/0x0c4314fffe6e886d/set' with data '{"state":"OPEN"}'
debug 2022-01-25 21:32:44: Publishing 'set' 'state' to '0x0c4314fffe6e886d'
debug 2022-01-25 21:32:44: TuYa_cover_control: Using state enums for _TZE200_nogaemzt:
                {"close":2,"open":0,"stop":1}
debug 2022-01-25 21:32:44: Received Zigbee message from '0x0c4314fffe6e886d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":48}' from endpoint 1 with groupID 0
debug 2022-01-25 21:32:45: Received Zigbee message from '0x0c4314fffe6e886d', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"seq":25600}' from endpoint 1 with groupID 0
info  2022-01-25 21:32:45: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe6e886d', payload '{"linkquality":149,"position":100,"running":true,"state":"OPEN"}'
debug 2022-01-25 21:32:45: Received Zigbee message from '0x0c4314fffe6e886d', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":7,"fn":0,"seq":25856}' from endpoint 1 with groupID 0
debug 2022-01-25 21:32:47: Received MQTT message on 'zigbee2mqtt/0x0c4314fffe6e886d/set' with data '{"state":"CLOSE"}'
debug 2022-01-25 21:32:47: Publishing 'set' 'state' to '0x0c4314fffe6e886d'
debug 2022-01-25 21:32:47: TuYa_cover_control: Using state enums for _TZE200_nogaemzt:
                {"close":2,"open":0,"stop":1}
debug 2022-01-25 21:32:47: Received Zigbee message from '0x0c4314fffe6e886d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":49}' from endpoint 1 with groupID 0
debug 2022-01-25 21:32:47: Received Zigbee message from '0x0c4314fffe6e886d', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"seq":26112}' from endpoint 1 with groupID 0
info  2022-01-25 21:32:47: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe6e886d', payload '{"linkquality":145,"position":0,"running":true,"state":"CLOSE"}'
debug 2022-01-25 21:32:47: Received Zigbee message from '0x0c4314fffe6e886d', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":7,"fn":0,"seq":26368}' from endpoint 1 with groupID 0
debug 2022-01-25 21:32:49: Received MQTT message on 'zigbee2mqtt/0x0c4314fffe6e886d/set' with data '{"state":"STOP"}'
debug 2022-01-25 21:32:49: Publishing 'set' 'state' to '0x0c4314fffe6e886d'
debug 2022-01-25 21:32:49: TuYa_cover_control: Using state enums for _TZE200_nogaemzt:
                {"close":2,"open":0,"stop":1}
debug 2022-01-25 21:32:49: Received Zigbee message from '0x0c4314fffe6e886d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":50}' from endpoint 1 with groupID 0
debug 2022-01-25 21:32:50: Received Zigbee message from '0x0c4314fffe6e886d', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":6,"fn":0,"seq":26624}' from endpoint 1 with groupID 0
warn  2022-01-25 21:32:50: TuYa_cover_control: Unhandled DP #6 for _TZE200_nogaemzt:
                {"seq":26624,"dp":6,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}
debug 2022-01-25 21:32:51: Received Zigbee message from '0x0c4314fffe6e886d', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"seq":26880}' from endpoint 1 with groupID 0
info  2022-01-25 21:32:51: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe6e886d', payload '{"linkquality":145,"position":0,"running":true,"state":"CLOSE"}'
debug 2022-01-25 21:32:52: Received MQTT message on 'zigbee2mqtt/0x0c4314fffe6e886d/set' with data '{"position":21}'
debug 2022-01-25 21:32:52: Publishing 'set' 'position' to '0x0c4314fffe6e886d'
debug 2022-01-25 21:32:52: Received Zigbee message from '0x0c4314fffe6e886d', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,79],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"seq":51}' from endpoint 1 with groupID 0
info  2022-01-25 21:32:52: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe6e886d', payload '{"linkquality":145,"position":21,"running":true,"state":"OPEN"}'
debug 2022-01-25 21:32:54: Received MQTT message on 'zigbee2mqtt/0x0c4314fffe6e886d/set' with data '{"options":{"motor_speed":62}}'
debug 2022-01-25 21:32:54: Publishing 'set' 'options' to '0x0c4314fffe6e886d'
info  2022-01-25 21:32:54: Setting motor speed to 62
debug 2022-01-25 21:32:57: Saving state to file /share/zigbee2mqtt/state.json
```
Position and speed - not working, motor behaviour does not change in any way. Stop - causes an error and motor stops.

### What did you expect to happen?

I would like to get all the features that are stated here https://www.zigbee2mqtt.io/devices/TS0601_cover.html

In addition to working "open" and "close", I would like to set position and stop without errors.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211217

### Adapter

SLS CC2652P stick v1.2

mvbl, 2022-01-25:
It's a shame, bought a lot of motors and they all have this behavior...:(

andreypopov, 2022-01-26:
Similar problem, but I can not control my motor, but I can see status updates

xoref, 2022-01-28:
I have the exact same, manual and remote control works, but it seems to completely ignore ZigBee commands send from z2m since a few days. Status updates are still reported and correct. I didn't perform any updates, so I will try recharging them and then pair again 🤞

mvbl, 2022-02-08:
> [image]
> 
> i have problem

Heh.... It's a pitty :( if you solve this problem, please, let me know.

mvbl, 2022-02-17:
@Koenkk can you help us?

closed 2022-02-18 by mvbl