Koenkk/zigbee2mqtt #11019: [New device support]: Aqara Presence Detector FP1 (RTCZCGQ11LM; lumi.motion.ac01)

Otnow, 2022-01-25:
### Link

https://www.aqara.com/cn/productDetail/Aqara-Presence-Detector-FP1

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x54ef44100042d6eb","nwkAddr":13551,"manufId":4660,"manufName":"aqara","powerSource":"DC Source","modelId":"lumi.motion.ac01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":65520,"inClusterList":[0,3,64704],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.motion.ac01","manufacturerName":"aqara","powerSource":4,"zclVersion":3,"appVersion":48,"stackVersion":2,"hwVersion":1,"dateCode":"20210121","swBuildId":""}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[3,40,28,5,33,1,0,8,33,48,1,10,33,0,0,12,32,20,16,32,1,18,32,0,101,32,1,102,32,6,103,32,0,104,32,0,105,32,1,106,32,1,107,32,3]},"322":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":48,"stackVersion":2,"hwVersion":1,"dateCode":"20210121","swBuildId":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1643135963340,"useImplicitCheckin":true}

### Comments

[image]

Review:
-
- [appearance and equipment (photos)](https://www.reddit.com/r/Aqara/comments/schrge/aqara_presence_detector_fp1_rtczcgq11lm/)
- [Aqara Home app (screenshots)](https://www.reddit.com/r/Aqara/comments/scht7o/aqara_presence_detector_fp1_rtczcgq11lm/)
- [disassembly (photos)](https://www.reddit.com/r/Aqara/comments/schunq/aqara_presence_detector_fp1_rtczcgq11lm/)

Buy:
-
- https://aliexpress.com/item/1005003772245093.html
- https://aliexpress.com/item/1005003781853869.html
- https://aliexpress.com/item/1005003766378575.html

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.motion.ac01'],
    model: 'RTCZCGQ11LM',
    vendor: 'Xiaomi',
    description: 'Aqara Presence Detector FP1',
    fromZigbee: [],
    toZigbee: [],
    exposes: [],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

z-x-f, 2022-01-26:
Wow, you got this new device so fast, I ordered it here and it hasn't arrived yet.

Otnow, 2022-01-26:
> Wow, you got this new device so fast, I ordered it here and it hasn't arrived yet.

Yes, lucky with the speed this time 🙂

It's good that it's coming to you too, so I hope we (and maybe someone else) will add support for it together, because I can hardly do it alone 🤝

A quick inspection of the exchange protocol again shows the abundant use of Aqara proprietary octet string 🤦‍♂️

Koenkk, 2022-01-31:
Added in https://github.com/Koenkk/zigbee2mqtt/issues/11019

closed 2022-01-31 by Koenkk

rdorys, 2022-01-31:
Hi @Otnow , do you know if the sensor can handle two people or more? Is there one point per people on the map ?

And what about pets? Can it detect it and see it's not human ?

Thx for your work

Otnow, 2022-01-31:
> Hi @Otnow , do you know if the sensor can handle two people or more? Is there one point per people on the map ?
> 
> And what about pets? Can it detect it and see it's not human ?
> 
> Thx for your work

Hi. I have not yet figured out the logic of this sensor well enough, so I can’t say for sure right now.

I did not check with several people, but judging by [this video review](https://www.youtube.com/watch?v=EhqXFNI6pmk), the sensor displays only 1 person on the map (this is only the first information so far, we will find out more accurately later).

I don't have pets, but it seems to me that the sensor will notice their movement, because. it seems that he reacts to everything that even slightly changes position in space (for example, a very slight swaying of a curtain). Will he be able to understand that this is not a person, I don’t know, but I doubt it.

The sensor has a very peculiar logic of operation, so in the process of adding the function of working with regions, I hope to understand it better.

qvntra, 2022-02-02:
Subscribed. Very interested to see the progress here.

Inrego, 2022-02-22:
Where can we track progress/discussion of more advanced features?

Otnow, 2022-02-22:
> Where can we track progress/discussion of more advanced features?

You can discuss here, so write if you have additional information or questions.

piggydamn, 2022-03-06:
it cant join z2m,the log:

warn  2022-03-06 10:18:59: `permit_join` set to  `true` in configuration.yaml.
warn  2022-03-06 10:18:59: Allowing new devices to join.
warn  2022-03-06 10:18:59: Set `permit_join` to `false` once you joined all devices.
warn  2022-03-06 10:19:00: Using experimental new availability feature
warn  2022-03-06 10:21:21: Device '0x54ef441000473044' with Zigbee model 'lumi.motion.ac01' and manufacturer name 'aqara' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
warn  2022-03-06 10:21:22: Received message from unsupported device with Zigbee model 'lumi.motion.ac01' and manufacturer name 'aqara'
warn  2022-03-06 10:21:22: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Otnow, 2022-03-10:
> Unfortunately, I tried many times, FP1 doesn't seem to support ZigBee 22 channel

Just checked, indeed the sensor does not want to work on channel 22.

Otnow, 2022-03-12:
[RTCZCGQ11LM converter changed](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3966) for support new functionality in [firmware №53](https://github.com/Koenkk/zigbee-OTA/pull/100): `motion_sensitivity` and `reset_nopresence_status` (is now available in the _[dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)_, and will appear in the _master branch_ at the beginning of next month):

[image]
[image]
[image]

_**Important**: before updating the sensor firmware to version 53, you must install a zigbee2mqtt version that supports this firmware version, otherwise `presence_event` will report incorrect data due to the changed logic of the firmware._

Otnow, 2022-03-30:
> hello, what progress has been made regarding the integration of the regions? There is an integrated Tuya occupancy sensor on Zigbee2mqtt and the regions are well integrated, the regions on the tuya sensors are so different from the FP1 sensors?

At the moment, there is no support for regions and, unfortunately, integration of Tuya sensor will not help in any way, because Aqara has a completely different logic of work.

jeezy51, 2022-04-14:
what is the logic of the sensor, is it digital data based on X (bottom left) and Y (top right) coordinates like Roborock vacuum cleaners? Data according to the number of boxes (4X7) Or data based on the coupling of approach distance and direction of travel? Do you currently know the key related to the information of the regions? because in debug I see 6 keys which are unknown and if I manage to know the digital info of the region I can process it via my home automation server.

jeezy51, 2022-04-14:
if I understood correctly, the region information is indeed uploaded in Zigbee2mqtt but you don't know how to interpret it, is that right?

Otnow, 2022-04-16:
@jeezy51, I'm not currently developing region support due to other priorities, so I can't answer your questions 🤷

Otnow, 2022-06-01:
> I'm not sure is my unit still with the old firmware but the sensitivity and resetting nobody is not available. I try to send a message to do OTA firmware check but return available: false.

First, I think you need to update _zigbee2mqtt_ to the current version ([1.25.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.25.2)).

What version do you have installed?

jayden-jw, 2022-06-02:
> First, I think you need to update _zigbee2mqtt_ to the current version ([1.25.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.25.2)).
>
> What version do you have installed?

Thanks for the tips, I didn't know since when all my add-on repositories are gone, so I didn't see any new update on zigbee2mqtt, now I have added back and upgraded to the latest version, everything working except the temperature always at 40c.

mihsu81, 2022-06-08:
I'm having the same issue. Both devices have firmware build date 20210121, and can't find an update available.
Z2MQTT is the latest version 1.25.2. Am I missing something in my config?

Thanks in advance.

[image]
[image]
[image]

mihsu81, 2022-06-08:
In the logs I can see:
```
Debug 2022-06-08 20:15:46Got OTA request '{"fieldControl":1,"manufacturerCode":4447,"imageType":8347,"fileVersion":53}'
Debug 2022-06-08 20:15:46ZigbeeOTA: downloaded main index
Debug 2022-06-08 20:15:46Is new image available for '0x54ef4410004d9819', current '{"fieldControl":1,"manufacturerCode":4447,"imageType":8347,"fileVersion":53}', latest meta '{"fileVersion":53,"fileSize":317370,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220124174454_OTA_lumi.motion.ac01_0.0.0_0053_20220124_93246C.ota","sha512":"2d7c7c0d3f3e58dbe2fd5a14224fb7bfcfa51c0763b9d7c3ff3203ab1b8b9ea832b1d619480e660c41d26d9860fbf27ad469670a4b2da424bf929f5f17577e25"}'
Debug 2022-06-08 20:15:46Update available for '0x54ef4410004d9819': NO
Info 2022-06-08 20:15:46No update available for 'MASTER-BEDROOM-PRESENCE'
Info 2022-06-08 20:15:46MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"No update available for 'MASTER-BEDROOM-PRESENCE'","meta":{"device":"MASTER-BEDROOM-PRESENCE","status":"not_available"},"type":"ota_update"}'
```

It's as if it is not recognized as a newer firmware.

mihsu81, 2022-06-08:
@Otnow How can it be the same when the firmware build date on my devices is "20210121" and the build date of the ota file is "20220124".

mihsu81, 2022-06-08:
> > How can it be the same when the firmware build date on my devices is "20210121" and the build date of the ota file is "20220124".
> 
> You need to ask _Aqara_ why they did not update the build date in the latest firmware 🤷‍♂️🙃

My apologies then. I was under the impression that there is a bug not allowing the update. Thanks for your help and for taking your time to explain. 🙂

Otnow, 2022-06-08:
> Does anyone know if it's possible to configure the regions of the Aqara FP1 with the Aqara Hub and move the FP1 with those settings into zigbee2mqtt?I don't have high hopes as you have to reset the device to pair with another controller 🙈, but contemplating getting an Aqara Hub just for being to configure regions in these sensors (as they are really awesome 😍).

No, unfortunately this will not work, because I checked this possibility and after re-pairing the sensor resets all settings (including regions).

ScarletSeven, 2022-06-11:
Anyone has an issue with this presence sensor not clearing presence state sometimes? It's just stuck until I hit the "reset_nopresence_status" button in Z2M

Otnow, 2022-06-11:
> Anyone has an issue with this presence sensor not clearing presence state sometimes? It's just stuck until I hit the "reset_nopresence_status" button in Z2M

Unfortunately, yes, rarely, but it happens.
I think this button was provided for this situation.

I hope that this problem, as well as the low speed of the start of presence detection, will be fixed in the new firmware.

Otnow, 2022-06-16:
@ScarletSeven, by the way, a [new firmware version](https://github.com/Koenkk/zigbee-OTA/pull/127) (№ **54**) has just been released, so you can update the sensor in _z2m_ on the _OTA_ tab and check if the problem you indicated has been fixed.

ScarletSeven, 2022-06-17:
Wow nice thanks, can’t wait to try updating.

I noticed that the presence sensor sometimes glitched when a person comes in and out at the edge of the detection range relatively quickly, that’s when the presence sensor gets stuck and doesn’t clear its status 

Regards,
Jonathan

Adscelerate.com

lst3245, 2022-06-28:
> Anyone has an issue with this presence sensor not clearing presence state sometimes? It's just stuck until I hit the "reset_nopresence_status" button in Z2M

I just experienced the same issue and searching online came across this issue. While I didn't find a solution, I think that I found out how is this reproduced (not 100% sure though).

My sensor is working perfectly fine at far approach distance and high sensitivity before. But yesterday it consistently failed as it stuck at the "approach" and "presence" state, not sometimes, but actually 7-8 out of 10 times. Originally I was running the stock firmware and tried upgrading it (the one Otnow mentioned), and it didn't make any difference for me.

After thinking a lot, I noticed that in my area it is getting much windier, and I have some hanging clothes within the detection area of the sensor. Indeed, after I closed my windows, my sensor started detecting "leave" and "no presence" again. So my hypothesis is like this: When there are interfering objects like moving clothes/curtain, or a fan, the sensor normally won't trigger by those objects alone, but if a human triggers the sensor and leave, the sensor sometimes detects those interfering objects as human, and hence stuck at "presence" state.

I went further to test this:
1) I put a fan in front of the sensor
2) Turned it on remotely, indeed the sensor was not triggered by it
3) I walked in front of the sensor and triggered it, and then leave
4) The sensor was stuck at "presence"
5) Turned off the fan remotely and after a while, the sensor went back to "leave"/"no presence"

So my temporary solution is to set the motion sensitivity to low, which seemed to be helping. But I don't think the approach distance setting will help as it doesn't affect presence detection, but just "how close to the sensor is regarded as approach or away". I wonder if others can reproduce the problem as mentioned. Removing possible interfering objects could help. Also, I don't have the aqara hub in hand, so if you got one, you could see if the official app detects those interfering objects.

mvgijssel, 2022-06-29:
Seems folks from deconz have figured out some of the region/area logic of the Aqara FP1! https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5928#issuecomment-1166545226

cook1emr, 2022-07-13:
> hello, what progress has been made regarding the integration of the regions? There is an integrated Tuya occupancy sensor on Zigbee2mqtt and the regions are well integrated, the regions on the tuya sensors are so different from the FP1 sensors?

Which tuya sensor are you talking of? Am I misunderstanding what regions are. I think regions is the device is capable of seeing where I am in a room. I cant see any documents that show any tuya occupancy sensor is capable of this.

MattL0, 2022-07-14:
Maybe this ? https://solution.tuya.com/projects/CMb6fp2yued4wt

jeezy51, 2022-07-18:
> Which tuya sensor are you talking of? Am I misunderstanding what regions are. I think regions is the device is capable of seeing where I am in a room. I cant see any documents that show any tuya [...]

I speak well of the zones, the tuya sensor manages them on zigbee2mqtt, it also manages the fall detection

https://fr.aliexpress.com/item/1005003461158730.html?spm=a2g0o.productlist.0.0.65ab22f6I9ow3l&algo_pvid=c75bc94c-4c47-4a0c-a7aa-8a96bdded761&algo_exp_id=c75bc94c-4c47-4a0c-a7aa-8a96bdded761-1&pdp_ext_f=%7B%22sku_id%22%3A%2212000025891550789%22%7D&pdp_npi=2%40dis%21EUR%21%21116.87%21%21%21%21%21%402100bdf016581851635038352e6f7b%2112000025891550789%21sea

cook1emr, 2022-07-19:
> I speak well of the zones, the tuya sensor manages them on zigbee2mqtt, it also manages the fall detection [...]

is this what "area" is doing? on the MIR-HE200-TY_fall. I.E can you set a zone on a map for instance and it knows if you walk into that zone? sorry looking to buy one of these but documentation is hard to come by.

mvgijssel, 2022-07-20:
This person https://percy10442.pixnet.net/blog/post/563851354-aqara-fp1-人體存在傳感器整合入home-assistant使用 integrated the FP1 into home assistant using the Aqara gateway integration. The article describes how the different regions and their states are represented by numerical values.

Thought that this might help figuring out how to integrate this into Z2M!

Almaz58, 2022-07-26:
Hello.
There is a sonoff dongle plus stick (version 20220219), Zigbee2MQTT (1.26.0) and aqara fp1 device (made in June 2022).
 Zigbee2MQTT cannot detect the device in any way, writes that it is not supported.
Tried everything:
1. Installed different versions of sonoff dongle plus, deleted data (erase).
2. Reinstalled Zigbee2MQT several times, cleared the folder, tried the EDGE version. I did everything with a reboot.
3. Changed channels.
4. Changed chargers for Aqara FP1.
In general, I tried everything that is possible. The logs say:
...
2022-07-26T11:52:57.513Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x54ef44100051847e', attempt 5
2022-07-26T11:52:57.514Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":5164,"nwkaddrofinterest":5164}
2022-07-26T11:52:57.514Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,44,20,44,20,35]
2022-07-26T11:52:57.520Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2022-07-26T11:52:57.521Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2022-07-26T11:52:57.521Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2022-07-26T11:52:57.522Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2022-07-26T11:52:57.523Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-07-26T11:53:07.525Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '5164' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2022-07-26T11:53:07.525Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 5164
2022-07-26T11:53:07.526Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":5164,"options":0,"radius":30}
2022-07-26T11:53:07.526Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,44,20,0,30,66]
2022-07-26T11:53:07.537Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2022-07-26T11:53:07.537Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2022-07-26T11:53:07.537Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2022-07-26T11:53:07.538Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2022-07-26T11:53:07.538Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-07-26T11:53:10.540Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":5164,"nwkaddrofinterest":5164}
2022-07-26T11:53:10.541Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,44,20,44,20,35]
2022-07-26T11:53:10.547Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2022-07-26T11:53:10.548Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2022-07-26T11:53:10.548Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2022-07-26T11:53:10.548Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2022-07-26T11:53:10.548Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-07-26T11:53:20.550Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2022-07-26T11:53:20.552Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2022-07-26T11:53:20.552Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x54ef44100051847e', attempt 6
...

Almaz58, 2022-07-26:
> Hello. There is a sonoff dongle plus stick (version 20220219), Zigbee2MQTT (1.26.0) and aqara fp1 device (made in June 2022). Zigbee2MQTT cannot detect the device in any way, writes that it is not [...]

Problem solved. You have to press the button for 10 seconds. (until it starts flashing). Then turn on the search in z2m. Then press the button again until it starts flashing (about 5 seconds).

Romario1972, 2022-07-28:
Hello! 
I've just received my new one FP1 and trying to add it to my HA Z2M add-on. But it joined as "Unsupported". I have in my logs:

 ```
Warning Received message from unsupported device with Zigbee model 'lumi.motion.ac01' and manufacturer name 'LUMI'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html. 
```
But it should connect without any external converters.
Zigbee2mqtt: Current version: 1.18.1-1
Zigbee module: ZigStar LAN Gateway (RF-STAR RF-BM-2652P)(CC2652P)

What should I do next?

definitio, 2022-07-28:
You should update zigbee2mqtt to 1.26.

hqsone, 2022-08-09:
I have the same problem as described above. I have a lot problems with joining and maximus what I can get is "unsupported".
Z2m version is 1.27.0
What channel do you use?

Lyr3x, 2022-08-09:
Is it possible to expose the reset state button to HA as well? That would be really handy to quickly set the sensors state to off again.

npty, 2022-08-13:
> I have the same problem as described above. I have a lot problems with joining and maximus what I can get is "unsupported". Z2m version is 1.27.0 What channel do you use? Update: Changed channel to 15, and it started work.
I've also changed to channel 15, but i can't make it works 🥲

hqsone, 2022-08-13:
> > I have the same problem as described above. I have a lot problems with joining and maximus what I can get is "unsupported". Z2m version is 1.27.0 What channel do you use? Update: Changed channel to 15, and it started work.
> 
> I've also changed to channel 15, but i can't make it works 🥲

Update: I had to change zigbee dongle to new one. Same chip but from another manufacturer.

npty, 2022-08-15:
> > I have the same problem as described above. I have a lot problems with joining and maximus what I can get is "unsupported". Z2m version is 1.27.0 What channel do you use? Update: Changed channel to 15, and it started work.
> 
> I've also changed to channel 15, but i can't make it works 🥲

Finally, i've made it work! The solution is to change to channel 25.

sorryusernameisalreadytaken, 2022-09-01:
> Is it possible to expose the reset state button to HA as well? That would be really handy to quickly set the sensors state to off again.

Is that possible, did you find a way?

> Anyone has an issue with this presence sensor not clearing presence state sometimes? It's just stuck until I hit the "reset_nopresence_status" button in Z2M

My sensor is on fw 54. My z2m docker on v1.27.2-dev commit: [178f72c](https://github.com/Koenkk/zigbee2mqtt/commit/178f72c)
Zigbee channel changed from  to 15 to 25. zStack3x0 Coordinator-Stick fw 20220219

Any idea why fp1 sensor just sometimes switch back to no-presence state?

husnainali09, 2022-09-04:
How can I update the firmware on this device ? I don't have a aqara hub (just Xiaomi hub which won't pair with this) and zigbee2mqtt says no firmware found, my firmware date shown in Z2m is from October 2021?

sorryusernameisalreadytaken, 2022-09-04:
> How can I update the firmware on this device ? I don't have a aqara hub (just Xiaomi hub which won't pair with this) and zigbee2mqtt says no firmware found, my firmware date shown in Z2m is from October 2021?

As far as I have understood correctly, the string that is displayed in the z2m web interface is the build_number and not the firmware!

To get the real firmware version of your device, you should read the following post:
https://community.home-assistant.io/t/new-aqara-human-presence-sensor-fp1/373471/186?u=jamieme

ykuijs, 2022-09-13:
I recently bought an Aqara FP1 via AliExpress and am trying to pair this device in Z2M, but the device keeps being detected as unsupported. I have already tried to remove the device and re-pair it, but this caused me having to reinstall Z2M before I was able to pair the device again.

I have set the channel to 25, which did not change the issue.

How can I get the device to be detected properly?

Idan37S, 2022-09-29:
Anyone knows if it works with channel 24?
I couldn't find any info about that

colinnewman2000, 2022-10-16:
> Anyone knows if it works with channel 24? I couldn't find any info about that

Sadly (for me and others) channels 21-24 are not supported :(

maxi1134, 2022-10-27:
Any update on the region integration?

patmalcolm91, 2022-12-18:
> Any update on the region integration?

I've just managed to write an external converter that adds basic support for regions (including entrances/exits, interference sources, and edges). See here: https://github.com/Koenkk/zigbee2mqtt/issues/13711#issuecomment-1356182773. There's still some work to be done to improve the UI/UX, but the basic functionality is there.

maxi1134, 2022-12-23:
> > Any update on the region integration?
> 
> I've just managed to write an external converter that adds basic support for regions (including entrances/exits, interference sources, and edges). See here: [#13711 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13711#issuecomment-1356182773). There's still some work to be done to improve the UI/UX, but the basic functionality is there.

The chosen one!

hansdampf01, 2023-02-03:
Hey guys,  I just started playing around with zigbee2mqtt and added an fp1. Works out of the box for me BUT Im missing an entities called sensor enter / sensor leave. I had those when I added the fp1 via ZHA but not on z2m. This is literally mission critical as the whole wife approval goes down with this. When I use the presence status for turning on/off  light there is a huge delay in recognizing. The sensor enter / leave works instantly.

DannyFeliz, 2023-02-03:
@hansdampf01 I had the same problem, but I fixed it by using the events enter for detection, here is a tutorial about how to do it https://smarthomescene.com/blog/aqara-fp1-tips-tricks-and-automations-examples/