Koenkk/zigbee2mqtt #11021: Unable to pair Danfoss 014G2460

rmk-ch, 2022-01-25:
### What happened?

I moved from Deconz to zigbee2mqtt and paired almost all devices (Aqara, Sonoff, Philips Hue) but I am not able to pair the Danfoss 014G2460. I do not see any hint in the log file that a new device is trying to pair. (related discussion: https://github.com/Koenkk/zigbee2mqtt/issues/10678)

The device was working with Deconz without any issues and in the exact same hardware setup before.

Things I tried:
- I updated Conbee2 to the latest stable firmware
- I replaced the battery
- I have Conbee2 connected via USB cable
- Conbee2 and z2m are connected/run on a standard laptop (not a raspberry pi)
- Moved the Conbee2 around (Danfoss device is ~2..3m away, other devices are much further away)
- Update to dev version (on Docker)

Let me know if I can help debugging the issue. I can also debug it if could point me where to start....
Thanks a lot!!

### What did you expect to happen?

Pair like any other device.
At least a log message to show where to look for.

### How to reproduce it (minimal and precise)

1. Enable pair (either via frontend or still enabled in the configuration.json from previous try)
2. Remove battery from device
3. Press button for a few seconds
4. Add battery again
5. Press button for a while until motor opens valve completly
6. I see "M" blinking on the devices display (saying it is in pairing mode)
7. Wait some minutes watching the zigbee2mqtt log showing nothing

### Zigbee2MQTT version

1.22.2-dev commit: c4c02d2

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/7936665/z2m.log)

lux73, 2022-01-26:
https://github.com/Koenkk/zigbee2mqtt/discussions/10119

not possible - seems the ConBee has serious Problems with pairing these

rmk-ch, 2022-01-26:
If it would be a hardware issue, why is it working with Deconz then? (Also, both adapter and device are listed as supported, so there seem to be setups where it works - otherwise documentation should be updated as well)
I am willing to debug the issue, just tell me where to start please...

closed 2022-03-05 by github-actions[bot]

rmk-ch, 2022-05-13:
Hi again. I bought a new zigbee gateway, the SONOFF Zigbee 3.0 USB Dongle Plus and installed the latest firmware as suggested. I was able to re-pair all of the devices but not the Danfoss Ally that is 3 meters away in the same room as the computer with the dongle connected via 1m USB cable. There is still no sign it is alive. I tried removing it from the radiator, switched batteries, updated zigbee2mqtt to the latest release. Nothing works. I tried enabling herdsman debug but I only see messages from other devices.

Some log details
`zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 21:45:25: Starting Zigbee2MQTT version 1.25.1 (commit #3f6a137)
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 21:45:25: Starting zigbee-herdsman (0.14.27)
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 21:45:25: zigbee-herdsman started (resumed)
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 21:45:25: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 21:45:25: Currently 9 devices are joined:
......
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 21:45:25: Zigbee: disabling joining new devices.
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 21:45:26: Connecting to MQTT server at mqtt://mqtt.ht.roman-koller.ch
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 21:45:26: Connected to MQTT server
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 21:45:26: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 21:45:26: Started frontend on port 0.0.0.0:8080
.....
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 22:00:09: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"ank3g-3"}'
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 22:00:11: Zigbee: allowing new devices to join.
zigbee2mqtt         | Zigbee2MQTT:info  2022-05-13 22:00:11: Succesfully changed options
`
I would really like to get this thing running.

Currently I am unsure whether the thermostat is defective or if zigbee2mqtt is the issue. Please tell me how I could debug this issue.

Thanks for any help.

dperezu76, 2023-09-17:
I am thinking of purchasing this device.  Did you fix the problem?

rmk-ch, 2024-01-09:
Finally solved today by 1. factory reset, 2. repairing as [discussed here](https://github.com/Koenkk/zigbee2mqtt/discussions/10119#discussioncomment-8071645). Please see [docs](https://www.zigbee2mqtt.io/devices/014G2461.html).