Koenkk/zigbee2mqtt #11022: [New device support]: Philips Hue Centura 3-pack (9290019531)

rtfmjoey, 2022-01-25:
### Link

https://www.philips-hue.com/nl-nl/p/hue-white-and-color-ambiance-3-pack-centura-inbouwspot/8719514342866

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x001788010b557a35","nwkAddr":40426,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"929003045301_01","epList":[11,2>

### Comments

These GU10 spots are the same as Philips 9290019532 but have a different model number: 9290019531 and the zigbee model is coded with a suffix (929003045301_01)

I've added the following to zigbee-herdsman-converters/devices/philips.js to make it work:

`    {
        zigbeeModel: ['929003045301_01', '929003045301_02', '929003045301_03'],
        model: '9290019531',
        vendor: 'Philips',
        description: 'Hue White and Color Ambiance GU10' (Centura),
        meta: {turnsOffAtBrightness1: true},
        extend: hueExtend.light_onoff_brightness_colortemp_color(),
        ota: ota.zigbeeOTA,
    },`

Haven't tried an external converter as OTA doesn't seem supported that way.

rtfmjoey, 2022-01-26:
Here it is:
Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

Koenkk, 2022-01-27:
added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-27 by Koenkk