Koenkk/zigbee2mqtt #11026: [New device support]: Woox R7048 Smart Humidity & Temperature Sensor

chrisob, 2022-01-25:
### Link

https://wooxhome.com/woox-r7048-smart-humidity-temperature-sensor-p48

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xbc33acfffe26e60d","nwkAddr":49852,"manufId":4098,"manufName":"_TZ3000_rusu2vzb","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,10,1,1026,1029],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"�>�)\u0011","modelId":"TS0201","manufacturerName":"_TZ3000_rusu2vzb","stackVersion":0,"dateCode":""}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2313}},"msRelativeHumidity":{"attributes":{"measuredValue":3997}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1643149318626,"useImplicitCheckin":true}

### Comments

By default, it seems this device is mistakenly recognized as a Tuya TS0201 (I think due to using the same `modelId`?). Probably I could have left it and it would've worked fine, but the incorrect picture is bugging me ;)

I'll probably submit a PR for this soon

### External converter

```shell
{
	    zigbeeModel: ['TS0201'],
	    fingerprint: [{manufacturerName: '_TZ3000_rusu2vzb'}],
	    model: 'R7048',
	    vendor: 'Woox',
	    description: 'Smart Humidity & Temperature Sensor',
	    fromZigbee: [fz.battery, fz.temperature, fz.humidity],
	    toZigbee: [],
	    exposes: [e.battery(), e.temperature(), e.humidity(), e.battery_voltage()],
}
```

chrisob, 2022-01-30:
PR for docs: https://github.com/Koenkk/zigbee2mqtt.io/pull/1186

chrisob, 2022-01-30:
PR for device support: https://github.com/Koenkk/zigbee-herdsman-converters/pull/3778

tyloch, 2022-02-13:
It does not detect my R7048 device properly.
Instead, it is identified by z2m (current installed version: 1.23.0-1 )
as https://www.zigbee2mqtt.io/devices/TS0201.html#tuya-ts0201
[image]

closed 2022-03-30 by github-actions[bot]

brechtvhb, 2023-01-10:
Hi,

I bought 2 of these sensors but when I added them to Z2M they did not et recognized as woox R7048 but as Tuya TS0201.
Technically not a big issue as they report the same sensors but how could I make sure zigbee recognizes it as the correct sensor?

This info might be useful for debugging:
[image]

It took also about half an hour before it started reporting the battery status but maybe that's normal?