Koenkk/zigbee2mqtt #11029: AwoX 33957 pairs as 33951/33948

Crocmagnon, 2022-01-26:
### What happened?

I bought 3 identical ceiling lights (not bulbs but I guess they use the same controller) from EGLO.

Two of them pair as AwoX 33957, which is surprising at first but I read online that EGLO uses AwoX controllers in some of their products. This mapping doesn't bother me, since it allows me to controll the white temperature of the lights.

However, the third one paired as AwoX 33951/33948, which only supports brightness.

I seem to have a problem similar to these threads : https://github.com/Koenkk/zigbee2mqtt/issues?q=is%3Aissue+is%3Aclosed+33951

[image]

[image]

Context : I'm using Z2M with the HA addon

### What did you expect to happen?

I expected the pairing to map the device to AwoX 33957, which would allow me to control the white temperature.

### Zigbee2MQTT version

1.22.2-dev commit: f0f7662

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

Koenkk, 2022-01-26:
Can you provide the data/database.db entry of this device?

Crocmagnon, 2022-01-26:
I'd be glad to, how should I proceed?

Koenkk, 2022-01-26:
You can open the file with a text editor, then search for the line of the device (based on the ieee address) and  copy-paste it here.

Crocmagnon, 2022-01-26:
There you go:
```
{"id":63,"type":"Router","ieeeAddr":"0xa4c1382a02091afa","nwkAddr":7509,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":2,"hwVersion":0}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1643214441156,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2022-01-26:
I cannot find the issue, both entries are identical in terms of identification. Could you check if restarting the addon fixes this?

Crocmagnon, 2022-01-26:
Well yeah it did 🤔

closed 2022-01-26 by Crocmagnon

Timple, 2023-05-11:
For the record, bug is still present.
But workaround (restart addon) still works