Koenkk/zigbee2mqtt #11032: Moes TVR not supported but its on the supported list

artphdent, 2022-01-26:
### What happened?

purchased moes TVR which looks exactly like BRT-100-TRV  in the supported devices list but is identified as TS0601 which is Unsupported and doesn't expose anything

### What did you expect to happen?

To be ID'd as BRT-100-TVR

### How to reproduce it (minimal and precise)

delete and repair

### Zigbee2MQTT version

1.18.1

### Adapter firmware version

20211217

### Adapter

Electrolama zzh

### Debug log

Debug Received Zigbee message from '0x84fd27fffea5ae0f', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,230],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":1,"transid":236}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_b6wax7g0'

Koenkk, 2022-01-26:
You are using an old z2m version (1.18.1), updating to the latest will fix this.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2022-01-26 by Koenkk

artphdent, 2022-01-26:
Thank you I had just come to the same conclusion after seeing that the latest version was 1.22.1 just changed over the repository and got the new version thanks for the help