Koenkk/zigbee2mqtt #11033: Tuya Smart Air Box (AirBox04) mixes up CO2 and Formaldehyde data

mateon1, 2022-01-26:
### What happened?

Connected a Zigbee-based Tuya AirBox04 sensor to Zigbee2MQTT, the device was detected as a [TS0601_air_quality_sensor](https://www.zigbee2mqtt.io/devices/TS0601_air_quality_sensor.html), and started rapidly reporting sensor values, but the "formaldehyd" value is suspiciously high, and "co2" value is suspiciously low. Comparing the data from this device to a near-identical WiFi-based device, I can tell the VOC is wired up correctly, but co2 and formaldehyde were swapped.

Example message:
```
info MQTT publish: topic 'zigbee2mqtt/[REDACTED]', payload '{"co2":2,"formaldehyd":368,"humidity":42,"last_seen":"2022-01-26T18:19:12+01:00","linkquality":123,"temperature":19.9,"voc":1}'
```

### What did you expect to happen?

All reported values were correct.

### How to reproduce it (minimal and precise)

Pair an AirBox04 device to Zigbee2MQTT, observe logs.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20210120

### Adapter

Some CCxx62 device with zStack3x0

mateon1, 2022-01-26:
I looked through the debug messages, to me it looks like:
The `"dp":2` messages should be formaldehyde.
The `"dp":18` - temperature, which they already are.
The `"dp":19` - humidity, permille, but always a multiple of 10. Seem to be handled correctly.
The `"dp":21` - seem to be voc, and already handled correctly.
The `"dp":22` - match up with CO2 on the Wifi device, currently affects "formaldehyde" in the output.

bigsem89, 2022-02-05:
i've also got a problem with this device.
My coordinator is breaking due to too spamming

frfstu, 2022-02-06:
I'm use zigbee2mqtt version 1.22.0-2, coordinator zStack3x0 firmware 20211217
[image]
[image]
[image]
[image]
[image]
[image]
TS0601 _TZE200_ryfmq5rl Air quality sensor not only are the output parameters of carbon dioxide and formaldehyde sensors mixed up, but this device sends so many data reports every second that my server can’t cope with their processing and freezes, doesn’t display (can’t load) the history of indicators, because of this, they work every other time  automation based on this device's data, spamming logs unnecessarily filling disk space and constantly writing and rewriting data to the disk has already disabled two microSD cards and one solid state drive.  I have previously written about this issue.
[image]

debnetjm, 2022-03-06:
I can confirm this issue as well. 
[image]

device is reporting values few times per second
`Zigbee2MQTT:info  2022-03-06 11:52:35: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":1,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1384acfbd1b52","manufacturerID":4417,"manufacturerName":"_TZE200_ryfmq5rl","model":"TS0601_air_quality_sensor","networkAddress":62755,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":156,"formaldehyd":358,"humidity":28,"last_seen":"2022-03-06T11:52:35+01:00","linkquality":131,"temperature":23.1,"voc":0}'
Zigbee2MQTT:info  2022-03-06 11:52:35: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":1,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1384acfbd1b52","manufacturerID":4417,"manufacturerName":"_TZE200_ryfmq5rl","model":"TS0601_air_quality_sensor","networkAddress":62755,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":138,"formaldehyd":358,"humidity":28,"last_seen":"2022-03-06T11:52:35+01:00","linkquality":123,"temperature":23,"voc":0}'
Zigbee2MQTT:info  2022-03-06 11:52:36: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":1,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1384acfbd1b52","manufacturerID":4417,"manufacturerName":"_TZE200_ryfmq5rl","model":"TS0601_air_quality_sensor","networkAddress":62755,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":342,"formaldehyd":358,"humidity":28,"last_seen":"2022-03-06T11:52:36+01:00","linkquality":123,"temperature":23,"voc":1}'
Zigbee2MQTT:info  2022-03-06 11:52:36: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":2,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1384acfbd1b52","manufacturerID":4417,"manufacturerName":"_TZE200_ryfmq5rl","model":"TS0601_air_quality_sensor","networkAddress":62755,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":131,"formaldehyd":358,"humidity":28,"last_seen":"2022-03-06T11:52:36+01:00","linkquality":123,"temperature":23,"voc":1}'
Zigbee2MQTT:info  2022-03-06 11:52:36: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":2,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1384acfbd1b52","manufacturerID":4417,"manufacturerName":"_TZE200_ryfmq5rl","model":"TS0601_air_quality_sensor","networkAddress":62755,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":140,"formaldehyd":368,"humidity":28,"last_seen":"2022-03-06T11:52:36+01:00","linkquality":131,"temperature":23,"voc":1}'
Zigbee2MQTT:info  2022-03-06 11:52:37: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":2,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1384acfbd1b52","manufacturerID":4417,"manufacturerName":"_TZE200_ryfmq5rl","model":"TS0601_air_quality_sensor","networkAddress":62755,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":656,"formaldehyd":368,"humidity":28,"last_seen":"2022-03-06T11:52:37+01:00","linkquality":127,"temperature":23,"voc":1}'
Zigbee2MQTT:info  2022-03-06 11:52:37: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":2,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1384acfbd1b52","manufacturerID":4417,"manufacturerName":"_TZE200_ryfmq5rl","model":"TS0601_air_quality_sensor","networkAddress":62755,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":142,"formaldehyd":368,"humidity":28,"last_seen":"2022-03-06T11:52:37+01:00","linkquality":131,"temperature":23.1,"voc":1}'
Zigbee2MQTT:info  2022-03-06 11:52:37: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":2,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1384acfbd1b52","manufacturerID":4417,"manufacturerName":"_TZE200_ryfmq5rl","model":"TS0601_air_quality_sensor","networkAddress":62755,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":31,"formaldehyd":368,"humidity":28,"last_seen":"2022-03-06T11:52:37+01:00","linkquality":127,"temperature":23.1,"voc":1}'
Zigbee2MQTT:info  2022-03-06 11:52:37: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":2,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1384acfbd1b52","manufacturerID":4417,"manufacturerName":"_TZE200_ryfmq5rl","model":"TS0601_air_quality_sensor","networkAddress":62755,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":120,"formaldehyd":368,"humidity":28,"last_seen":"2022-03-06T11:52:37+01:00","linkquality":127,"temperature":23.1,"voc":0}'
Zigbee2MQTT:info  2022-03-06 11:52:37: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":1,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1 [... cut ...]
Zigbee2MQTT:info  2022-03-06 11:52:39: MQTT publish: topic 'zigbee2mqtt/Czujnik CO2 75', payload '{"co2":1,"device":{"applicationVersion":64,"dateCode":"","friendlyName":"Czujnik CO2 75","hardwareVersion":1,"ieeeAddr":"0xa4c1384acfbd1b52","manufacturerID":4417,"manufacturerName":"_TZE200_ryfmq5rl","model":"TS0601_air_quality_sensor","networkAddress":62755,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":141,"formaldehyd":358,"humidity":28,"last_seen":"2022-03-06T11:52:39+01:00","linkquality":131,"temperature":23.1,"voc":0}'`

Koenkk, 2022-04-01:
Swapping should already have been fixed in z2m 1.25.0 (https://github.com/Koenkk/zigbee-herdsman-converters/pull/3982)

debnetjm, 2022-04-02:
Seems it is not (maybe not for all devices ?)

latest-dev: Zigbee2MQTT version 1.25.0-dev commit: [ad24c4f]

{
    **"co2": 2,**
    "device": {
        "applicationVersion": 64,
        "dateCode": "",
        "friendlyName": "Czujnik CO2 75",
        "hardwareVersion": 1,
        "ieeeAddr": "0xa4c1384acfbd1b52",
        "manufacturerID": 4417,
        "manufacturerName": "_TZE200_ryfmq5rl",
        "model": "TS0601_air_quality_sensor",
        "networkAddress": 20635,
        "powerSource": "Mains (single phase)",
        "stackVersion": 0,
        "type": "Router",
        "zclVersion": 3
    },
    **"formaldehyd": 368,**
    "humidity": 34,
    "last_seen": "2022-04-02T12:22:15+02:00",
    "linkquality": 105,
    "temperature": 19.2,
    "voc": 1,
    "elapsed": 414
}

frfstu, 2022-04-02:
This is embarrassing!  The prospect of reconfiguring half of the server
configuration, additional installation of core library components, flashing
the stick (which, as the list of bug reports shows, also does not work
quite correctly).  And how the system will behave with these innovations is
not yet clear!  At the moment, I don’t have the opportunity (mainly related
to redundancy) to somehow contribute to the development and test new
versions of the software, no matter how trite it sounds, I renamed their
names in the display form  solves the problem itself with the correctness
of sending and the frequency of updating values, but purely visually
acceptable and correct.

Koenkk, 2022-04-02:
@debnetjm also fixed it for the `_TZE200_ryfmq5rl` now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

skandragon, 2022-04-02:
> 

I don't get what you are saying here.  It's just the zigbee2mqtt that needs updated here, not the firmware on the zigbee devices.

frfstu, 2022-04-02:
Jokers .. I read all this yesterday (crossed out today) as I imagined that
I would have to re-configure almost everything (including some
automations), I was just indescribably delighted!

debnetjm, 2022-04-04:
@Koenkk something's changed, co2 seems to be OK I am not sure about formaldehyd

formaldehyd seems to be too high and too similar to co2 reading (see graph)

[image]

using latest-dev

{
    "co2": 358,
    "device": {
        "applicationVersion": 64,
        "dateCode": "",
        "friendlyName": "Czujnik CO2 75",
        "hardwareVersion": 1,
        "ieeeAddr": "0xa4c1384acfbd1b52",
        "manufacturerID": 4417,
        "manufacturerName": "_TZE200_ryfmq5rl",
        "model": "TS0601_air_quality_sensor",
        "networkAddress": 20635,
        "powerSource": "Mains (single phase)",
        "stackVersion": 0,
        "type": "Router",
        "zclVersion": 3
    },
    "elapsed": 1109,
    "formaldehyd": 368,
    "humidity": 37,
    "last_seen": "2022-04-04T11:51:17+02:00",
    "linkquality": 102,
    "pm25": 1,
    "temperature": 19.1,
    "voc": 1
}

kevincaradant, 2022-04-05:
Hi

I just fall on this thread but maybe you should remove the device completely and add it again because I just connected mine yesterday, and I got value close to 0 until 10 for the formaldehyd attribute so I don't understand yours :)
[image]

Koenkk, 2022-04-05:
@debnetjm could be that yours uses different data point ids, to figure them out follow https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

debnetjm, 2022-04-06:
@Koenkk they seems to be OK but DPs are not fixed for Smart Air Box in tuya.js

    // tuya smart air box
    tuyaSabCO2: 2,
    tuyaSabTemp: 18,
    tuyaSabHumidity: 19,
    tuyaSabVOC: 21,
    tuyaSabFormaldehyd: 22,

DP 18: temperature 2022-04-06 09:55:13	Report	Current Temperature	22.10℃	device itself

DP 2: CH20 (formaldehyd) 2022-04-06 09:53:34	Report	CH2O value	0.01mg/m³	device itself

DP 19: humidity 2022-04-06 09:54:00	Report	Current Humidity	39.00%	device itself

DP 21: VOC 2022-04-06 09:56:08	Report	VOC value	0.10ppm	device itself	

DP 22: CO2 2022-04-06 09:56:00	Report	CO2 value	368ppm	device itself

My device Smart Air Box (AirBox04) does not support PM25, does not show in tuya app as well.
Seems my airbox04 was detected as Smart Air House Keeper which probably support pm25 ?

Koenkk, 2022-04-06:
@debnetjm are correct values reported when paired through the TuYa app and can you provide the data/database.db entry of your device?

debnetjm, 2022-04-06:
[image]

`{"id":42,"type":"Router","ieeeAddr":"0xa4c1384acfbd1b52","nwkAddr":43621,"manufId":4417,"manufName":"_TZE200_ryfmq5rl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\"��)f\"��)\u0012","65506":32,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_ryfmq5rl","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1649266861521,"defaultSendRequestWhen":"immediate"}`

debnetjm, 2022-04-25:
@Koenkk any update? Do you need any other information ?

Koenkk, 2022-04-25:
@debnetjm pushed another fix, can you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Mirarkitty, 2022-04-26:
Just got my TS0601 air quality thing with pm25 today, came here for this bug. I guess it's fixed now but not in the released branch?

In latest master:
{"co2":397,"formaldehyd":396,"humidity":41.8,"linkquality":153,"pm25":6,"temperature":25.4,"voc":52}

Previously co2 was 5-7... I'll see if I can get around to test the dev branch

debnetjm, 2022-04-26:
@Koenkk indeed seems to be much better

two things:
1)
formaldehyd - missing unit name (mg/m³)
voc - I think should be ppm not ppb

2)
scale for formaldehyd and voc, I think itshould be 0.01-0.02 (formaldehyd) and 0.0-0.1 for voc as per screenshot from tuya application and definition from tuya platform ?

ch2o_value | Integer | { "unit": "mg/m³", "min": 0, "max": 1000, "scale": 2, "step": 1 }
voc_value | Integer | { "unit": "ppm", "min": 0, "max": 999, "scale": 1, "step": 1 }

`
Zigbee2MQTT:info  2022-04-26 14:27:17: MQTT publish: topic 'zigbee2mqtt/0xa4c1384acfbd1b52', payload '{"co2":368,"formaldehyd":2,"humidity":40,"last_seen":"2022-04-26T14:27:16+02:00","linkquality":138,"temperature":24.3,"voc":1}'
Zigbee2MQTT:info  2022-04-26 14:27:32: MQTT publish: topic 'zigbee2mqtt/0xa4c1384acfbd1b52', payload '{"co2":368,"formaldehyd":2,"humidity":39,"last_seen":"2022-04-26T14:27:31+02:00","linkquality":149,"temperature":24.3,"voc":1}'
Zigbee2MQTT:info  2022-04-26 14:28:02: MQTT publish: topic 'zigbee2mqtt/0xa4c1384acfbd1b52', payload '{"co2":368,"formaldehyd":2,"humidity":39,"last_seen":"2022-04-26T14:28:01+02:00","linkquality":149,"temperature":24.3,"voc":1}'
Zigbee2MQTT:info  2022-04-26 14:28:09: MQTT publish: topic 'zigbee2mqtt/0xa4c1384acfbd1b52', payload '{"co2":368,"formaldehyd":2,"humidity":39,"last_seen":"2022-04-26T14:28:08+02:00","linkquality":153,"temperature":24.3,"voc":1}'
Zigbee2MQTT:info  2022-04-26 14:28:16: MQTT publish: topic 'zigbee2mqtt/0xa4c1384acfbd1b52', payload '{"co2":358,"formaldehyd":1,"humidity":39,"last_seen":"2022-04-26T14:28:15+02:00","linkquality":149,"temperature":24.3,"voc":0}'
Zigbee2MQTT:info  2022-04-26 14:28:46: MQTT publish: topic 'zigbee2mqtt/0xa4c1384acfbd1b52', payload '{"co2":358,"formaldehyd":1,"humidity":39,"last_seen":"2022-04-26T14:28:45+02:00","linkquality":123,"temperature":24.3,"voc":0}'
`

`{
    "last_seen": "2022-04-26T14:17:57+02:00",
    "linkquality": 149,
    "co2": 358,
    "formaldehyd": 1,
    "humidity": 40,
    "temperature": 24.1,
    "voc": 1
}`

[image]

Mirarkitty, 2022-04-26:
I'm seeing a lot of trash pm25 values, should those be filtered on zigbee2mqtt level or later? 

{ "co2": 378, "formaldehyd": 396, "humidity": 34.5, "linkquality": 153, "pm25": 43691, "temperature": 25.6, "voc": 37 }

43691 is 0xaaab, basically the bitfield "1010101010...". Seeing 43692 too (0xaaac). Normal pm25 for the sensor is around 5.

Koenkk, 2022-04-27:
@debnetjm so basically ch20 should be divided by 100 and voc by 10?

Koenkk, 2022-04-28:
@debnetjm fixed now

@Mirarkitty `0xaaab` `pm25` will be ignored now.

@Aaike can you try with the latest dev first? 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ravimohta, 2022-05-11:
> @debnetjm fixed now
> 
> @Mirarkitty `0xaaab` `pm25` will be ignored now.
> 
> @Aaike can you try with the latest dev first? 
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Hi,

The erroneous pm25 value is 43692, which is 0xaaac (and not 0xaaab).

I have 4 of Tuya Air Quality monitor sensors and I switched to using the Edge version of the Z2m yesterday night. But I can still see the high pm25 coming.

Could you please check and fix this?

Regards

ravimohta, 2022-05-11:
[[image]

Koenkk, 2022-05-11:
@ravimohta fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Mirarkitty, 2022-05-11:
> The erroneous pm25 value is 43692, which is 0xaaac (and not 0xaaab).

It's both, as what I wrote above:

> 43691 is 0xaaab, basically the bitfield "1010101010...". Seeing 43692 too (0xaaac).

ravimohta, 2022-05-11:
> > The erroneous pm25 value is 43692, which is 0xaaac (and not 0xaaab).
> 
> It's both, as what I wrote above:
> 
> > 43691 is 0xaaab, basically the bitfield "1010101010...". Seeing 43692 too (0xaaac). 
> 
> Possibly other values too, I get random spikes, but I haven't investigated more. My pm25 sensor seems to only give noise near zero otherwise, I can't get it to read any dust. 
> 
> I opened it and the pm25 sensor is jammed shut towards the bottom and top of the box (???), could be the reason. Will have to print a new case or run it open, I guess. Also co2, formaldehyde and voc are weirdly linked. None of this is the fault of zigbee2mqtt, though, I'm sure.
> 
> [image]
> 

Sorry 🙏 my fault.. I agree, it is both these value.

Also please let share if running it open makes any difference or not.

Thanks

Koenkk, 2022-05-11:
Ignored both values now (`0xaaac` and `0xaaab`)

ravimohta, 2022-05-11:
Here is the min/max parameters, as mentioned in the paper that came with the sensor
[image]

ravimohta, 2022-05-11:
I have reinstalled the Z2m Edge on home assistant and I am still seeing the high values every now and then.

How do I check if I have got the changes you applied via z2m Edge in home assistant?

Mirarkitty, 2022-05-12:
To make it less spammy I recommend debounce or set cache flags to false. https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#mqtt-behaviour

I burned a paper next to mine now. Zero difference on the pm25, still same noise between 3 and 10 units.

kevincaradant, 2022-05-17:
Hi

I'm not sure it will be enought. Because I'm agree in Z2M and HA we will have only few messages. 
But I saw that my CC1352-P2 has many issues few days after and I think it's because the Tuya device will spam and 'kill' after few days the stability of the Zigbee coordinator.

Am I right to think this ? 

Since I disconnected the device, I didn't have any issues after that :(

Thank you for your advice on this

Mirarkitty, 2022-05-18:
It could very well be. It's sending 5 messages every 5 seconds or so. I don't seem to have issues with my CC2652 luckily. It's probably outside the control of zigbee2mqtt since it doesn't seem to have a parameter for how often it should send.

Since I don't get **__any__** useful data from the device I will probably disconnect it anyway and I really can't recommend it. Maybe I can use it as a temp/hum sensor...

Mirarkitty, 2022-05-26:
If anyone got any useful data out of the device I'd like to know :D

ravimohta, 2022-05-27:
I still occasionally see pm2.5 values greater than 1000 (more in the range of around 40k or so). Since filtering the pm2.5 values, they have become lot less. And every day or two I run a sql query on the HA database and delete the records with pm2.5 values greater than 1000.

This has helped a lot in seeing the actual values of pm2.5

Next I played around with the debounce setting and tried different numbers (and I have 4 of these Tuya Air Quality devices) .. setting to 1 or 2 didnt make much of a different, so I tried 10 for a few days. Meaning I set the debounce on each of my Tuya air quality devices to 10 and let it run for 3-4 days to see what difference it made. After this period i realised that value of 10 really made the device useless .. the measured values of pm2.5, co2, voc, formaldehyde etc would change once in a few hours .. which made them useless really.

Eventually, after some more trial and error I have settled for a value of 5 or 6 as that slows down the refresh of readings from these devices to roughly once every 30 seconds or so, and I am happy with that.

Hope that helps.

Mirarkitty, 2022-05-30:
How do your graphs look like? Mine are incredibly stupid:

[image]

This is definitely not a problem with z2m. (I turned off the cache myself, so I only get values when they come in and not all of them when any value comes in.)

ravimohta, 2022-05-30:
The values I am getting now looks OK.

ravimohta, 2022-05-30:
The values I am getting now seems ok. Please eee screenshots

ravimohta, 2022-05-30:
[image]

ravimohta, 2022-05-30:
[image]

ravimohta, 2022-05-30:
[image]

ravimohta, 2022-05-30:
[image]

ravimohta, 2022-05-30:
[image]

ravimohta, 2022-05-30:
[image]

ravimohta, 2022-05-30:
My only issue now is that the formaldehyde, voc and co2 seem to be linked, or very similar all the times.

buddy31130, 2022-05-30:
mine are OK now
[image]
Sorry, I don't use temperature nor humidity

Mirarkitty, 2022-05-31:
> My only issue now is that the formaldehyde, voc and co2 seem to be linked, or very similar all the times.

Yes, this. They seem to be based on the same measurement, but with different scales and baselines. I don't think the firmware does what it should, because they have different sensors. pm25 just gives random spikes for me, I should put it outside the box and see if it helps...

buddy31130, 2022-06-02:
After updating, the debounce does not seem to work anymore. It generates tons of data... About every 1-4 sec with debouce set to any value

aredridel, 2022-06-04:
I get seemingly okay data from the pm25 sensor if I delete anything over 40k; not just the filtering added recently.

[image]

closed 2022-07-13 by github-actions[bot]

frfstu, 2022-10-11:
I apologize for the inaccurate translation, but I think the general meaning
is clear.  I am now suing the state authorities of the Russian Federation
(I have a court date on April 13, 2022), whose officials fabricate medical
and other documents related to the accident that happened to me, exposing
it as a chronic disease, hinting at the fact that it is hereditary, thereby
depriving me of all the prescribed  the law of benefits, benefits, care,
treatment, because of this I cannot receive compensation from the state,
the payment of insurance, and I am paid a minimum pension which is barely
enough to buy food for a month (and by law, multi-million compensation must
be paid for an industrial injury).