Koenkk/zigbee2mqtt #11034: Receiving errors publishing to some devices

Coder2012, 2022-01-26:
### What happened?

Every now and then I get errors in the logs that a publish has an error whilst other devices continue to operate OK. Also sometimes I see the publish in the logs, but device doesn't respond for a second or more. Or they stack up messages then get hit with lots one after the other.

### What did you expect to happen?

Devices respond to a publish without error

### How to reproduce it (minimal and precise)

This is specific to my setup, using Ikea Trafri and Innr bulbs.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

v2.7.1 from 20210120

### Adapter

Slaesh - CC2652RB

### Debug log

```
Zigbee2MQTT:error 2022-01-26 17:09:06: Publish 'set' 'state' to 'study-light' failed: 'Error: Command 0x00158d0003101219/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
Zigbee2MQTT:info  2022-01-26 17:09:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'study-light' failed: 'Error: Command 0x00158d0003101219/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"study-light"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:info  2022-01-26 17:12:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'front-light' failed: 'Error: Command 0xccccccfffe887b4d/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"front-light"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2022-01-26 17:12:04: Publish 'set' 'state' to 'rear-light' failed: 'Error: Command 0xccccccfffebe8b76/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
Zigbee2MQTT:info  2022-01-26 17:18:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"front-light","ieee_address":"0xccccccfffe887b4d"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-01-26 17:18:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"front-light"},"type":"device_announced"}'
Zigbee2MQTT:error 2022-01-26 17:18:59: Failed to read state of 'front-light' after reconnect (Read 0xccccccfffe887b4d/1 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))

```

Koenkk, 2022-01-26:
`MAC channel access failure' (225))` means wireless interference. https://www.zigbee2mqtt.io/guide/faq/#common-error-codes

Coder2012, 2022-01-26:
Thank you for finding the cause of my problem. I have changed the channel and rep-paired my devices. I will see how this goes and report back after one day of operation.

Coder2012, 2022-01-27:
@Koenkk I think this has fixed my mac error problem, but it still delays messages. This is something I've been fighting with for months. I even got a new Slaesh stick in the hope it would fix it, any ideas on things I could try?

Koenkk, 2022-01-27:
@Coder2012 do you get any errors on the delays?

Coder2012, 2022-01-28:
Not that I can tell, what does happen though is that zigbee restarts frequently "due to changes" as I run it using nodemon, because when it used to crash I wanted it to come up again.

Coder2012, 2022-02-01:
I will close this ticket, as my original problem is solved. In fact is been running great for a few days much happier now. Thank you @Koenkk

closed 2022-02-01 by Coder2012