Koenkk/zigbee2mqtt #11036: Too many published Siren variables not working

kiat-huang, 2022-01-26:
### What happened?

Siren - https://www.zigbee2mqtt.io/devices/HS2WD-E.html
Many of the published variables either have no effect or stop the Siren from working (so have to be avoided).

### What did you expect to happen?

I expected these published variables to work and.....
     to not prevent the Siren from sounding off:
         mode (enum): police, police_panic, fire_panic, emergency_panic
    not to have no effect whatsoever (there is only one kind of alarm, with no strobe)
         mode (enum): burglar, fire, emergency
         strobe_level (enum): Intensity of the strobe. Allowed values: low, medium, high, very_high
         strobe (binary): Turn on/off the strobe (light) during warning. Allowed values: true or false
         strobe_duty_cycle (numeric): Length of the flash cycle
         duration (numeric): Duration in seconds of the alarm

### How to reproduce it (minimal and precise)

1. Install the Siren via z2m.
2. Activate the Siren using an MQTT publish client or via HA (script or switch template)
   e.g.  mosquitto_pub -h 192.168.1.250 -u REDACTED -P REDACTED -t 'zigbee2mqtt/Siren Hall/set' -m '{"warning": {"mode": "fire", "strobe": true, "duration": 300, "strobe_level": "low" }}'
3. Siren sounds off, as expected, but note the strobe does not com on, nor is the duration respected (it does not stop)
4. Deactivate the siren
  e.g.  mosquitto_pub -h 192.168.1.250 -u REDACTED -P REDACTED -t 'zigbee2mqtt/Siren Hall/set' -m '{"warning": {"mode": "stop" }}'
5. The siren stops sounding off
6. Try some other options
   e.g.  mosquitto_pub -h 192.168.1.250 -u REDACTED -P REDACTED -t 'zigbee2mqtt/Siren Hall/set' -m '{"warning": {"mode": "burglar", "strobe": true, "duration": 10, "strobe_level": "high" }}'
7. Note that the Siren sounds off but none of the variables are respected
8. Try one of the modes from: police, police_panic, fire_panic, emergency_panic
   e.g. mosquitto_pub -h 192.168.1.250 -u REDACTED -P REDACTED -t 'zigbee2mqtt/Siren Hall/set' -m '{"warning": {"mode": "police"}}'
9. The Siren does not work.

### Zigbee2MQTT version

1.22.2-1

### Adapter firmware version

20211217

### Adapter

https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/

### Debug log

{kiat@newton:~/sshfs/hass/config/zigbee2mqtt ?} tail -f log/2022-01-26.18-03-47/log.txt | grep -i siren
debug 2022-01-26 18:07:55: Received MQTT message on 'zigbee2mqtt/Siren Hall/set' with data '{"warning": {"mode": "stop"}}'
debug 2022-01-26 18:07:55: Publishing 'set' 'warning' to 'Siren Hall'
debug 2022-01-26 18:08:46: Received Zigbee message from 'Siren Hall', type 'attributeReport', cluster 'ssIasWd', data '{"61440":0}' from endpoint 1 with groupID 0
debug 2022-01-26 18:08:46: Received MQTT message on 'zigbee2mqtt/Siren Hall/set' with data '{"warning": {"mode": "fire", "strobe": true, "duration": 1, "strobe_level": "low" }}'
debug 2022-01-26 18:08:46: Publishing 'set' 'warning' to 'Siren Hall'
debug 2022-01-26 18:08:46: Received Zigbee message from 'Siren Hall', type 'attributeReport', cluster 'ssIasWd', data '{"61440":3}' from endpoint 1 with groupID 0
debug 2022-01-26 18:08:48: Received MQTT message on 'zigbee2mqtt/Siren Hall/set' with data '{"warning": {"mode": "stop"}}'
debug 2022-01-26 18:08:48: Publishing 'set' 'warning' to 'Siren Hall'
debug 2022-01-26 18:08:48: Received Zigbee message from 'Siren Hall', type 'attributeReport', cluster 'ssIasWd', data '{"61440":2}' from endpoint 1 with groupID 0
debug 2022-01-26 18:08:50: Received MQTT message on 'zigbee2mqtt/Siren Hall/set' with data '{"warning": {"mode": "police", "strobe": true, "duration": 1, "strobe_level": "low" }}'
debug 2022-01-26 18:08:50: Publishing 'set' 'warning' to 'Siren Hall'
debug 2022-01-26 18:08:53: Received MQTT message on 'zigbee2mqtt/Siren Hall/set' with data '{"warning": {"mode": "stop"}}'
debug 2022-01-26 18:08:53: Publishing 'set' 'warning' to 'Siren Hall'

==============================================================================================

https://REDACTED:8443/hassio/addon/45df7312_zigbee2mqtt/logs

Zigbee2MQTT:info  2022-01-26 18:10:05: MQTT publish: topic 'zigbee2mqtt/Siren Hall', payload '{"battery":100,"linkquality":40,"voltage":4100}'
2022-01-26T18:10:05.144Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-01-26T18:10:05.145Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-01-26T18:10:05.146Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-01-26T18:10:05.146Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-01-26T18:10:05.147Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-26T18:10:05.149Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,54,240]
2022-01-26T18:10:05.149Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,54,240]
2022-01-26T18:10:05.149Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,54] - 240
2022-01-26T18:10:05.150Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":54}
2022-01-26T18:10:05.150Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-26T18:10:07.467Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,109,27,1,252,234,229]
2022-01-26T18:10:07.467Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,109,27,1,252,234,229]
2022-01-26T18:10:07.467Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [109,27,1,252,234] - 229
2022-01-26T18:10:07.468Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":7021,"relaycount":1,"relaylist":[60156]}
2022-01-26T18:10:07.468Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-26T18:10:07.522Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,4,11,109,27,1,1,0,69,0,184,233,187,0,0,8,8,10,10,5,5,33,141,95,252,234,28,246]
2022-01-26T18:10:07.522Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,4,11,109,27,1,1,0,69,0,184,233,187,0,0,8,8,10,10,5,5,33,141,95,252,234,28,246]
2022-01-26T18:10:07.523Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,109,27,1,1,0,69,0,184,233,187,0,0,8,8,10,10,5,5,33,141,95,252,234,28] - 246
2022-01-26T18:10:07.523Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":7021,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":69,"securityuse":0,"timestamp":12315064,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[8,10,10,5,5,33,141,95]}}
2022-01-26T18:10:07.528Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":10,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1285,"dataType":33,"attrData":24461}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":7021,"endpoint":1,"linkquality":69,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'

writepatrick, 2022-01-28:
Hi kiat-huang,
I have experience an similar issue here: https://github.com/Koenkk/zigbee2mqtt/issues/10965
In my care an off-brand Heiman sirene was picked up as TuYa TS2016 (instead of Heiman)
My sirene is only acting as a switch (on/off), instead of the function at least the TuYa is having (alarm volume)

I have ordered an Heiman HS2WD-E right now, but fear that I will run into the same problems like you are.
I am not sure where these publishing variables are even coming from. (Who confirmed this, since it is not included in the heiman.js)

closed 2022-03-07 by github-actions[bot]

jhonny001, 2022-04-08:
Up. Having the same issues here...