Koenkk/zigbee2mqtt #11049: OTA Update lead to weird firmware version (Danfoss 014G2461)

rubenkr, 2022-01-27:
### What happened?

- OTA update conducted for Danfoss Zigbee devices
- Device updated to 01.18.0008 01.18 work with external sensors
- Devices updated to AAAABBBBCCCCDDDD do **not** work with external sensor
[image]

### What did you expect to happen?

Update to firmware 01.18.0008 01.18 and or repair process 

### Zigbee2MQTT version

1.22.2-dev commit: 1a2943d

### Adapter firmware version

20211217

### Adapter

CC2652RB

lux73, 2022-01-27:
I had to repair my Danfoss Ally after updating to show the correct Firmware

rubenkr, 2022-01-27:
@lux73 Michael how did you repair it? Could you please elaborate?

lux73, 2022-01-27:
i simply mean delete & re-pair with z2m - only after this my System shows the updatet Firmware. without re-pair it stucks with the old Firmware showing

artemd93, 2022-02-01:
It also shows battery level > 100%

stefangehn, 2022-02-04:
I ran into a similar (related?) issue when updating one of my thermostats. While still on 1.22.2 the update got stuck at 100% (#10666, fixed in 1.23.0).
I now upgraded to 1.23.0 and manually cleaned the update state of `bathroom_thermostat` in `state.json`, however the firmware version and date do not fit to eachother.
For comparison, `livingroom_thermostat` finished its OTA update to version 01.18 and also got its firmware date updated:
[image]

I don't know if it makes sense but if the "reconfigure" action would also fetch the current firmware date and version to fix this then I wouldn't have to re-pair the device to get both fields updated on the z2m side. OTOH unless I see actual problems I will probably just leave things as they are now because the thermostat works great.

Tenkc, 2022-02-10:
I experienced the same behaviour with my Popp [701721](https://www.zigbee2mqtt.io/devices/701721.html#popp-701721), which is based on a Danfoss Ally. 

Updated two Popp's, one showed the correct firmware, the other AAAABBBBCCCCDDDD and a battery status of 127.5%. Retried repairing a couple of times with the TRV in place without any luck. Then I removed the TRV, restarted Z2M and repaired nearer to a router. Coincidentally this worked, now it is showing the right firmware and battery status again.

stefangehn, 2022-02-11:
Same issue happened for me now that I upgraded all 4 thermostats I have.

[image]

The first two were updated with z2m 1.22.2, the last two with 1.23.0.
So regardless of the z2m version it looks like fetching the new values for build date or version occasionally fails. I doubt this is of much use for fixing in z2m though.

A debug log of a working and a failing firmware update might have helped (I assume there'd be a difference between the two cases in the zigbee payloads logged) but I did not think about that before trying updating with 1.23.0.

magicus, 2022-03-03:
I have run into the very same problem. :-( I recently bought Allys for all my radiators, and a few of them updated alright. And then suddenly two got this "AAAABBBBCCCCDDDD" version. I think I might have updated z2m in Home Assistant between the point it was working and when it wasn't, indicating that this is indeed a recent regression. I am currently using version 1.24.0-1 of the hass add-on.

@lux73 Did you mean that you could fix the problem by factory reset and re-pair of the TRV?

lux73, 2022-03-03:
Correct, after Factory Reset & re-pair both of my Allys shown correct Firmware Version

ascheucher, 2022-03-18:
Maybe noteworthy...

My whole repair and factory reset workflow for the Danfoss Ally TRVs. It's faster to note it down and look up then to forget and come up with it again.

* copy the IEEE Address and the friendly name of the affected devices into a list
* force remove the Danfoss Ally TRV
* stop ZigBee2MQTT
* Reset the affected TRVs (only half of them showed the wrong firmware versioen) with following steps:
  * press button
  * remove battery
  * hold button
  * add battery
  * still hold button, till alle the symbols on the LCD flash and the motor takes away all the pressure from the valve. you can hear the water... then release
* long press the button, till it starts to close the valve again
* start ZigBee2MQTT again
* short press to activate the display
* short press again to initiate ZigBee pairing.
* check device page and wait 
* rename the devices (check *Update Home Assistant entity ID*) to the former names from the list.

That are quite some steps to get in the correct order. Also to leave them as note to myself for the next time.

magicus, 2022-03-18:
@ascheucher And you managed to get a proper firmware with correctly reported version by doing this dance? I've tried doing parts of it (basically everything but restarting zigbee2mqtt), and it did not help; my updated Allys still show the AAAA... firmware version. (I've updated just a few; I dare not touch the rest.)

ascheucher, 2022-03-18:
@magicus Yes, this worked for all of them. I have got 10 of them. After the update half of them showed AAABBB... whatever. One showed a wrong older version (1.12 maybe?).

After going through this annoying dance, all of them showed the correct version. I also did a firmware update of my Sonoff ZigBee 3.0 dongle. Which wasn't easy first (wrong tutorials for Mac and Windows), but following a youtube video doing it on Linux worked like a charm. I had some paring issues with TuYa wall switches. Turned out it had been outdated CR2032 batteries... :D

magicus, 2022-03-18:
@ascheucher Thank you sir! With the restart of z2m, it worked for me as well. A relief. (But still an annoying bug; hope it gets fixed)

mpuff, 2022-03-19:
I gave the same on my popp 701721

surrealo, 2022-04-01:
@ascheucher  Thanks for those steps, i was having the exact same problem and managed to have the firmware show up properly with your steps.

robertalexa, 2022-04-05:
@Tenkc and @mpuff Would one of you be kind to send me a screenshot of the Popp TRV on the OTA screen? Basically like this:
[image]

Currently trying to figure out if the Hive TRV (which has a custom firmware version) can be "force" flashed with the official Danfoss Ally ota file.

Am i right in thinking that you did an OTA to 1.18 by just pressing the button with no extra shenanigans?

(I am basically trying to understand if Popp describes itself as Popp but works on the Danfoss firmware from the factory, unlike Hive TRV)

Thank you both

Tenkc, 2022-04-06:
@robertalexa Here you go!

[image]

Yes, I did an OTA to 1.18 just by pressing the button.

robertalexa, 2022-04-06:
Thanks a lot

magicus, 2022-05-18:
This is still a problem, even if the bot detects no activity. :-( Can someone with the proper permissions remove the stale label?

closed 2022-05-26 by github-actions[bot]

javnik36, 2022-07-31:
Bought recently 2 of these Danfoss TRVs and updated one right away. Got the same issue with weird software version. Below log when z2m v1.25.2 finished updating it (only "info" lvl unfortunately, but maybe it will help somehow. I will try to remember to change to debug when updating 2nd TRV 😄). After "repair process" z2m show correct sf version (1.18).

`info  2022-07-21 19:47:33: Update of '0x0c4314fffe5de312' at 100.00%
info  2022-07-21 19:47:33: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe5de312', payload '{"adaptation_run_settings":true,"adaptation_run_status":"none","algorithm_scale_factor":1,"battery":82,"day_of_week":"thursday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","linkquality":196,"load_balancing_enable":true,"load_estimate":-33,"load_room_mean":-8000,"local_temperature":28.14,"mounted_mode_active":true,"mounted_mode_control":true,"occupied_heating_setpoint":21,"pi_heating_demand":0,"preheat_status":null,"programming_operation_mode":null,"radiator_covered":false,"regulation_setpoint_offset":null,"running_state":"idle","setpoint_change_source":null,"system_mode":"heat","thermostat_vertical_orientation":false,"trigger_time":660,"update":{"progress":100,"remaining":16,"state":"updating"},"update_available":null,"viewing_direction":true,"window_open_external":false,"window_open_feature":true,"window_open_internal":"closed"}'
info  2022-07-21 19:48:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe5de312","ieee_address":"0x0c4314fffe5de312"},"type":"device_announce"}'
info  2022-07-21 19:48:00: Finished update of '0x0c4314fffe5de312'
info  2022-07-21 19:48:00: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe5de312', payload '{"adaptation_run_settings":true,"adaptation_run_status":"none","algorithm_scale_factor":1,"battery":82,"day_of_week":"thursday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","linkquality":196,"load_balancing_enable":true,"load_estimate":-33,"load_room_mean":-8000,"local_temperature":28.14,"mounted_mode_active":true,"mounted_mode_control":true,"occupied_heating_setpoint":21,"pi_heating_demand":0,"preheat_status":null,"programming_operation_mode":null,"radiator_covered":false,"regulation_setpoint_offset":null,"running_state":"idle","setpoint_change_source":null,"system_mode":"heat","thermostat_vertical_orientation":false,"trigger_time":660,"update":{"state":"idle"},"update_available":null,"viewing_direction":true,"window_open_external":false,"window_open_feature":true,"window_open_internal":"closed"}'
info  2022-07-21 19:48:00: Configuring '0x0c4314fffe5de312'
info  2022-07-21 19:48:00: Device '0x0c4314fffe5de312' was updated from '{"dateCode":"20220331","softwareBuildID":"01.12.0008 01.12"}' to '{"dateCode":"20220331","softwareBuildID":"AAAABBBBCCCCDDDD"}'
info  2022-07-21 19:48:01: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20220331","software_build_id":"01.12.0008 01.12"},"id":"0x0c4314fffe5de312","to":{"date_code":"20220331","software_build_id":"AAAABBBBCCCCDDDD"}},"status":"ok","transaction":"soqrl-2"}'
error 2022-07-21 19:48:11: Failed to configure '0x0c4314fffe5de312', attempt 1 (Error: Bind 0x0c4314fffe5de312/1 hvacThermostat from '0x00124b0014b938f3/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))
info  2022-07-21 19:48:17: Configuring '0x0c4314fffe5de312'
info  2022-07-21 19:48:17: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe5de312', payload '{"adaptation_run_settings":true,"adaptation_run_status":"none","algorithm_scale_factor":1,"battery":82,"day_of_week":"thursday","external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","linkquality":196,"load_balancing_enable":true,"load_estimate":-33,"load_room_mean":-8000,"local_temperature":28.18,"mounted_mode_active":true,"mounted_mode_control":true,"occupied_heating_setpoint":21,"pi_heating_demand":0,"preheat_status":null,"programming_operation_mode":null,"radiator_covered":false,"regulation_setpoint_offset":null,"running_state":"idle","setpoint_change_source":null,"system_mode":"heat","thermostat_vertical_orientation":false,"trigger_time":660,"update":{"state":"idle"},"update_available":null,"viewing_direction":true,"window_open_external":false,"window_open_feature":true,"window_open_internal":"closed"}'
error 2022-07-21 19:48:27: Failed to configure '0x0c4314fffe5de312', attempt 2 (Error: Bind 0x0c4314fffe5de312/1 genPowerCfg from '0x00124b0014b938f3/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))
info  2022-07-21 19:49:34: Configuring '0x0c4314fffe5de312'
info  2022-07-21 19:49:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe5de312","ieee_address":"0x0c4314fffe5de312"},"type":"device_announce"}'
[... cut ...] "mounted_mode_control":true,"occupied_heating_setpoint":21,"pi_heating_demand":0,"preheat_status":false,"programming_operation_mode":null,"radiator_covered":false,"regulation_setpoint_offset":null,"running_state":"idle","setpoint_change_source":null,"system_mode":"heat","thermostat_vertical_orientation":false,"trigger_time":660,"update":{"state":"idle"},"update_available":null,"viewing_direction":true,"window_open_external":false,"window_open_feature":true,"window_open_internal":"quarantine"}'
`

javnik36, 2022-11-24:
I did upgrade of my other Danfoss TRV, this time with full logging and sniffing ([herdsman log excerpt](https://github.com/Koenkk/zigbee2mqtt/files/10086019/herdsman_stripped.txt), [debug log excerpt](https://github.com/Koenkk/zigbee2mqtt/files/10086021/debug_stripped.txt)). After upgrade experienced the issue again. I did the sniffing for the first time, but hopefully did it correctly - 3 packets which I found interesting (although I have no knowledge how to interpret this):

> ZigBee Application Support Layer Data, Dst Endpt: 13, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 13
    Cluster: OTA Upgrade (0x0019)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 246
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...0 .... = Disable Default Response: False
    Sequence Number: 42
    Command: **Upgrade End Request** (0x06)
    Payload
        Status: Success (0x00)
        Manufacturer Code: Danfoss A/S (0x1246)
        Image Type: Manufacturer Specific (0x0100)
        File Version: 0x00000112
            0000 0000 .... .... .... .... .... .... = Application Release: 0
            .... .... 0000 0000 .... .... .... .... = Application Build: 0
            .... .... .... .... 0000 0001 .... .... = Stack Release: 1
            .... .... .... .... .... .... 0001 0010 = Stack Build: 18

And first 2 responses from TRV regarding software and battery percentage:
> ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 1
    Cluster: Basic (0x0000)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 90
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 175
    Frame Control Field: Profile-wide (0x08)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...0 .... = Disable Default Response: False
    Sequence Number: 175
    Command: **Read Attributes Response** (0x01)
    Status Record, String: 20220331
        Attribute: Date Code (0x0006)
        Status: Success (0x00)
        Data Type: Character String (0x42)
        String: 20220331
    Status Record, String: AAAABBBBCCCCDDDD
        Attribute: Software Build Id (0x4000)
        Status: Success (0x00)
        Data Type: Character String (0x42)
        String: AAAABBBBCCCCDDDD

> ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 1
    Cluster: Power Configuration (0x0001)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 94
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 177
    Frame Control Field: Profile-wide (0x08)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...0 .... = Disable Default Response: False
    Sequence Number: 177
    Command: **Read Attributes Response** (0x01)
    Status Record
        Attribute: Battery Percentage Remaining (0x0021)
        Status: Success (0x00)
        Data Type: 8-Bit Unsigned Integer (0x20)
        Remaining Battery Percentage: 127,5 [%]

fiodorow, 2022-12-11:
Thank you Good Sirs for this thread. I landed here looking for solution to my problems with Popp thermostat, which is sometimes stuck and not communicating with z2m. I had done OTA update on 2 devices right after installation and wasn’t even aware they have not updated correctly (AAAA… firmware version). With instruction by ascheucher I now have both thermostats with correct FW version, hoping this will solve my issues. 

Thanks!

helgek, 2022-12-12:
Hi,

I ran the OTA with Z2M with two devices and both end up with wrong Firmware build date. I followed the hard reset procedure of ascheucher multiple times but the Firmware build date remains wrong:

Firmware build date
20210623
Firmware version
01.18.0008 01.18

I also tried the "force" firmware update procedure provided by Z2M but the device obviously rejects to overwrite the firmware again with the same version and a previous version ("Timeout: device did not request any image blocks").

I wanted to update the TRVs because of external sensors support ("radiator_covered" / "external_measured_room_sensor").

brusiniak, 2023-01-05:
Ok, I am experiencing the same symptoms. After update 2 Popp TRV devices (eT093WRO/ D5X84YU) one works smoothly (with hw firmware 20210126) and second behave very unstable (hw firmware 20221003):

1. After updated it showed a firmware AAAABBBBCCCCDDD
2. After first re-connection it shows unsuccessful pairing device (interview with device fail)
3. Then after several tryouts, pairing becoming successful and firmware version showed correctly (    "01.18.0008 01.18") but battery status was wrong -> 127,5%
4. After hour or so battery level become normal and TRV works normal as designed
5. After day or so of working device stop responding to "set" messages, so I cannot set the temperate remotely however if I change it manually it updates the values on z2m

Device reset and reconnecting  helps but steps are repeatable (except 1) . Does anyone knows if the downgrade helps?

mpuff, 2023-01-05:
has anyone the problem, that also after a delete of the device and repairing, the firmware still shows AAAABBBBCCCCDDD?

surrealo, 2023-01-05:
> has anyone the problem, that also after a delete of the device and repairing, the firmware still shows AAAABBBBCCCCDDD?

yup, I seem to recall that only by following ascheucher's steps **exactly** do things work properly. an important step is to stop z2m

atus, 2023-01-06:
Just had the same problem on [1.29.0]. Is this a bad firmware or a Z2M bug?

magicus, 2023-01-06:
My understanding is that it is a Z2M bug. There are no similar reports for alternative zigbee software.

atus, 2023-01-07:
So.. two devices completed update with 1h in between. One has the ABCD firmware, the other has a correct firmware version. I can confirm firmware date remains incorrect no matter what I do.
[image]

helgek, 2023-01-07:
Hi all, I might have an explanation why these firmware updates - at least partly - don't complete successfully: I ordered the original Ally hub recently and since I hadn't updated two of my four Allys yet I wanted to see if an update would be offered. But even after a couple of days no update was offered at all! So maybe my devices with 01.12.0008 01.12 were never meant to be updated at all. Luckily I can return my Allys to Amazon since I ordered them there. I will start a new round with either Allys or Bosch Thermostat II (once they support Conbee II).

theDiverDK, 2024-01-11:
The problem does still occur.

Yesterday 10th of January 2024 I was told all my Danfoss Ally's had an update.

I updated just one to test it, and now it's shown as:

[image]

The thermostat still works and head my room correctly.

Currently updating a single other thermostat.

surrealo, 2024-01-11:
Indeed, I have the same problem. Out of an abundance of caution I reset /re-pair (exact procedure is in this thread) the TRV so that the firmware version shows up properly. It's a pity that we still need to do that.

theDiverDK, 2024-01-11:
Hi Surrealo

I agree.

The second Ally i updated, shows the same firmware date as before, but now
also version AAABBBCCCDDD so i'll stop updating.

And will try the pair/reset trick when i get home.

Have a nice day

surrealo, 2024-01-11:
@theDiverDK 

[image]

the pair/reset works but is quite a hassle since the TRV also have to be reconfigured. On the plus side with this firmware for the first time the TRV connected directly to the coordinator.

surrealo, 2024-01-11:
I don't suppose anybody could find a changelog for version 1.28?

KennethLavrsen, 2024-01-24:
I also just upgraded Danfoss Ally and ended up with AAAABBBBCCCCDDDD so as of january 2024 this issue is still not resolved

magicus, 2024-01-24:
All my Allys started claiming there is an update available a short time ago. I doubt this is a real update. How does Z2M determine that there is an update and that it is applicable? I'm suspecting that source is incorrect.

theDiverDK, 2024-10-04:
> For anyone gettng here from search: On devices page , press blue 'i' button to reinterview. After that is showed correct (updated) version.
> 
> ![image](https://private-user-images.githubusercontent.com/10474989/373562918-d9f8b928-75c2-429f-9bbf-bf7bec3a9064.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgwMzMwMzksIm5iZiI6MTcyODAzMjczOSwicGF0aCI6Ii8xMDQ3NDk4OS8zNzM1NjI5MTgtZDlmOGI5MjgtNzVjMi00MjlmLTliYmYtYmY3YmVjM2E5MDY0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA0VDA5MDUzOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVjYTNhNzFiYjYxMzJhOWE0Nzc4ZmI0NjkyZWJlZTkxOWQ3NjBhNWM5MmU2MDM1NjBhNTZlN2FhOTUwMzk3YzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.rwg5CHwZPw9_Gh-LB_hk4kYMD7M0qNriazTR4qPE_5Q)

OMG MarioZG

Thanks a lot for your answer, it works perfectly, and is WAY easier than removing/reconnecting the thermostats.

THANKS

zsiti, 2024-10-13:
> For anyone gettng here from search: On devices page, press blue 'i' button to reinterview. After that is showed correct (updated) version.
> 

+1 Thanks @MarioZG, this indeed works!

Ramblurr, 2024-10-22:
Oh very nice :) With another winter coming up maybe I can get these guys to be up to date finally.

Ilnahro, 2024-12-27:
> For anyone gettng here from search: On devices page , press blue 'i' button to reinterview. After that is showed correct (updated) version. [...]

Life-saver. So much less effort than the steps listed above. 7/8 of mine had the issue after updating. All fixed in the first attempt with this method.

magicus, 2025-01-11:
I opened a PR to add this vital piece of information to the documentation: https://github.com/Koenkk/zigbee2mqtt.io/pull/3419.