Koenkk/zigbee2mqtt #11056: Legrand 067773 switches send two ON events when pressed once

Crocmagnon, 2022-01-27:
### What happened?

Two "on" events are sent when pressed once, which make my automations based on the number of presses fail.

### What did you expect to happen?

When pressed once, only one "on" event is sent.

### How to reproduce it (minimal and precise)

Bind a legrand wireless switch, and press the on button

### Zigbee2MQTT version

1.22.2-dev commit: 1a2943d

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

```
info  2022-01-27 13:26:59: MQTT publish: topic 'zigbee2mqtt/Interrupteur couloir centre', payload '{"action":"on","action_group":1,"battery":0,"linkquality":255,"voltage":2500}'
info  2022-01-27 13:26:59: MQTT publish: topic 'zigbee2mqtt/Interrupteur couloir centre', payload '{"action":"","battery":0,"linkquality":255,"voltage":2500}'
info  2022-01-27 13:26:59: MQTT publish: topic 'zigbee2mqtt/Interrupteur couloir centre/action', payload 'on'
info  2022-01-27 13:26:59: MQTT publish: topic 'zigbee2mqtt/Interrupteur couloir centre', payload '{"action":"on","battery":0,"linkquality":255,"voltage":2500}'
info  2022-01-27 13:26:59: MQTT publish: topic 'zigbee2mqtt/Interrupteur couloir centre', payload '{"action":"","battery":0,"linkquality":255,"voltage":2500}'
info  2022-01-27 13:26:59: MQTT publish: topic 'zigbee2mqtt/Interrupteur couloir centre/action', payload 'on'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Couloir', payload '{"brightness":254,"color_mode":"color_temp","color_temp":183,"state":"ON"}'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Ampoule couloir est', payload '{"brightness":254,"color_mode":"color_temp","color_temp":183,"color_temp_startup":65535,"linkquality":255,"state":"ON"}'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Ampoule couloir sud', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"color_temp_startup":65535,"linkquality":255,"state":"ON"}'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Couloir', payload '{"brightness":177,"color_mode":"color_temp","color_temp":183,"state":"ON"}'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Ampoule couloir est', payload '{"brightness":177,"color_mode":"color_temp","color_temp":183,"color_temp_startup":65535,"linkquality":255,"state":"ON"}'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Couloir', payload '{"brightness":177,"color_mode":"color_temp","color_temp":183,"state":"ON"}'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Ampoule couloir nord', payload '{"brightness":177,"color_mode":"color_temp","color_temp":250,"color_temp_startup":65535,"linkquality":255,"state":"ON"}'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Ampoule couloir sud', payload '{"brightness":254,"color_mode":"color_temp","color_temp":250,"color_temp_startup":65535,"linkquality":255,"state":"ON"}'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Couloir', payload '{"brightness":177,"color_mode":"color_temp","color_temp":250,"state":"ON"}'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Ampoule couloir sud', payload '{"brightness":177,"color_mode":"color_temp","color_temp":250,"color_temp_startup":65535,"linkquality":255,"state":"ON"}'
info  2022-01-27 13:27:00: MQTT publish: topic 'zigbee2mqtt/Ampoule couloir nord', payload '{"brightness":177,"color_mode":"color_temp","color_temp":250,"color_temp_startup":65535,"linkquality":255,"state":"ON"}'
```

Koenkk, 2022-01-27:
Looks like the fix for https://github.com/Koenkk/zigbee2mqtt/issues/10972 only needs to be applied to the `off` action.

could you check if all possible actions of this device work properly when using the following external converter:

```js
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('../lib/tuya');

const fzLocal = {
    command_off: {
        cluster: 'genOnOff',
        type: 'commandOff',
        convert: (model, msg, publish, options, meta) => {
            return {action: 'off'};
        },
    },
};

const definition = {
    zigbeeModel: [
        ' Remote switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
    model: '067773',
    vendor: 'Legrand',
    // led blink RED when battery is low
    description: 'Wireless remote switch CUSTOM',
    fromZigbee: [fz.identify, fz.command_on, fzLocal.command_off, fz.command_toggle, fz.legacy.cmd_move, fz.legacy.cmd_stop, fz.battery],
    exposes: [e.battery(), e.action(['identify', 'on', 'off', 'toggle', 'brightness_move_up',
        'brightness_move_down', 'brightness_stop'])],
    toZigbee: [],
    meta: {battery: {voltageToPercentage: '3V_2500'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'genLevelCtrl']);
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)

Crocmagnon, 2022-01-27:
I'm sorry, however I try to configure the custom converter it doesn't seem to change the device's description.
I already have one external converter that works, I tried removing it to leave only yours but it still doesn't load (or doesn't change the device's description).

Koenkk, 2022-01-27:
Not sure why it doesn't work, anyway I've committed the changes (since I expect it will fix it), please let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Crocmagnon, 2022-01-28:
After updating the edge addon, everything seems to work fine with these wall switches now. Thank you very much! 🎉 🙏🏻

closed 2022-01-28 by Crocmagnon