Koenkk/zigbee2mqtt #11067: Wrong temperature_unit_convert value

stomko11, 2022-01-27:
### What happened?

Using Nous E6 sensors, they publish following state to MQTT:
"temperature_unit_convert":"°C"

HA core log throws errors:
2022-01-27 23:21:01 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.guest_room_e6_sensor_temperature_unit_convert: '°C' (valid options: ['celsius', 'fahrenheit'])

In that same regard, when I select the device - exposes - temperature_unit_convert and select "celsius", display on device switches to Fahrenheit so I assume that selection is wrong. You can't switch it any way back unless you double click button on device (that switches display)

[image]

### What did you expect to happen?

that "celsius" (or fahrenheit) would be published instead of "°C" so that HA core log is not flooded with such errors AND when selecting celsius (or Fahrenheit) under device settings, that it will switch display accordingly

### How to reproduce it (minimal and precise)

select "celsius" or "fahrenheit" under "exposes" - display should switch to celsius/fahrenheit, but it does not. It switches to fahrenheit only (as a fallback?)

### Zigbee2MQTT version

 1.22.2-dev commit: 23bdc18

### Adapter firmware version

 0x26700700

### Adapter

ConBee2/RaspBee2

### Debug log

```
Debug Received Zigbee message from 'Living Room E6 Sensor', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,248],"type":"Buffer"},"datatype":2,"dp":1}],"seq":7493}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/Living Room E6 Sensor', payload '{"battery":100,"humidity":23,"linkquality":255,"max_temperature":0,"min_temperature":0,"temperature":24.8,"temperature_alarm":null,"temperature_sensitivity":0.1,"temperature_unit_convert":"°C"}'
```

HA Core log shows:
```
2022-01-27 23:20:22 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.living_room_e6_sensor_temperature_unit_convert: '°C' (valid options: ['celsius', 'fahrenheit'])
2022-01-27 23:20:22 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.living_room_e6_sensor_temperature_unit_convert: '°C' (valid options: ['celsius', 'fahrenheit'])
2022-01-27 23:20:22 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.living_room_e6_sensor_temperature_unit_convert: '°C' (valid options: ['celsius', 'fahrenheit'])
```

McGiverGim, 2022-01-28:
I don't know too much about the code of zigbee2mqtt, but I think the problem is that this defines the words `celsius` and `farenheit` for the field:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/f082fd537893c3474c2fba209a98d8c06017d1b2/devices/nous.js#L35

And the converters use `°C` and `°F`.
https://github.com/Koenkk/zigbee-herdsman-converters/blob/f082fd537893c3474c2fba209a98d8c06017d1b2/converters/fromZigbee.js#L1830-L1831
https://github.com/Koenkk/zigbee-herdsman-converters/blob/f082fd537893c3474c2fba209a98d8c06017d1b2/converters/toZigbee.js#L5149-L5151

I suppose it needs to use the same unit in all places.

stomko11, 2022-02-15:
Looking at the device https://www.zigbee2mqtt.io/devices/E6.html
All of other options are °C and °F, only temperature convert is celsius or fahrenheit.

closed 2022-03-26 by github-actions[bot]