Koenkk/zigbee2mqtt #11068: Can not set operation_mode for QBKG24LM

lufton, 2022-01-27:
### What happened?

After joining QBKG24LM I'm trying to set operation_mode for left and right buttons. But I'm getting:
`No converter available for 'operation_mode_left' ("decoupled")` and
`No converter available for 'operation_mode_right' ("decoupled")`
Other option `control_left_relay/control_right_relay` don't work either.

### What did you expect to happen?

Operation mode for buttons should change according to [documentation](https://www.zigbee2mqtt.io/devices/QBKG24LM.html#operation-mode-enum-left-endpoint).

### How to reproduce it (minimal and precise)

1. Join the device
2. Open device page
3. Open `Exposes` tab
4. Select any value for Left or Right endpoint `operation_mode` option.

### Zigbee2MQTT version

1.22.2-dev commit: 0b06483

### Adapter firmware version

20211217

### Adapter

Egony Stick V4

### Debug log

<details>
<summary>Debug log</summary>
2022-01-27T23:55:28.986Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,6,0,243,47,2,1,0,171,0,200,75,215,0,0,14,24,20,10,0,0,16,0,0,240,35,0]
2022-01-27T23:55:28.988Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [243,47,3,243,47,29,20]
2022-01-27T23:55:28.988Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,6,0,243,47,2,1,0,171,0,200,75,215,0,0,14,24,20,10,0,0,16,0,0,240,35,0,243,47,3,243,47,29,20]
2022-01-27T23:55:28.989Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 34 - 2 - 4 - 129 - [0,0,6,0,243,47,2,1,0,171,0,200,75,215,0,0,14,24,20,10,0,0,16,0,0,240,35,0,243,47,3,243,47,29] - 20
2022-01-27T23:55:28.989Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":12275,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":171,"securityuse":0,"timestamp":14109640,"transseqnumber":0,"len":14,"data":{"type":"Buffer","data":[24,20,10,0,0,16,0,0,240,35,0,243,47,3]}}
2022-01-27T23:55:28.989Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":20,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0},{"attrId":61440,"dataType":35,"attrData":53474048}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":12275,"endpoint":2,"linkquality":171,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-01-28 01:55:28: Received Zigbee message from '0x00158d00077e8c61', type 'attributeReport', cluster 'genOnOff', data '{"61440":53474048,"onOff":0}' from endpoint 2 with groupID 0
2022-01-27T23:55:28.992Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2022-01-28 01:55:28: MQTT publish: topic 'zigbee2mqtt/0x00158d00077e8c61', payload '{"action":null,"linkquality":171,"operation_mode_left":null,"operation_mode_right":null,"power":0,"state_left":"OFF","state_right":"OFF","update":{"state":null},"update_available":null}'
2022-01-27T23:55:29.408Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,34,68,129,0,0,6,0,243,47,2,1,0,182,0,184,178,215,0,0,14,24,21,10,0,0,16,1,0,240,35,0]
2022-01-27T23:55:29.409Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,6,0,243,47,2,1,0,182,0,184,178,215,0,0,14,24,21,10,0,0,16,1,0,240,35,0]
2022-01-27T23:55:29.411Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [243,47,3,243,47,29,128]
2022-01-27T23:55:29.411Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,6,0,243,47,2,1,0,182,0,184,178,215,0,0,14,24,21,10,0,0,16,1,0,240,35,0,243,47,3,243,47,29,128]
2022-01-27T23:55:29.411Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 34 - 2 - 4 - 129 - [0,0,6,0,243,47,2,1,0,182,0,184,178,215,0,0,14,24,21,10,0,0,16,1,0,240,35,0,243,47,3,243,47,29] - 128
2022-01-27T23:55:29.412Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":12275,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":182,"securityuse":0,"timestamp":14135992,"transseqnumber":0,"len":14,"data":{"type":"Buffer","data":[24,21,10,0,0,16,1,0,240,35,0,243,47,3]}}
2022-01-27T23:55:29.413Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":21,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":1},{"attrId":61440,"dataType":35,"attrData":53474048}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":12275,"endpoint":2,"linkquality":182,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-01-28 01:55:29: Received Zigbee message from '0x00158d00077e8c61', type 'attributeReport', cluster 'genOnOff', data '{"61440":53474048,"onOff":1}' from endpoint 2 with groupID 0
2022-01-27T23:55:29.415Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2022-01-28 01:55:29: MQTT publish: topic 'zigbee2mqtt/0x00158d00077e8c61', payload '{"action":null,"linkquality":182,"operation_mode_left":null,"operation_mode_right":null,"power":0,"state_left":"OFF","state_right":"ON","update":{"state":null},"update_available":null}'
2022-01-27T23:55:29.832Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,34,68,129,0,0,6,0,243,47,2,1,0,178,0,116,26,216,0,0,14,24,22,10,0,0,16,0,0,240,35,0]
2022-01-27T23:55:29.832Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,6,0,243,47,2,1,0,178,0,116,26,216,0,0,14,24,22,10,0,0,16,0,0,240,35,0]
2022-01-27T23:55:29.835Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [243,47,3,243,47,29,237]
2022-01-27T23:55:29.835Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,6,0,243,47,2,1,0,178,0,116,26,216,0,0,14,24,22,10,0,0,16,0,0,240,35,0,243,47,3,243,47,29,237]
2022-01-27T23:55:29.835Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 34 - 2 - 4 - 129 - [0,0,6,0,243,47,2,1,0,178,0,116,26,216,0,0,14,24,22,10,0,0,16,0,0,240,35,0,243,47,3,243,47,29] - 237
2022-01-27T23:55:29.835Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":12275,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":178,"securityuse":0,"timestamp":14162548,"transseqnumber":0,"len":14,"data":{"type":"Buffer","data":[24,22,10,0,0,16,0,0,240,35,0,243,47,3]}}
2022-01-27T23:55:29.836Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":22,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0},{"attrId":61440,"dataType":35,"attrData":53474048}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":12275,"endpoint":2,"linkquality":178,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-01-28 01:55:29: Received Zigbee message from '0x00158d00077e8c61', type 'attributeReport', cluster 'genOnOff', data '{"61440":53474048,"onOff":0}' from endpoint 2 with groupID 0
2022-01-27T23:55:29.837Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2022-01-28 01:55:29: MQTT publish: topic 'zigbee2mqtt/0x00158d00077e8c61', payload '{"action":null,"linkquality":178,"operation_mode_left":null,"operation_mode_right":null,"power":0,"state_left":"OFF","state_right":"OFF","update":{"state":null},"update_available":null}'
2022-01-27T23:55:30.208Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,34,68,129,0,0,6,0,243,47,2,1,0,182,0,71,118,216,0,0,14,24,23,10,0,0,16,1,0,240,35,0]
2022-01-27T23:55:30.209Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,6,0,243,47,2,1,0,182,0,71,118,216,0,0,14,24,23,10,0,0,16,1,0,240,35,0]
2022-01-27T23:55:30.211Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [243,47,3,243,47,29,182]
2022-01-27T23:55:30.211Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,6,0,243,47,2,1,0,182,0,71,118,216,0,0,14,24,23,10,0,0,16,1,0,240,35,0,243,47,3,243,47,29,182]
[... cut ...] left":"OFF","state_right":"ON","update":{"state":null},"update_available":null}'
Zigbee2MQTT:info  2022-01-28 01:55:32: MQTT publish: topic 'zigbee2mqtt/0x00158d00077e8c61', payload '{"action":"","linkquality":182,"operation_mode_left":null,"operation_mode_right":null,"power":0,"state_left":"OFF","state_right":"ON","update":{"state":null},"update_available":null}'
Zigbee2MQTT:info  2022-01-28 01:55:32: MQTT publish: topic 'zigbee2mqtt/0x00158d00077e8c61/action', payload 'single_right'
</details>

Koenkk, 2022-01-28:
Fixed by https://github.com/Koenkk/zigbee2mqtt/pull/11072

closed 2022-01-28 by Koenkk

lufton, 2022-01-28:
@Koenkk, are you sure this issue related to #11072? Because to me it looks it is something different.

z-x-f, 2022-01-29:
yes it hasn't been fixed.

Koenkk, 2022-01-29:
Is it fixed after: https://github.com/Koenkk/zigbee2mqtt/pull/11083 ?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

lufton, 2022-01-30:
Yes, the issue has been fixed now in dev branch.