Koenkk/zigbee2mqtt #11069: JASCO, 43096 - Enbrighten Zigbee Plug-in Dual-Outlet Smart Dimmer

jaydhary14, 2022-01-28:
### Link

https://manuals.plus/m/0a8133f3a13da066c19ec08248d83bb8ddf301b089c78ad4f5eeee27ec136046

### Database entry

"configured_reportings":[],"scenes":[]},"242":{"bindings":[],"clusters":{"input":[],"output":["greenPower"]},"configured_reportings":[],"scenes":[]}},"friendly_name":"0xec1bbdfffefade2f","ieee_address":"0xec1bbdfffefade2f","interview_completed":true,"interviewing":false,"manufacturer":"Jasco Products","model_id":"43096","network_address":7418,"power_source":"Mains (single phase)","supported":false,"type":"Router"}]'

### Comments

I tried to add / configure this file as per the instructions provided 

created 43096_1.js in the same directory where configuration.yaml is 

/opt/zigbee2mqtt/data$ pwd
/opt/zigbee2mqtt/data

ubuntu@zigbee:/opt/zigbee2mqtt/data$ ls -lrt
total 64
-rwxrwx--- 1 ubuntu ubuntu  3494 Jan 19 11:31 state.json.backup
-rwxrwx--- 1 ubuntu ubuntu  2171 Jan 19 11:31 database.db.backup
**-rwxrwx--- 1 ubuntu ubuntu  2325 Jan 27 19:30 configuration.yaml**
-rw-r--r-- 1 ubuntu ubuntu 19828 Jan 27 19:35 database.db
**-rw-rw-r-- 1 ubuntu ubuntu  1052 Jan 27 19:36 43096_1.js**
drwxrwx--- 3 ubuntu ubuntu  4096 Jan 27 19:36 log
-rwxrwx--- 1 ubuntu ubuntu  3910 Jan 27 19:46 state.json

updated configuration.yaml and added below entry 

  external_converters:
    - 43096_1.js
  legacy_api: false
  log_level: debug

### External converter

```shell
Tried these two configuration 

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  zigbeeModel: ['43096'],
          model: '43096',
          vendor: '‎Jasco',
          description: 'ZigBee dimmer ',
          extend: extend.light_onoff_brightness({noConfigure: true}),
          configure: async (device, coordinatorEndpoint, logger) => {
              await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
              const endpoint = device.getEndpoint(1);
              await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
              await reporting.onOff(endpoint);
};

module.exports = definition;

============================== 2nd try with new JS =======================

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['43096'],
    model: '43096',
    vendor: 'Jasco Products',
    description: 'My super switch!',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

jaydhary14, 2022-01-28:
I can see support for this device is already added 

https://github.com/Koenkk/zigbee-herdsman-converters/pull/3692

I am not sure how I can add this to my existing setting.. I tried to update zigbee2mqtt but new changes are not coming in. 

please advice what's the next step. 

I have tried to restart the services multiple time but still the status is "Unsupported"

Koenkk, 2022-01-28:
This device has been added this month, so it is not supported in 1.22.2.

- It will be in the next release on 1 February
- It is already available in the dev branch (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-28 by Koenkk

jaydhary14, 2022-01-28:
great thanks will there be any communication after release ?

Koenkk, 2022-01-29:
@jaydhary14 yes, under https://github.com/Koenkk/zigbee2mqtt/releases