Koenkk/zigbee2mqtt #11080: [New device support]: Philips Hue Amaze 4023330P6

nikolovnet, 2022-01-28:
### Link

https://www.philips-hue.com/cs-cz/p/hue-white-ambiance-zavesne-svitidlo-amaze/4023330P6

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0x0017880109ef25d0","nwkAddr":41345,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"4023330P6","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"4023330P6","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20190529","swBuildId":"1.56.7_r29434"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20190529","swBuildId":"1.56.7_r29434","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1642921265044,"useImplicitCheckin":true}

### Comments

This seems to be the same device as 4023330P7, which is already supported by zigbee2mqtt.

Koenkk, 2022-01-29:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

nikolovnet, 2022-01-29:
'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}

Thank you!

Koenkk, 2022-01-29:
added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-29 by Koenkk