Koenkk/zigbee2mqtt #11081: TS1101_dimmer_module_1ch TUYA dimmer switch not fully supported

sjors98, 2022-01-28:
### What happened?

I just bought the [Tuya dimmer switch: TS1101_dimmer_module_1ch](https://www.zigbee2mqtt.io/devices/TS1101_dimmer_module_1ch.html) which is listed as supported on the site, however after pairing I get this in response: 
```{"message":"interview_successful","meta":{"friendly_name":"0xa4c1380601b91d49","supported":false},"type":"pairing"}```
I tried repairing a couple of times next to the zigbee module, but I can't get it to work. I am running Zigbee2MQTT 1.22.2 (supported since this version according to https://github.com/Koenkk/zigbee2mqtt/discussions/10473 )

### What did you expect to happen?

I expect it to pair and recognize the device so that supported will be true

### How to reproduce it (minimal and precise)

- Get this device
- Put it in pairing mode
- Hold next to the zigbee stick
- Observe the zigbee2mqtt/log logs in a mqtt broker

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

2018090201

### Adapter

Texas Instruments CC2531

sjors98, 2022-01-28:
It has been a while since I last used zigbee2mqtt and I see some major updates, nice work @Koenkk !!
Could my issue be caused by an outdated firmware on the stick? I know it's a Texas Instruments CC2531 and you don't recommend this, but it's all I have at the moment.. Will look into upgrading soon!
The firmware on this stick is from 2018 (pre covid firmware holy crap), could it be caused by this outdated firmware? If so I'll give upgrading the stick a shot (a friend of mine has a CC debugger so can't do that right now :/ )

Koenkk, 2022-01-29:
@sjors98 device support (in most cases) has nothing to do with the stick firmware. Probably your device uses a slightly different identifier. Could you provide the data/database.db entry of your device? Adding it [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/72b18217beb0e187061e692be7d6dd634c5493ec/devices/tuya.js#L1237) should make it supported

sjors98, 2022-01-30:
Thanks for the quick reply @Koenkk !! 
This is what I find in the database.db for this device:
`{modelID: 'TS110E', manufacturerName: '_TZ3210_zxbtub8r'}`

Seems to be different indeed so this is probably what's causing the issue :( 
I'm unable to make a PR for zigbee herdsman right now, any chance you can add this? :)

sjors98, 2022-01-31:
Hej @Koenkk ! I was on my phone yesterday, but made a [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3784) just now :) This is what you meant right?

sjors98, 2022-02-01:
As discussed in: https://github.com/Koenkk/zigbee-herdsman-converters/pull/3784
This device was already supported on the dev branch 🥳

closed 2022-02-01 by sjors98