Koenkk/zigbee2mqtt #11082: [New device support]: Orvido Multi-functional 3 gang relay. Same device, different ID?

CaptFennec, 2022-01-28:
### Link

https://www.orvibo.com/en/product/controlbox.html

### Database entry

?

### Comments

Hi everyone,

Sorry I'm just a noob. I've bought Orvido Multi-functional relays and they're not supported.
The strange thing is 2 very similar devices are already supported:
https://www.zigbee2mqtt.io/devices/T18W3Z.html
https://www.zigbee2mqtt.io/devices/RL804QZB.html

After pairing, the zigbeeModel I got is : 396483ce8b3f4e0d8e9d79079a35a420
Could it be that the only difference is the ID?

I'm don't know how to test that myself :(

Thanks for any help you can give!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['396483ce8b3f4e0d8e9d79079a35a420'], // different from "82c167c95ed746cdbd21d6817f72c593" on https://zigbee.blakadder.com/Orvibo_CM10ZW.html
    model: 'CM10ZW', // 
    vendor: 'ORVIBO', // Vendor of the device
    description: 'Multi-functional 3 gang relay', // "Multi-functional Relay" on device's back
    fromZigbee: [], // Nothing here
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.switch().withEndpoint('l3')],
    endpoint: (device) => {
        return {l1: 1, l2: 2, l3: 3};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
        const endpoint3 = device.getEndpoint(3);
        await reporting.bind(endpoint3, coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;
```

sj8023ld, 2022-01-29:
same device，different ID
the newest model is CM10ZW

CaptFennec, 2022-01-29:
OK, thanks.
I've added the database.db entry.

How can I add that new ID (even if it's local/manual).

Thanks again!

Koenkk, 2022-01-31:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3785

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-31 by Koenkk