Koenkk/zigbee2mqtt #11087: 'Write for 'BUFFER' not available' when running touch link identify

JoeGandy, 2022-01-29:
### What happened?

Ran touch link scan,
Ran touch link identify with the results of scan.

### What did you expect to happen?

Expected light to blink

### How to reproduce it (minimal and precise)

- Run touchlink scan
- Run touchlink identify with results

### Zigbee2MQTT version

dev branch latest

### Adapter firmware version

20211217

### Adapter

CC2652P

### Debug log

[31mZigbee2MQTT:error[39m 2022-01-29 09:52:17: Request 'zigbee2mqtt/bridge/request/touchlink/identify' failed with error: 'Write for 'BUFFER' not available'

[34mZigbee2MQTT:debug[39m 2022-01-29 09:52:17: Error: Write for 'BUFFER' not available

at BuffaloZnp.write (/app/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:274:19)
at ZpiObject.createPayloadBuffer (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:118:21)
at ZpiObject.toUnpiFrame (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:52:29)
at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:305:34)
at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:42)
at /app/node_modules/zigbee-herdsman/src/utils/queue.ts:21:18
at new Promise (<anonymous>)
at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:19:16)
at Znp.request (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:302:27)
at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:835:28)
[32mZigbee2MQTT:info [39m 2022-01-29 09:52:17: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/identify', payload '{"data":{},"error":"Write for 'BUFFER' not available","status":"error"}'

Koenkk, 2022-01-30:
I cannot reproduce this on the latest-dev:

```
Zigbee2MQTT:debug 2022-01-30 13:43:13: Received MQTT message on 'zigbee2mqtt/bridge/request/touchlink/identify' with data '{"channel":20,"ieee_address":"0x90fd9ffffe92968e","transaction":"lyx87-3"}'
Zigbee2MQTT:info  2022-01-30 13:43:13: Start Touchlink identify of '0x90fd9ffffe92968e' on channel 20
Zigbee2MQTT:info  2022-01-30 13:43:16: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/identify', payload '{"data":{"channel":20,"ieee_address":"0x90fd9ffffe92968e"},"status":"ok","transaction":"lyx87-3"}'
Zigbee2MQTT:debug 2022-01-30 13:43:20: Received MQTT message on 'zigbee2mqtt/bridge/request/touchlink/scan' with data '{"transaction":"lyx87-4","value":true}'
Zigbee2MQTT:info  2022-01-30 13:43:20: Start Touchlink scan
Zigbee2MQTT:info  2022-01-30 13:43:31: Finished Touchlink scan
Zigbee2MQTT:info  2022-01-30 13:43:31: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{"found":[{"channel":20,"ieee_address":"0x90fd9ffffe92968e"}]},"status":"ok","transaction":"lyx87-4"}'
```

Are you sure you are on the latest-dev, what is shown under frontend -> settings -> about

JoeGandy, 2022-02-11:
Hi apologies on the slow reply

I must've been doing something wrong, I was initially using the api to do this request, but when I rand it through the fron end it works fine.

closed 2022-02-11 by JoeGandy