Koenkk/zigbee2mqtt #11101: [Feature request]: Configure a global color_mode for reading colors

MartinMuzatko, 2022-01-29:
### Is your feature request related to a problem? Please describe

I'm always frustrated when I want to retrieve the color of a light and I need to convert x/y to rgb.

```
#publish
mosquitto_pub -t "zigbee2mqtt/0x00124b0021f2ec79/get" -m '{"state":{}}'

#message received
zigbee2mqtt/0x00124b0021f2ec79 {"brightness":254,"color":{"x":0.1355,"y":0.0399},"color_mode":"xy","color_temp":574,"linkquality":69,"state":"ON"}
```

This also applies to the `expose` I retrieve when pairing a new device.

### Describe the solution you'd like

I would like to be able to set a global `color_mode` for all light exposes of a device.
By default it would be xy, but would also support rgb.

### Describe alternatives you've considered

Trying to set the color_mode does not do anything at all.

```
mosquitto_pub -t "zigbee2mqtt/0x00124b0021f2ec79/set" -m '{"color_mode":"rgb"}'
```
On next retrieve I still get `"color_mode": "xy"`

### Additional context

I'm on docker image `koenkk/zigbee2mqtt:latest-dev`
Using Slae.sh Zigbee USB Stick.

sjorge, 2022-01-29:
There is no RGB colorMode defined in the ZCL.

https://raw.githubusercontent.com/Koenkk/zigbee-herdsman/master/docs/07-5123-08-Zigbee-Cluster-Library.pdf
[image]

Also color_mode is not settable at all, for bulbs with color support it is the mode that it is currently using.

Most likely this will be xy or color_temp as z2m convert a hex or rgb value if you pass it, to xy.

MartinMuzatko, 2022-01-30:
But it would be possible independent from color_mode to configure the output of lights/colors to transform to rgb before it is sent to the broker. I'm willing to help to implement that, if you agree that this something users should be able to configure.

sjorge, 2022-01-30:
I think it already includes r, g, b, sub values if some option was set. I’ll have a closer look tomorrow when I am at my desk. I’m like 80% sure I touched that code when working on the color sync stuff.

sjorge, 2022-01-31:
Couldn't find it.

@Koenkk did we not output `{"r": 0, "g": 0, "b": 0}` values in some cases? I vaguely remember if HA was enabled?

Koenkk, 2022-01-31:
@sjorge 

>  I vaguely remember if HA was enabled?

I think you are talking about the h/s thing: https://github.com/Koenkk/zigbee2mqtt/blob/1a0a9a6eabf58855d3a6feda60bb79a1c34d17f8/lib/extension/homeassistant.ts#L1314

I'm not sure if we should publish an rgb at all since it is not a Zigbee color mode (only xy and hs are)

MartinMuzatko, 2022-01-31:
It makes zigbee2mqtt hard to use for any node-red applications, since xy to rgb conversions nodes/plugins are very rare and barely usable. 
I found one node that takes xyz color code, but not xy :D

A global configuration in the settings would make this a lot easier, but I understand if it is outside the scope of this software.

sjorge, 2022-01-31:
I have a node on my flow that sets this when it is triggered by a startup inject.

```node
const color = {
    // convert mired shift to tegrees kelvin
    mired_to_kelvin: function (mired) {
        return Math.floor(1000000 / mired);
    },

    // convert degrees kelvin to mired shift
    kelvin_to_mired: function (kelvin) {
        return Math.floor(1000000 / kelvin);
    },

    // convert degrees kelvin to rgb
    kelvin_to_rgb: function (kelvin) {
        /* Return an RGB color from a color temperature in Kelvin.
         * This is a rough approximation based on the formula provided by T.Helland
         * http://www.tannerhelland.com/4435/convert-temperature-rgb-algorithm-code/
         */
        /* internal helpers */
        // limit given color component to range
        function _clamp(component, minimum = 0.0, maximum = 255.0) {
            // NOTE: ensure params are floats
            component = component * 1.0;
            minimum = minimum * 1.0;
            maximum = maximum * 1.0;

            return parseInt(
                Math.min(
                    Math.max(component, minimum),
                    maximum
                ),
            );
        }

        // get red component of temperature in RGB space
        function _get_red (kelvin) {
            if (kelvin <= 66)
                return 255

            const tmp = 329.698727446 * Math.pow(kelvin - 60, -0.1332047592)
            return _clamp(tmp);
        }

        // get red component of temperature in RGB space
        function _get_green(kelvin) {
            let tmp;
            if (kelvin <= 66) {
                tmp = 99.4708025861 * Math.log(kelvin) - 161.1195681661;
            } else {
                tmp = 288.1221695283 * Math.pow(kelvin - 60, -0.0755148492);
            }
            return _clamp(tmp);
        }

        // get blue component of temperature in RGB space
        function _get_blue(kelvin) {
            if (kelvin >= 66) {
                return 255
            } else if (kelvin <= 19) {
                return 0
            }
            const tmp = 138.5177312231 * Math.log(kelvin - 10) - 305.0447927307;
            return _clamp(tmp);
        }

        // NOTE: range check
        if (kelvin < 1000) {
            kelvin = 1000
[... 155 lines cut ...]

        return [
            parseInt(r * 255),
            parseInt(g * 255),
            parseInt(b * 255),
        ];
    },

    // convert hex notation to rgb
    hex_to_rgb: function (hex) {
        if (hex.startsWith("#")) {
            hex = hex.substring(1);
        }

        return [
            parseInt(hex.substring(0, 2), 16),
            parseInt(hex.substring(2, 4), 16),
            parseInt(hex.substring(4, 6), 16),
        ];
    },
};
flow.set("libcolor", color);
```

I can then grab it using `const color = flow.get("libcolor");` in other functions nodes on the flow.

sjorge, 2022-01-31:
> I'm not sure if we should publish an rgb at all since it is not a Zigbee color mode (only xy and hs are)

Yeah I did that was indeed the conclusion from the whole colorSync discussion too.

closed 2022-03-11 by github-actions[bot]