Koenkk/zigbee2mqtt #11106: SyntaxError: Unexpected end of JSON input -> help point to the file in error

lunarok, 2022-01-29:
### What happened?

After a restart of server for something not related to z2m, the container didn't start anymore with this error :

SyntaxError: Unexpected end of JSON input

I try restart, remove key ... then take a look at issues. I found that one :
https://github.com/Koenkk/zigbee2mqtt/issues/7907

And it helps me point out that yes my coordinator backup was empty, so I insert a backup I have, restart container and all was ok.

### What did you expect to happen?

The previous issue was closed by owner, but will be nice if some error message enhancement could be done like pointing to the file that cannot be parsed.

### How to reproduce it (minimal and precise)

delete content of coordinator backup json

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210218

### Adapter

ZigStar Stick v4

yuriyb80, 2022-02-04:
Zigbee2mqtt
Current version: 1.23.0-1

was same error  
 this solution it work  after delete of coordinator backup json  

Thanks !

Koenkk, 2022-02-04:
Added this now;
- A better error is shown when the backup is corrupt
- Chances of this happening have been greatly reduced (first a temp file is written which is renamed once write is finished)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

KuzinHouse, 2022-02-07:
Done. Deleted last device in database.

closed 2022-03-20 by github-actions[bot]