Koenkk/zigbee2mqtt #11109: [New device support]:

rmuk, 2022-01-30:
### Link

https://lonsonho.com/products/zigbee-smart-led-light-5w-gu10-holder-lamp-lights-rgb-cw-dimmable-2-pack

### Database entry

{"id":42,"type":"Router","ieeeAddr":"0x00124b00226d594e","nwkAddr":14772,"manufId":0,"manufName":"eWeLight","powerSource":"Mains (single phase)","modelId":"ZB-CL01","epList":[1,13],"endpoints":{"1":{"profId":49246,"epId":1,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"ZB-CL01","manufacturerName":"eWeLight","powerSource":1,"zclVersion":1,"appVersion":13,"stackVersion":2,"hwVersion":1,"dateCode":"20201216","swBuildId":"1.0.13"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorMode":1,"currentX":40237,"currentY":24445,"colorTemperature":370}},"genLevelCtrl":{"attributes":{"currentLevel":3}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":13,"stackVersion":2,"hwVersion":1,"dateCode":"20201216","swBuildId":"1.0.13","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1643503033095,"useImplicitCheckin":true}

### Comments

The bulb is identified in Zigbee2MQTT as:

Manufacturer: Lonsonho
Model: ZB-RGBCW
Zigbee Manufacturer: eWeLight
Zigbee Model: ZB-CL01

They are similar to appearance to the ones shown in the Product URL, but there are some minor differences in detailing and proportions compared to what's shown on that page; either the bulbs I have are an earlier revision or a knock-off (I got them in an unmarked pack of 8 from a charity shop for £5, so that's possible).

The issue it turning the bulbs off: setting the brightness to 1/255 turns the bulbs off no problem, but setting the bulbs to 0/255 brightness or explicitly turning them off has no effect and with "Publish Optimistic State" turned off this is correctly reflected in Zigbee2MQTT.

I tried to make an external converter with the "turnsOffAtBrightness1" parameter included since I suspect that will solve the issue but I'm a bit out of my depth. If anyone wants me to test anything please let me know. FWIW, I'm using Zigbee2MQTT as part of a Home Assistant setup, so I'm working around the issue at the moment with a HA automation.

### Supported color modes

color temperature, color (rgb)

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}

reaper7, 2022-01-30:
I have the same problem with this model.
There is no way to turn it off.

rmuk, 2022-01-30:
@reaper7 It's not ideal, but I've found a workaround. Firstly, I have four three-bulb lamps and I was able to shuffle my existing bulbs around so each cluster has two of these Lonsonho bulbs and one trustworthy bulb, either a Ikea or Lidl one. Then I made a quick automation in Home Assistant to set the Lonsonho bulbs to 1/255 brightness when the Ikea/Lidl bulb was turned off:

`alias: Shit Bulbs - Bedroom Spotlights
description: >-
  The shit bulbs don't turn off it told to (by calling the light.turn_off
  service by setting the brightness to 0); instead they turn off when their
  brightness is set to 1/255. This automation does that if another nearby,
  trustworthy, bulb is turned off.
trigger:
  - platform: device
    type: turned_off
    device_id: a7c3c36a4d76ad4eb247637a4415fddd
    entity_id: light.0x804b50fffea54e77
    domain: light
condition: []
action:
  - service: light.turn_on
    data:
      brightness: 1
      transition: 1
    target:
      device_id:
        - 9754040dfbacdf13ced7448d0f4349ae
        - 5d2f214ebab4163ba21c3c163bdf00be
        - 46c608790fe7351da9fbd10c0cf8b0b1
  - service: light.turn_off
    target:
      device_id:
        - 9754040dfbacdf13ced7448d0f4349ae
        - 5d2f214ebab4163ba21c3c163bdf00be
        - 46c608790fe7351da9fbd10c0cf8b0b1
mode: single`

It's far from ideal, but it does the job. Hopefully that helps until someone smarter than me can put a permanent fix in place!

reaper7, 2022-01-30:
@rmuk - in your case brightness 1 turns off this bulb ? or only sets the minimum brightness ?

rmuk, 2022-01-30:
@reaper7 Brightness level 1 turns them off. Brightness level 2 sets them to the lowest brightness setting. Level 0 does nothing. Also, these are brightness levels out of 255, they're not brightness percentages.

reaper7, 2022-01-30:
@rmuk - you're right, works by HA service:
setting the brightness [0-255] to 1 turns the light off

but not via HA scene :(
```
- id: kitchen_lights_off
  name: Kitchen Lights Off
  entities:
    light.kitchen_light_two:
      brightness: 1
      color_temp: 370
      transition: 1
      state: on
```
in this case the minimum brightness will then be achieved

rmuk, 2022-01-30:
@reaper7 Try getting rid of the `color_temp: 370` line. On my bulbs they only turn off if the brightness value is the only thing set. If I include colour temperature, rgb, etc they stay on.

reaper7, 2022-01-30:
works! tnx!

closed 2022-03-09 by github-actions[bot]

messyjoes1, 2022-07-05:
This is still an issue, bulb can't be turned off with a simple payload like - mqtt.send('zigbee2mqtt/bulb/set', { state: 'OFF' }), only brightness=1 hack works

pvince83, 2022-08-05:
I confirm,

Anyone to help?

I can make tests, captures.... if it can be help

PCServices2, 2022-09-01:
Still having to same problem

rombarcz, 2022-10-08:
I've had issue with Lonsonho ZB-RGBCW: it would not turn off, setting brigthess to 1 was switching it off.

I've managed to make it work again by REMOVING from `configuration.yaml` parameters:
```
transition: 0
color_sync: false
```
1. Stop zigbee2MQTT
2. edit configuration.yaml (create backup first!)
3. Start zigbee2MQTT

cmiguelcabral, 2023-02-16:
Hi there,
I created this external converter to fix the problem of turning off these bulbs, with it they work as any other bulb:
[lonsonho.zip](https://github.com/Koenkk/zigbee2mqtt/files/10750591/lonsonho.zip)

Actually, this is the default implementation of `light_onoff_brightness`, only changed what's on line  131 (on my file).

suchkovdu, 2023-08-27:
> Привет, я создал этот внешний преобразователь, чтобы решить проблему с выключением этих лампочек, с ним они работают как любая другая лампочка: [lonsonho.zip](https://github.com/Koenkk/zigbee2mqtt/files/10762036/lonsonho.zip)
> 
> На самом деле, это реализация по умолчанию `light_onoff_brightness`, изменено только то, что написано в строке 131 (в моем файле).

please tell me where this file is located?